mikakaraila/node-red-contrib-opcua #1: Connection errors

industrialinternet, 2015-07-02:
Mika thanks for this great news.
I'm encountering the following errors.
Have followed install instructions & nmp modules are up to date.
I feel there is something I'm missing?
 
 Jul 21:01:34 - [info] Settings file  : ./settings
2 Jul 21:01:34 - [info] User Directory : /home/pi/.node-red
2 Jul 21:01:34 - [info] Flows file     : /home/pi/.node-red/flows_NRdev.json
2 Jul 21:01:34 - [info] Server now running at http://127.0.0.1:1880/
2 Jul 21:01:34 - [info] Starting flows
Test server (read items) connecting to opc.tcp://localhost:4334
Test server (browse) connecting to opc.tcp://localhost:4334
Test server (write items) connecting to opc.tcp://localhost:4334
Test server (subscribe item) connecting to opc.tcp://localhost:4334
2 Jul 21:01:34 - [info] Started flows
2 Jul 21:01:34 - [info] [OpcUaClient:Test server (read items)] Test server (read items) client : process terminated with an error
2 Jul 21:01:34 - [info] [OpcUaClient:Test server (read items)]  error :
2 Jul 21:01:34 - [info] [OpcUaClient:Test server (read items)] Error: connect ECONNREFUSED
    at errnoException (net.js:905:11)
    at Object.afterConnect [as oncomplete] (net.js:896:19)
2 Jul 21:01:34 - [info] [OpcUaClient:Test server (browse)] Test server (browse) client : process terminated with an error
2 Jul 21:01:34 - [info] [OpcUaClient:Test server (browse)]  error :
2 Jul 21:01:34 - [info] [OpcUaClient:Test server (browse)] Error: connect ECONNREFUSED
    at errnoException (net.js:905:11)
    at Object.afterConnect [as oncomplete] (net.js:896:19)
2 Jul 21:01:34 - [info] [OpcUaClient:Test server (write items)] Test server (write items) client : process terminated with an error
2 Jul 21:01:34 - [info] [OpcUaClient:Test server (write items)]  error :
2 Jul 21:01:34 - [info] [OpcUaClient:Test server (write items)] Error: connect ECONNREFUSED
    at errnoException (net.js:905:11)
    at Object.afterConnect [as oncomplete] (net.js:896:19)
2 Jul 21:01:34 - [info] [OpcUaClient:Test server (subscribe item)] Test server (subscribe item) client : process terminated with an error
2 Jul 21:01:34 - [info] [OpcUaClient:Test server (subscribe item)]  error :
2 Jul 21:01:34 - [info] [OpcUaClient:Test server (subscribe item)] Error: connect ECONNREFUSED
    at errnoException (net.js:905:11)
    at Object.afterConnect [as oncomplete] (net.js:896:19)

mikakaraila, 2015-07-03:
If i remember async is missing from package....

industrialinternet, 2015-07-04:
Hi I had installed async. No start-up connection Errors if you run the node-opcua sample_server.js then Node-Red. 

Then you try and Browse

/4/2015, 9:34:00 AM[47119c58.b8ee64]
ns=0;i=61 : [msg.payload] : string
[{"id":"4b12ca9b.e7e184","type":"OpcUaItem","item":"ns=0;i=61","datatype":"opcua.DataType.Double","value":"66.6","name":"FolderType","x":251,"y":334,"z":"30ffd2ee.59fdd6","wires":[["70dd1397.3c8e44"]]}]
7/4/2015, 9:34:00 AM[47119c58.b8ee64]
ns=1;i=1001 : [msg.payload] : string
[{"id":"4b12ca9b.e7e184","type":"OpcUaItem","item":"ns=1;i=1001","datatype":"opcua.DataType.Double","value":"66.6","name":"MyVariable1","x":251,"y":334,"z":"30ffd2ee.59fdd6","wires":[["70dd1397.3c8e44"]]}]
7/4/2015, 9:34:00 AM[47119c58.b8ee64]
ns=4;b=1020ffaa : [msg.payload] : string
[{"id":"4b12ca9b.e7e184","type":"OpcUaItem","item":"ns=4;b=1020ffaa","datatype":"opcua.DataType.Double","value":"66.6","name":"MyVariable2","x":251,"y":334,"z":"30ffd2ee.59fdd6","wires":[["70dd1397.3c8e44"]]}]
7/4/2015, 9:34:00 AM[47119c58.b8ee64]
ns=4;s=free_memory : [msg.payload] : string
[{"id":"4b12ca9b.e7e184","type":"OpcUaItem","item":"ns=4;s=free_memory","datatype":"opcua.DataType.Double","value":"66.6","name":"FreeMemory","x":251,"y":334,"z":"30ffd2ee.59fdd6","wires":[["70dd1397.3c8e44"]]}]
7/4/2015, 9:34:00 AM[Address.txt]
[msg] : string
failed to append to file : Error: ENOENT, open './public/Address.txt'
7/4/2015, 9:34:00 AM[OPC UA Items]
[msg] : string
failed to write to file : Error: ENOENT, open './lib/templates/OPCUA/FolderType.js'
7/4/2015, 9:34:00 AM[Address.txt]
[msg] : string
failed to append to file : Error: ENOENT, open './public/Address.txt'
7/4/2015, 9:34:00 AM[OPC UA Items]
[msg] : string
failed to write to file : Error: ENOENT, open './lib/templates/OPCUA/MyVariable1.js'
7/4/2015, 9:34:00 AM[Address.txt]
[msg] : string
failed to append to file : Error: ENOENT, open './public/Address.txt'
7/4/2015, 9:34:00 AM[OPC UA Items]
[msg] : string
failed to write to file : Error: ENOENT, open './lib/templates/OPCUA/MyVariable2.js'
7/4/2015, 9:34:00 AM[Address.txt]
[msg] : string
failed to append to file : Error: ENOENT, open './public/Address.txt'
7/4/2015, 9:34:00 AM[OPC UA Items]
[msg] : string
failed to write to file : Error: ENOENT, open './lib/templates/OPCUA/FreeMemory.js'

mikakaraila, 2015-10-07:
Now I have time to look this... updating parts of code.

mikakaraila, 2016-01-15:
There was general mistake with session handling.
I added empty subscription that will keep session alive now.

closed 2016-01-15 by mikakaraila

industrialinternet, 2016-01-21:
Thanks so much for looking at this will test when I have time hopefully in next few weeks