mikakaraila/node-red-contrib-opcua #3: Only one node visible: OpcUaItem - missing OpcUaClient & OpcUaBrowse

thfrei, 2015-11-30:
I cloned the current node-red repository (v0.12.2 according to node-red package.json), `npm install`ed it, and installed the node-red-contrib-opcua via `npm install git://github.com/mikakaraila/node-red-contrib-opcua`.
However, I can only find the _OpcUaItem_ under _Advanced_, and I am missing the `OpcUaClient` `OpcUaBrowse` node. (see attached screenshot)
Any ideas why? 

Thank you for your feedback.
Please let me know, if you need more information.
[image]

thfrei, 2015-11-30:
Just as I posted it, I thought about checking it on Linux (I was previously running Windows 10, where there seems to be quite a lot of certificate issues). Under Linux it works.

mikakaraila, 2015-12-08:
Ack,

I will try to find time to update code. I modified it as I tested it on IBM
Bluemix...

Br,
Mika

mikakaraila, 2016-01-15:
New refactored with better browse and session handling updated.
Please test with the latest one.

mikakaraila, 2016-01-15:
I added new example flow:
http://flows.nodered.org/flow/f2e2c45b96dbc7ce22b5
Package is available also in npm.

markrey, 2016-01-21:
Hi Mika,

Thank you for the update.
Still one node is missing named "OPCUA".

As defined in,
https://github.com/mikakaraila/node-red-contrib-opcua/blob/649392de36baa4491fe29112e820970bc0205b4d/103-opcua.js

I would like to put username and password to get write access in OPC Server.

Thanks

mikakaraila, 2016-01-22:
Ack,

I will look these during weekend...

Br,
Mika

mikakaraila, 2016-01-24:
Browse fixed and example added. I will add later username & password and perhaps security into client parameters.

mikakaraila, 2016-03-28:
Klaus refactored code and implemented good improvements.

closed 2016-04-04 by mikakaraila