mikakaraila/node-red-contrib-opcua #11: OpcUa-Client Node Missing

synergeticon, 2016-05-02:
After installing with `npm install node-red-contrib-opcua` there is no node "npm install node-red-contrib-opcua" available in node-red.

[image]

biancode, 2016-05-02:
I've made a npm uninstall and npm install and it works well on different systems
[image]

biancode, 2016-05-02:
please check your installation by starting command verbose mode ->  node-red -v 
may your node-opcua installation is not done with exit 0  -  on Windows it is important that your Username /Users/ ...  contains no spaces

biancode, 2016-05-02:
you need to see this

Executing  node bin/generate_opcua_classes.js --clear 
done ... (0)
Executing  node bin/crypto_create_CA.js 
done ... (0)

sandro-k, 2016-05-03:
I am getting the following error 

```
[OpcUa-Client] Error: Cannot find module 'collections/set'
```

synergeticon, 2016-05-03:
My Node Version: v4.2.4
Hey, yes I get this:

`Executing node bin/generate_opcua_classes.js --clear
done ... (0)
Executing node bin/crypto_create_CA.js
done ... (0)`

I tried it on a virtual machine with Xubuntu on my PC... Same Result. **But** on a different Server it worked!

sandro-k, 2016-05-03:
It looks like, that `node-red-contrib-opcua` or require needs `collections/set` to be installed globally to find it. I tried to install  `collections/set` into the `node_modules` folder of `node-red` and in  `node-red-contrib-opcua/node_modules`, both did not work. After installing `collections` globally the `OpcUa Client` can be loaded without a problem.

biancode, 2016-05-03:
OK, next version will bring this dependency.

mikakaraila, 2016-05-03:
Ack, merge & publish done.

closed 2016-10-30 by mikakaraila