mikakaraila/node-red-contrib-opcua #16: Read OPC Item Double Message

yonas124, 2016-10-28:
Hi Mikakaraila,

I tried to read opc item using flow below:
[image]

As you can see the message shows 2 time which makes the gauge reading awkward.
Is this the correct flow? Or did I do it incorrectly? If I use OpcUa-Item then inject, the result is the same.

node-red-contrib-opcua version: 0.2.7
Node-RED version: v0.15.1
Node.js  version: v4.6.1
Linux 4.4.11-v7+ arm LE

allatrue, 2016-10-28:
I can confirm this. Also using opcua client subscribe.

yonas124, 2016-10-29:
In the end this is my workaround:
[image]
The 'Switch' is passing the value to ouput if input is > -1 (inject number is -1)

mikakaraila, 2016-10-30:
Hmm most probably automatic typing in javscript causes this. 
Have you tried with OPC Item node and use Double with it?
[image]

mikakaraila, 2016-10-30:
Above is support for other basic types, I will push new version after some testing.

yonas124, 2016-10-30:
mikakaraila
You did a good job on the module, it's easy to use and works right away without too much effort.
Yes I did try with OpcUa-Item and used different type of data with same result, msg goes 2 times.

Anyway, my other issues are: I don't know where to change the timeout duration, subscription rate, and my OpcUa-Client Write doesn't seem to work (it shows 'active writing' but tag isn't written).

mikakaraila, 2016-10-30:
Uups, those are not taken out as parameters. They are inside code as defaults.
I will look them later. Now just testing all basic type conversions.

mikakaraila, 2016-10-30:
I just pushed fixes, could you test your case. I will then publish this to npm.

yonas124, 2016-10-31:
How do I apply the fixes?

allatrue, 2016-10-31:
The problem still stays for me. I get **first** Msg from OpcUa-Client using SUBSCRIBE with the value set in OpcUa-Item. Also if connection failed.
In your last commit to opcuaclient.js you have overwritten changes from #17. Is there any reason for that?

mikakaraila, 2016-10-31:
Ups, my mistake yesterday... have to check later today.

mikakaraila, 2016-10-31:
#17 merged back. I will check flow with subscription later.

mikakaraila, 2016-10-31:
Ack now I can repeat problem. Inject sends "empty" payload that will come
out through the opc-node.
Normally I use inject once to create subscription. I have not looked this
so closely...
Fastest way is to use subscription (predefined default parameters), but I
can look this today at home.

mikakaraila, 2016-10-31:
Found it! Some testing and fix will be available.

closed 2016-10-31 by mikakaraila

allatrue, 2016-11-01:
Thanks a lot! But it worked just partly for me.
If connection cannot be established I get warning message "can't work without OPC UA session" and msg goes through the opcua-client node with initial payload.
```
           if (!node.client || !node.session) {
                verbose_warn("can't work without OPC UA Session");
                reset_opcua_client(connect_opcua_client);
                node.send(msg); // is it necessary?
                return;
            }
```

mikakaraila, 2016-11-01:
Ok, I will fix this after normal working day. I will add some types at the
same time... I did not finalize everything yesterday.

mikakaraila, 2016-11-01:
As a solution 4 lines commented out. So in case of error node.send(msg) is
not used.

commit 61533092 (2016-10-31): Fixed #16, inject sends payload that goes through opc client node.