mikakaraila/node-red-contrib-opcua #18: Failed to write OPC item

yonas124, 2016-11-05:
Hi again,

I got this msg when I tried to write opc item using the latest version (0.2.9):
> msg : error
> AssertionError: this.identifierType !== NodeIdType.NUMERIC || (this.value >= 0 && this.value <= 0xFFFFFFFF)

Payload is:
> ns=2;s=OPCsim.Plant3.PV1.DC_Voltage : msg.payload : number
> 123

Data type is int16

Please advice

mikakaraila, 2016-11-05:
Ok, for some reason it seems to parse node address as numeric like
ns=2;i=2004 not as you have ns=2;s=StringAddress

C0ppy, 2016-11-05:
I have the exact same problem.
[image]
Could this be the problem?

mikakaraila, 2016-11-05:
No but if there is a space as a first character...

C0ppy, 2016-11-05:
Read and Subscribe works for the exactly same address.

mikakaraila, 2016-11-05:
Strange... I have IoT panel starting soon. I have to take closer look
tomorrow.

C0ppy, 2016-11-05:
Okay thanks

mikakaraila, 2016-11-05:
This line had wrong index, do you have time to modify & test?
102-opcuaclient.js:
if (msg.topic.substring(5,6)=='s')

C0ppy, 2016-11-05:
yes it works. I have made a pull request. (I'm very new to Github) 
I've have written down an idea to make this line more "generalized"

closed 2016-11-06 by mikakaraila