mikakaraila/node-red-contrib-opcua #20: Problem with reading and writing

C0ppy, 2016-11-14:
I have some problems with reading and writing values. Subscribing is working fine.
With KEPServerEX V5.11 I am getting these errors...
[image]
[image]
When I try to read and write a value I am getting an error. When I retry this after some secondes it works.
On the Prosys Testserver I am getting other errors...
[image]
[image]

I have created an "opcuaitem" by myself with reading a Jason file.

Subscribing works fine with the same processdata as reading on both servers. They are both running on the same PC as the node-red server.

mikakaraila, 2016-11-14:
Ack, I will try to investigate more. Have to install Prosys OPC UA server.
This seems to be something related to session and item address format.
I have one server that supports only ns=2;i=1004 numeric address format
when writing items.

C0ppy, 2016-11-14:
Okay thanks.

Some further Information...

I have created the "opcuaitems" by myself with a json string from a file. But with the opcua item node i have the same problem.

When I constantly write and read with the "client-read" and "client-Write" node to any address for every second, i can read/write to any address how often I want without any problem. Maybe it have something to do with an "keep-alive" or something like this.

mikakaraila, 2016-11-16:
I got opc simulation server and I got error only on deploy flow.
No other errors, read, subscription & write are working.
So I expect there is something in the flow or parameters how you are using
nodes.
Do you have simple example that I could look & use for testing?

mikakaraila, 2016-11-16:
Now I got the same error, I suppose other sessions are not using keepalive.
So if there is no write/read in 10s session is closed.
Subscription has keepalive so it keeps on running.
I have to look & validate this...

C0ppy, 2016-11-16:
Okay thanks.

mikakaraila, 2016-11-20:
Just added keepalive and session check for read&write. Please test I will
then publish new version.

C0ppy, 2016-11-21:
Thanks a lot. It`s working now without any errors.

mikakaraila, 2016-11-21:
v0.2.11 published into npm.

closed 2016-11-21 by mikakaraila