mikakaraila/node-red-contrib-opcua #21: Cannot locate certificate file

dharmeshpurohit, 2016-12-09:
Hi ,

I am trying to create OPC UA Browser using node-red and I am following example https://www.youtube.com/watch?v=LaUmhhMdoyY

I am trying to connect to PLC system and Allen-Bradley ControlLogix Ethernet - ControlLogix 5500 using

opc.tcp://MyIPAddress:44818

When I try to view the UI Page what I have developed I am getting following error and I have no idea what does it mean.

Error: cannot locate certificate file C:\Users\<USER>\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua\certificates\client_selfsigned_cert_1024.pem

Look forward for your reply.

Dharmesh

mikakaraila, 2016-12-10:
Have you checked that node-opcua installation created that directory and it
contains those files?
Look certificates directory!

Br,
Mika

mikakaraila, 2016-12-10:
I just notice that there is this issue in the node-opcua open:
https://github.com/node-opcua/node-opcua/issues/235

dharmeshpurohit, 2016-12-12:
Hi ,

In the installation directory the is not certificates folder being created.

Regards,

Dharmesh

dharmeshpurohit, 2016-12-12:
Hi tried to run 
node crypto_create_CA.js

but I am getting following error how ever I have installed  openssl 

set OPENSSL_CONF =C:/Users/<user>/AppData/Roaming/npm/node_modules/node-red-contrib-opcua/node_modules/node-opcua/certificates/PKI/CA/conf/caconfig.cnf
checking presence of  "C:\Users\<user>\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua\bin\openssl\openssl.exe"
openssl seems to be missing and need to be installed
 process.env.PROCESSOR_ARCHITEW6432  = AMD64
downloading http://indy.fulgan.com/SSL/openssl-1.0.2e-x64_86-win64.zip
Server responded with unhandled status: 400

Thanks 
Dharmesh

biancode, 2016-12-12:
May there is a space in your Windows path to the node package. That doesn't work. Like C:/users/my name/....

mikakaraila, 2016-12-13:
> set OPENSSL_CONF =C:/Users//AppData/Roaming/npm/node_modules/node-red-

Hmm, it should be run already in as node-opcua install runs postinstall and
it creates all those CA, PKI and other folders under node-opcua. Please
check first that under node-opcua is "certificates" directory created.

dharmeshpurohit, 2016-12-13:
Under node-opcua  "certificates" directory is not getting created, though I can see PKI folder and under that I have "untrusted" folder that has 6b5f82dc738bb783defb466d533813c6d6bd4975.pem file.

erossignon, 2016-12-13:
Default certificates are created during the installation of the "node-opcua" package.
For some reasons, this has failed during the installation on your machine when the installer tried to download the "openssl" binaries.
You may have a issue with your local proxy, or  the dowload server might have been unavailable.
Please check that you can download and access "http://indy.fulgan.com/SSL/openssl-1.0.2j-x64_86-win64.zip"

dharmeshpurohit, 2016-12-14:
> openssl-1.0.2j-x64_86-win64.zip"

Yes I am able to download the zip file from this location. But I don't know
how use it after downloading.

mikakaraila, 2017-01-31:
Ping, any news / updates?

WilliansDutra, 2017-07-28:
Hi, in my case I have the folder certificate but not the file client_selfsigned_cert_1024.pem. How to proceed?

RedCross0815, 2019-02-14:
Iam getting the same error, but in addition to this, i get the error "cant read/.../cer1024.pem"
But i can find and open the file...!
Any Ideas?

mikakaraila, 2019-02-14:
Did you use the latest version? Add some debug "node-red -v FLOW.json"
How / in which folder you start node-red? Because there is relative path to find certificates.
Like ../../../node-opcua-client/certificates/ etc.

kaushalkumarjoshi, 2019-02-20:
Dear @mikakaraila ,

Hi,

I installed node-red-contrib-opcua and tried to create an opc ua server.
After getting the certificate error, I soon realized that there is an issue related to path of the certificate.
The path has additional 'node_modules' folder which is not there.

[image]
[image]
[image]
[image]

Kindly resolve the same as it has to be done manually!

Best Regards
Kaushal

mikakaraila, 2019-02-20:
Yes I know the reason is related to client cetificate file handling. I did not yet fixed server / browse.
I try to find some later today at home.

mikakaraila, 2019-02-20:
Ok, works in my environment, please test.

kaushalkumarjoshi, 2019-02-21:
Now seems to take 2 .node-red in the path:

"Error: Certificate file must exist :C:\Users\Z003Z5MH\.node-red**\.node-red**\node_modules\node-opcua-server\certificates\server_selfsigned_cert_2048.pem"

Still an issue :(

mikakaraila, 2019-02-21:
Ok, it seems to be some difference in my dev. environment vs. installation environment.
I need following information:
1) How node-red-contrib-opcua is installed
2) What is installation folder (absolute path)
3) How you run flow e.q. where you start node-red and how flow JSON is given

Just to find out reason for double .node-red / node_modules. 
Of course there is relative path now in code.

mikakaraila, 2019-02-21:
I made server certificate finding with get-installed-path, please test if this will solve the problem.

kaushalkumarjoshi, 2019-02-22:
Works as expected!!
Cheers :) 👍

kaushalkumarjoshi, 2019-02-22:
Have you worked on opc ua timestamps??

mikakaraila, 2019-02-22:
Yes and arrays, 2-D tables, alarms & conditions etc.  even OPC UA Historical data access.
I have implemented our OPC UA Server and OPC UA Client with all features.

kaushalkumarjoshi, 2019-02-22:
Can you tell me the name of this product? :)

mikakaraila, 2019-02-22:
It is Valmet DNA (DCS system) and Valmet DNA-OPCUA-Server and Valmet-DNA-OPCUA-Client for
 Linux and Windows. Realtime communication and other high availability.

kaushalkumarjoshi, 2019-02-23:
I am looking for some option where the opc ua client will send the timestamped data and the opc ua server will use the timestamp from data source when publishing data to other clients.
Is it possible to do with these nodes?

mikakaraila, 2019-02-25:
I will close this one, new own issue for timestamp handling.

closed 2019-02-25 by mikakaraila

smalhao, 2019-09-16:
> Dear @mikakaraila ,
>
> Hi,
>
> I installed node-red-contrib-opcua and tried to create an opc ua server.
> After getting the certificate error, I soon realized that there is an issue related to path of the [...]
Can you share your code for me for seeing who you configure the server?
Best regards

chrisgray13247, 2020-01-03:
I am experiencing the same certificate issue with OPC UA Server in Docker instance of node-red with OPC UA contribution.

I have searched the node-red session but there is no certificates to be found.  Error I get in node-red is: 

Error: Certificate file must exist :/data/node_modules/node-opcua-server/certificates/server_selfsigned_cert_2048.pem

I also looked in the following directories:
/data/node_modules/node-opcua-server/certificates
/data/node_modules/node-opcua-server-discovery/certificates
/data/node_modules/node-opcua-client/certificates

How can I get a certificate file to copy to this folder?

mikakaraila, 2020-01-03:
There has been error already in installation, try to solve it first:
2020-01-03T02:07:37.148Z [err] CWD = .
2020-01-03T02:07:37.149Z [err] Command failed: "openssl" version
2020-01-03T02:07:37.149Z [err] /bin/sh: openssl: not found

So node-opcua installation creates self signed certificates and it has failed because script has not found openssl command from your environment.

chrisgray13247, 2020-01-03:
Thank Mikakaraila,

Yes, it would appear that the openssl is not installed within the Docker container (however it is installed on the host).  Foregive me for asking, but do you know how to install the Openssl within a Docker container (or where I would find the app to install)?

mikakaraila, 2020-01-07:
Did you try something like this?
RUN apk upgrade --update-cache --available && \
    apk add openssl && \
    rm -rf /var/cache/apk/*

jhottell, 2020-01-07:
Here are the two lines added to my docker file to install openssl:

```
USER root
RUN apk add --update openssl && rm -rf /var/cache/apk/*
```

schimao, 2023-10-18:
Hi,
I use Docker containers, where are the certificates located and what do I need to specify here?

[image]

mikakaraila, 2023-10-19:
Code generates them automatically, you can see path at console output.