mikakaraila/node-red-contrib-opcua #22: Problems reading from OPC server it gets "Connection erro" after 60 seconds.

Anknet, 2017-01-02:
Hey!
Thanks for a great plugin.
I have problems reading from OPC server it gets "Connection erro" after 60 seconds. Writing is no trouble.

in debug
-------------------------------------------------- -------
2017-01-02 16: 31: 5560db8389.74461c
msg: string [51]
OPC UA Connection Error: Transaction has timed out
-------------------------------------------------- ----------

 Timeout waiting for .... Response {/ * RequestHeader * /
 authenticationToken / NodeId * * /: ns = 0; i = 0
 timestamp / * UtcTime * /: 2017-01-0
2T21: 02: 10.588Z
 request handle / * IntegerId * /: 1
       0x1
 returned diagnostics / * uint32 * /: 0
       0x0
 auditEntryId / * UASTRING * /: null
 timeout hint / * uint32 * /: 0
       0x0
 additionalHeader / * Extension Object * /: null
};
January 2 22:03:10 - [error] [OpcUa-Client: 60db8389.74461c] OPC UA Connection Error
: Transaction has timed out
warning: Disconnection: pending closing session
-------------------------------------------------- -------------------------

Change something and press Deploy none so work it in 60 seconds.

It does not matter if I have the update interval of 1s 10s or 30s.
I set the on 70s that is, it has never asked so will the same result.

This now came up after update 0.2.8 0.2.11 ran before, and then worked good reading.

I also have trouble if you restart the OPC server, you will lose contact with the server and does not reconnect again when the server is back up.

Then I have to restart the "node-red".

I would like to have a function that once every 30 seconds a trustee does not reconnect again. From skratch.

I run "Node-red" 0.15.1 on a viritulel Hyper-V Server win2012r2
I run  Bechoff TwinCAT PLC 2.11 Beckhoff OPC UA Server v1.4 

Have also tried to run another PLC Beckhoff opcserver with the same result.
 
Do you have any idea I was doing wrong.
Sincerely Erik

mikakaraila, 2017-01-03:
Hmm,

do you inject read at start or just wait 60sec? Can you provide example
flow?

Br,
Mika

Anknet, 2017-01-03:
[image]

[{"id":"d17a994a.ba7cb8","type":"inject","z":"2cda9d07.7ee782","name":"Subscribe (1s)","topic":"","payload":"","payloadType":"str","repeat":"1","crontab":"","once":true,"x":201,"y":141.5,"wires":[["c582aed.b8d4e5"]]},{"id":"54f1d9bd.6aa998","type":"debug","z":"2cda9d07.7ee782","name":"","active":true,"console":"false","complete":"true","x":1213,"y":144,"wires":[]},{"id":"c582aed.b8d4e5","type":"OpcUa-Item","z":"2cda9d07.7ee782","item":"ns=4;s=Mqtt.mInUpptid","datatype":"String","value":"","name":"","x":547,"y":140,"wires":[["60db8389.74461c"]]},{"id":"60db8389.74461c","type":"OpcUa-Client","z":"2cda9d07.7ee782","endpoint":"32de5554.c50e0a","action":"read","time":"1","timeUnit":"s","name":"","x":887,"y":142,"wires":[["54f1d9bd.6aa998"]]},{"id":"32de5554.c50e0a","type":"OpcUa-Endpoint","z":"","endpoint":"opc.tcp://192.168.210.16:4840","login":false}]

Anknet, 2017-01-03:
[image]

mikakaraila, 2017-01-04:
I cannot reproduce the problem. One case can be that embedded OPC UA server
runs out of resources as the flow injects new read once a second.
More efficient way to do it is to inject once at start and then use
SUBSCRIPTION on 'OPC UA Client' node with 1s subscription.

Or you have multiple tabs for flows and one OPC UA Client is with the wrong
endpoint. As I look your examples client unique flow node-id is not the
same with the flow and error message you have given.

Anknet, 2017-01-05:
Hello again!
Thanks for your reply
I have reinstalled everything on a raspberry pi instead
The problem was there anyway! stop working after 60 sec
But I think I see the problem "Inject once at startup" do I remove the check mark, so reserve it as they work.

But I can not reconnect to the OPC server after they have been a crime
the. in komunikationen.

It would be great if it could make the reconnection from the beginning in any way at regular intervals.

Sincerely Erik

mikakaraila, 2017-01-11:
Can you test with this one:
[{"id":"5f8ed109.9e4d8","type":"inject","z":"77b6f9da.f4aed8","name":"Subscribe","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":true,"x":110.20002746582031,"y":78.5,"wires":[["e9e3fd91.2e46c"]]},{"id":"674af0a8.f79b7","type":"debug","z":"77b6f9da.f4aed8","name":"","active":true,"console":"false","complete":"true","x":667.2000923156738,"y":78.49999809265137,"wires":[]},{"id":"e9e3fd91.2e46c","type":"OpcUa-Item","z":"77b6f9da.f4aed8","item":"ns=4;s=Mqtt.mInUpptid","datatype":"String","value":"","name":"","x":316.2000274658203,"y":79,"wires":[["c170e87d.056348"]]},{"id":"c170e87d.056348","type":"OpcUa-Client","z":"77b6f9da.f4aed8","endpoint":"5abacb6e.640414","action":"subscribe","time":"1","timeUnit":"s","name":"","x":507.2000274658203,"y":79,"wires":[["674af0a8.f79b7"]]},{"id":"5abacb6e.640414","type":"OpcUa-Endpoint","z":"","endpoint":"opc.tcp://
192.168.210.16:4840","login":false}]

mikakaraila, 2017-01-11:
I can run both READ & SUBSCRIPTION based on my servers without any problem.
So I suspect that your server has some problem with String or with READ
every second. Have you tested flow above with SUBSCRITION or READ every 5th
second?

mikakaraila, 2017-01-31:
Ping, any updates?

closed 2017-02-02 by mikakaraila