mikakaraila/node-red-contrib-opcua #23: Not connecting!

tnq13, 2017-01-13:
I'm trying to get the data from an Siemens WinCC OPC Server into InfluxDB using node-red, but it's not working... seems that the OPC Client is not working properly...

When I watched the videos, there's always an icon showing that the OPC is connected, it's not appearing in my case.

I'm sure the OPC UA Server is workin, as you could see in the Unified Automation UAExpert Client.

[image]

[image]

[image]

Do you have any clue?

biancode, 2017-01-14:
please try ns=4;i=%AI_TS-.....  or ns=4;s=$%AI_TS-.....

i means Identifier
s means BrowseName

biancode, 2017-01-14:
Is the endpoint string the same as in UaExpert for the client in node-red?

tnq13, 2017-01-14:
Sorry... I couldn't understand what you asked... is the connection info?
If yes, see below the connection settings for UA Client

[image]

As for the node/tag configurations, none worked!
It seems that the Client is not working properly... Is there anyway to test the connection between the OPC client and OPC server??

tnq13, 2017-01-14:
I've just tried to use an public OPC Server (opc.tcp://opcua.demo-this.com:51210/UA/SampleServer)

Tried the node: ns=2;i=10847 and ns=2;s=NumberValeu (the same node, using adress and name)

Result: just the same. This confirm, for me, that the problem must to be in the Client (maybe I've configured wrongly?)

mikakaraila, 2017-01-14:
Item node is missing from the flow. I think inject does not contain type
information. it is the reason.

biancode, 2017-01-14:
see https://github.com/mikakaraila/node-red-contrib-opcua/tree/master/examples

tnq13, 2017-01-14:
**Mikakaraila**, I can't understand what you said...

This is the code I'm using:

`[
    {
        "id": "6be9fb3e.c2cc34",
        "type": "inject",
        "z": "3edf4287.9548be",
        "name": "",
        "topic": "ns=4;s=%AI_TS-FT001_OUT_EU",
        "payload": "",
        "payloadType": "str",
        "repeat": "",
        "crontab": "",
        "once": true,
        "x": 174.4444580078125,
        "y": 185.5555763244629,
        "wires": [
            [
                "e9060b7e.3e63a8"
            ]
        ]
    },
    {
        "id": "74382d31.c72a34",
        "type": "debug",
        "z": "3edf4287.9548be",
        "name": "",
        "active": true,
        "console": "true",
        "complete": "payload",
        "x": 757.7777777777777,
        "y": 193.33333333333331,
        "wires": []
    },
    {
        "id": "e9060b7e.3e63a8",
        "type": "OpcUa-Client",
        "z": "3edf4287.9548be",
        "endpoint": "6e50f038.2d317",
        "action": "subscribe",
        "time": "1",
        "timeUnit": "s",
        "name": "",
        "x": 477.7777862548828,
        "y": 190.00001335144043,
        "wires": [
            [
                "74382d31.c72a34"
            ]
        ]
    },
    {
        "id": "6e50f038.2d317",
        "type": "OpcUa-Endpoint",
        "z": "",
        "endpoint": "ocp.tcp://192.168.9.44:4870",
        "login": false
    }
]`

**biancod**

I've tried some of the examples (ExampleFlow.json, IGSS_OPCUA.json, OPCUA_AE_TEST.json) without success: when I inject, there is no output.

mikakaraila, 2017-01-15:
This is the correct way to do it, please check the type!

[{"id":"2aacb21f.d2e95e","type":"inject","z":"74b8c033.c7e52","name":"","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":true,"x":122.19999694824219,"y":110.19999694824219,"wires":[["444b737b.14bbbc"]]},{"id":"566747c1.cb3908","type":"debug","z":"74b8c033.c7e52","name":"","active":true,"console":"true","complete":"payload","x":707.5333404541016,"y":113.97775268554688,"wires":[]},{"id":"dfe63d6b.8a5e2","type":"OpcUa-Client","z":"74b8c033.c7e52","endpoint":"b1675416.4955c8","action":"subscribe","time":"1","timeUnit":"s","name":"","x":515.5333251953125,"y":114.64443397521973,"wires":[["566747c1.cb3908"]]},{"id":"444b737b.14bbbc","type":"OpcUa-Item","z":"74b8c033.c7e52","item":"ns=4;s=%AI_TS-FT001_OUT_EU","datatype":"Float","value":"","name":"","x":312.50001525878906,"y":112.59999084472656,"wires":[["dfe63d6b.8a5e2"]]},{"id":"b1675416.4955c8","type":"OpcUa-Endpoint","z":"","endpoint":"ocp.tcp://
192.168.9.44:4870","login":false}]

tnq13, 2017-01-16:
*Mikakaraila*

No deal! Not working... just copied what you sent and nothing!
I manage to get data from the OPC Server to a InfluxDB using [Node-OPCUA](http://node-opcua.github.io/) in a Ubuntu Distro.

A new info... maybe this is the secret:

[image]

tnq13, 2017-01-16:
It was indeed the certificate. Working!!

mikakaraila, 2017-01-16:
Good,
I have had problems with certificates, but first look I did not notice that
you used certificate on client, but anyway good that you find it.

Mfg,
Meta

mikakaraila, 2017-01-16:
Closed, certificate missing was reason for not accepting connection => timeout.

closed 2017-01-16 by mikakaraila