mikakaraila/node-red-contrib-opcua #24: Cannot restart node-red after critical OPC server error.

AndreGuerra123, 2017-01-31:
Hello, 
I implemented an OPC item flow to an OPC Server and I got the following error in console: 
[image]
I cannot even restart node-red now.
Do you know the cause?
Seems that the server is still open but I checked netstat in cmd and I didn't find any 53880 port opened.
Best regards,
André

mikakaraila, 2017-01-31:
Do you have more than one OPC UA server node in the flow? If yes they should have own port.
Or did you modify & deploy flow again. It can be some problem in deployment....
Try to reproduce error and attach example / sequency then I can look this more detailed.

AndreGuerra123, 2017-02-01:
Yes, you cannot modify & deploy flow while running. That caused the error.  Just killed the node process tree and restarted. Ty

mikakaraila, 2017-02-01:
Can you provide flow? Simple one to repeat the problem. I tested but I did
not find out yet to repeat this one.

closed 2017-02-02 by mikakaraila