mikakaraila/node-red-contrib-opcua #25: Connecting inject, function (msg.payload) to OPC Item to OPC Server.

AndreGuerra123, 2017-01-31:
Could not find an example of a OPC Item connect to a Server and to an inject simultaneously?
The data seems not be passing along the flow. How to connect an inject to a Server  without the translation function provided in the videos?
Regards
PS: Which software is used on those videos as OPC Client?

mikakaraila, 2017-01-31:
Which video you mean? I have been using UA Expert to browse and check OPC
UA Server address space.
Can you attach your flow? It will be easier to understand what you try to
do...

Br,
Mika

AndreGuerra123, 2017-02-01:
Hello, Mika.
So, I'm building a process model with ODEs which simulates a process. I have the obtained variables being produced asynchronously, and I need them to be deployed in an OPC UA Server. I have a function which gives me a payload with respective variables. To my understanding, I have to connect this function to an OPC Item before I attach to an OPC Server.
I was talking about this video: https://www.youtube.com/watch?v=J2pfXmsxYvw  , but in this case, it is an inject attached directly to the OPC Server.
Best regards,
André

mikakaraila, 2017-02-01:
> v=J2pfXmsxYvw , but in this case, it is an inject attached directly to

I tested with the flow: OPCUAServer_Flow.json from the examples folder.
Server creates variables according injected opcuaCommands that are injected.

mikakaraila, 2017-02-01:
If I understand you want to add new variables into the OPC UA Server
address space by defining them in inject.
For example:
topic = "ns=4;s=MyNewVariable;datatype=Int32"
I have to add some code then you can do it, but you need to use item node
and client node to read / write values.

mikakaraila, 2017-02-01:
OK I implemented it for Int32 type as first proto-type. What are the basic
types you want to use?

mikakaraila, 2017-02-01:
[image: Inline image 1]

mikakaraila, 2017-02-01:
Here is UA Expert view:
[image: Inline image 1]

mikakaraila, 2017-02-02:
I added most of bacic types: UInt16, Uint32, Int16, Int32, String, Float, Double, Boolean.
OPCUA-TEST.json flow in the folder examples contains practical examples.

closed 2017-02-02 by mikakaraila

AndreGuerra123, 2017-02-02:
Sorry to raise this issue again, but...what I need to understand is that some authors are calling items/params and groups inside an OPC UA Server.  (See Communication Solution for Industrial Control Applications with Multi-agents Using OPC Servers)
So, from my understanding, the Items should be inside the Server. Correct me if I'm wrong, the client should retrieve those items...
Another issue is how to inject Alarms and Events into an OPC Server in node-red and differentiated?
Ty for your time, 
André

mikakaraila, 2017-02-02:
In the github is new OPCUA-TEST.json flow file in the examples. And if you
look 104-opcuaserver.js it contains code that will create folders and
variables into the server.
Alarms and events are longer story. I have implemented our OPC UA server
with full address space and alarms & events support.
In case of A&E you have to prepare address space with needed A&E class and
instantiate A&E object plus raise alarm.
It is very complex... you can take a look into specifications (more than
hundred pages).
Br,
Mika

AndreGuerra123, 2017-02-06:
Hello, Mika. 
Sorry for reopening this issue twice. I've been testing your OPCUA-TEST.json flow recently. So I got a couple of issues I hope you may help me with.

1) When I import your example and deploy I instantaneouly receive this error: " OPC UA connection error: Cannot read property 'createSession' of null" but if I wait a bit then OPC UA Server seems to be running and both deployed OPC clients  have an active session.

2)When I inject New Variable (Int32) the direct debug states 2 messages: 

2/6/2017, 2:56:26 PMnode: f05cf121.f9875
msg : string[22]
"unknown OPC UA Command"
2/6/2017, 2:56:26 PMnode: 4cb5e2f5.2bcdfc
ns=4;s=TestInt32;datatype=Int32 : msg : Object
{ _msgid: "5e023d36.6fce94", topic: "ns=4;s=TestInt32;datatype=Int3…", payload: object }

and  the Server debug states:

2/6/2017, 2:57:17 PMnode: a632cab2.947458
ns=4;s=TestInt32;datatype=Int32 : msg.payload : Object
{ opcuaCommand: "addVariable" }

 which seems correct. So, far to my understanding node-red just added a new variable to the address space of the respective Server endpoint.

3)Now let's say that I want to Inject a variable value (write) to this recently added variable. I go to the last inject of the example and inject 13 as payload. (Shift the wire to top Client) When I do this the following error just comes up in Client debug (Client status shifts to error): 

" Cannot write value (13) to msg.topic:ns=4;s=TestInt32;datatype=Int32 error:Error: _socket has been disconnected by third party
 
If I switch the wire to the last inject to the first Item connected to the first Client I can successfully inject.
So, I assume is the correct way of doing it.

4) Now it's time to read, with the Second Client, when I press the in the read Inject i got the following error:
2/6/2017, 3:09:26 PMnode: d4a8272b.da89a8
msg : string[59]
"→Bad read: [1m[31mBadNodeIdUnknown (0x80340000)[39m[22m"
2/6/2017, 3:09:26 PMnode: d4a8272b.da89a8
msg : string[40]
"→Message:ns=4;s=TestInt32 dataType:Int32"
2/6/2017, 3:09:26 PMnode: d4a8272b.da89a8
msg : string[205]
"→Data:{"statusCode":{"value":2150891520,"description":"The node id refers to a node that does not exist in the server address space.","name":"BadNodeIdUnknown"},"sourcePicoseconds":0,"serverPicoseconds":0}"

Hope I am not doing something really stupid but well I attach a video, with my probably wrong doings....
Best regards, André

mikakaraila, 2017-02-06:
>    1. Now it's time to read, with the Second Client, when I press the in

Split server to own tab and clients to another. It will resolve some of the
issues above.
I will look later today those other ones...

AndreGuerra123, 2017-02-06:
https://www.useloom.com/share/ad1e3aa0ec7f11e689c9c95ad27ead9a
By the way, I dropped an example video...that may help you to understand my flow issues. 
Best regards and thank you.

mikakaraila, 2017-02-06:
Ack, found it.

mikakaraila, 2017-02-06:
These should be now fixed. I tested with new OPCUA-TEST.json (separated
server & client to own flows).