mikakaraila/node-red-contrib-opcua #26: OPC UA Client gets stuck on Active Reading

ajgbarnes, 2017-02-01:
Maybe this is a user error and quite possibly as I'm new to these libraries and nodes.

I'm testing against the Prosys OPC UA Simulation Server right now too.

1. I have a simple flow which has

Inject -> Opc-ua-Client -> Template -> Debug 

and an equivalent using the Opc-ua-item node:

Inject -> Opc-ua-item Opc-ua-node -> Template -> Debug

2. The actual test Node-RED flow is as below:

[{"id":"a3631fd0.4642b","type":"inject","z":"a73edfd3.234468","name":"","topic":"ns=2;s=MyEnumObject","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":189,"y":199,"wires":[["679e58b9.b0ad2"]]},{"id":"679e58b9.b0ad2","type":"OpcUa-Client","z":"a73edfd3.234468","endpoint":"4c718b59.85a844","action":"read","time":"","name":"Test server (read items)","x":532,"y":195.75,"wires":[["a9c05ba8.7b0cf8"]]},{"id":"a9c05ba8.7b0cf8","type":"template","z":"a73edfd3.234468","name":"","field":"payload","fieldType":"msg","format":"handlebars","syntax":"mustache","template":"This is the payload","x":649,"y":279.5,"wires":[["b25e574c.3ad488"]]},{"id":"603a079a.1725c8","type":"inject","z":"a73edfd3.234468","name":"Read","topic":"ns=2;s=MyEnumObject","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":151,"y":97.75,"wires":[["e0785202.82a4a8"]]},{"id":"e0785202.82a4a8","type":"OpcUa-Item","z":"a73edfd3.234468","item":"ns=2;s=MyEnumObject","datatype":"Int32","value":"","name":"FreeMemoryItem","x":323,"y":104.75,"wires":[["679e58b9.b0ad2"]]},{"id":"b25e574c.3ad488","type":"debug","z":"a73edfd3.234468","name":"Read value","active":true,"console":"false","complete":"true","x":681,"y":347.75,"wires":[]},{"id":"4c718b59.85a844","type":"OpcUa-Endpoint","z":"","endpoint":"opc.tcp://localhost:53530/OPCUA/SimulationServer","login":false}]

3. The flow that uses which uses the Topic field of the inject node always returns an error along with the read.  However the Info documentation for the Opc-ua-client states that you only have to set the Topic field of the inject node to be the NodeId e.g. ns=<int>;i=<int> or ns=<int>;s=<string> however using an Opc-ua-item ad setting the Item to the NodeId performs a clean read.  

4. Additionally, when using the Inject node / Topic combination, the flow never progresses beyond the Opc-ua-client, it gets stuck (or bails out) on that node.  And just leaves the state in Active Reading which still works for subsequent injects but I'd expect to see an output flow on error?

So I guess two possible issues (I may have misunderstood the expected behaviour):

1. I cannot see a way to dynamically query in Node-RED using these nodes, as Opc-ua-item hard codes the NodeId into the node.  No way of dynamically added a NodeId using e.g. msg.topic without hitting the errors above and not getting any output from the Opc-ua-client?

2.  Why does the opc-ua-client hang on an error and not provide an error output?

Any help/thoughts/comments are appreciated,

Many thanks, Andy

mikakaraila, 2017-02-01:
>    1. The actual test Node-RED flow is as below:
>    the inject node to be the NodeId e.g. ns=;i= or ns=;s= however using an

Oh, I think it missed type. Fix will need type in the topic. I will add
needed code and new example...

ajgbarnes, 2017-02-01:
Thanks for responding so quickly - what would the syntax be?  Separated using a semi-colon?  e.g. ns=1;i=12345;datatype=Double ?

mikakaraila, 2017-02-01:
Good I can implement that later today

ajgbarnes, 2017-02-01:
That would be an incredible turnaround - and is most appreciated, thank you.

mikakaraila, 2017-02-01:
Good this is now fixed.

ajgbarnes, 2017-02-01:
Thank you - will test in the morning :)

mikakaraila, 2017-02-01:
I did not yet push it back to git...

ajgbarnes, 2017-02-02:
I know it's been less than 24 hours since I raised this, and just wondered if you'd made any progress with this?

mikakaraila, 2017-02-02:
OK test flow works now with addVariable, addFolder, setFolder. I will push
it to github for testing.

closed 2017-02-02 by mikakaraila