mikakaraila/node-red-contrib-opcua #27: parsing data error in 0.2.11 ?

jmikerq, 2017-02-01:
Hello, I've just encountered an error in node-red-contrib-opcua@0.2.11 that the OPC node tries to parse some data from kepserver and return error in reading the values. The same flow works fine when I switch to use node-red-contrib-opcua@0.2.7

I think the error was thrown at line 362 in opcuaclient.js

so here is the DataValue object I got when I add `console.log(dataValue)` after line 335:

```
DataValue {
  value:
   Variant {
     '$dataType': { [Number: 6] key: 'Int32', value: 6 },
     '$arrayType': { [Number: 0] key: 'Scalar', value: 0 },
     value: 55,
     dimensions: null },
  statusCode: StatusCode { value: 0, description: 'No Error', name: 'Good' },
  sourceTimestamp: null,
  sourcePicoseconds: 0,
  serverTimestamp: null,
  serverPicoseconds: 0 
}
```

and here is the error message it throw in cmd:
 
>1 Feb 13:25:40 - [error] [OpcUa-Client:OPC Read]        Bad read: Good (0x00000)
>1 Feb 13:25:40 - [error] [OpcUa-Client:OPC Read]        Message:ns=2;s=Painting.Sprayer.Sprayer1 dataType:undefined
>1 Feb 13:25:40 - [error] [OpcUa-Client:OPC Read]        Data:{"value":{"dataType":"Int32","arrayType":"Scalar","value":55},"statusCode":{"value":0,"description":"No Error","name":"Good"},"sourcePicoseconds":0,"serverPicoseconds":0}

mikakaraila, 2017-02-02:
There has been read error if you are adding console logging to line 335.
OR another case can be that datatype is null or empty. Are you using inject
or item node in your flow? With item node check that you have selected type.
I just added yesterday datatype into inject so it cannot work yet.

mikakaraila, 2017-02-02:
Please test.

jmikerq, 2017-02-02:
The console logging that I added was just to verify I am receiving a non-null dataValue object. I was receiving this error before I added in console logging lines.

I used an inject node to trigger action of the opcua node.

So I further debug this code, and I think the problem was from this block 

```
if (msg.datatype.localeCompare(dataValue.value.dataType.toString())!=0) {
    node.error("\tMessage types are not matching: " + msg.topic + " types: " + msg.datatype + " <> " + dataValue.value.dataType.toString());
}
```
The `msg.datatype` object in this case is undefined.

mikakaraila, 2017-02-02:
Thanks. Can send that psrt of the flow?
I can now prevent exception but I would like to add meaningful error
message...

2.2.2017 8.50 ip.

mikakaraila, 2017-02-02:
Code could be this way 354-359. Can you test?
if (msg.datatype!=null &&
msg.datatype.localeCompare(dataValue.value.dataType.toString())!=0) {
node.error("\tMessage types are not matching: " + msg.topic + " types: " +
msg.datatype + " <> " + dataValue.value.dataType.toString());
}
if (msg.datatype==null) {
node.warn("msg.datatype == null, if you use inject check topic is format
'ns=2;s=MyLevel;datatype=Float'");
}

jmikerq, 2017-02-02:
Now the flow doesn't return the bad read error. The flow is able to return some data, but also keep showing the warning string.

One of the flows I have is
```
[{"id":"fc3b8b24.c7d038","type":"inject","z":"268a2eb6.954ee2","name":"Poll Temperature","topic":"ns=2;s=Painting.Oven.Temperature","payload":"","payloadType":"date","repeat":"5","crontab":"","once":false,"x":167,"y":114,"wires":[["c9d30906.9ba398"]]},{"id":"c9d30906.9ba398","type":"OpcUa-Client","z":"268a2eb6.954ee2","endpoint":"376da0ee.763d8","action":"read","time":"5","timeUnit":"s","name":"OPC Read","x":354,"y":114,"wires":[["7ada865d.73b0c8"]]},{"id":"7ada865d.73b0c8","type":"switch","z":"268a2eb6.954ee2","name":"Filter","property":"payload","propertyType":"msg","rules":[{"t":"neq","v":"null","vt":"str"}],"checkall":"true","outputs":1,"x":503,"y":114,"wires":[["3f70cadf.e735c6","233384d4.ef854c"]]},{"id":"233384d4.ef854c","type":"mqtt out","z":"268a2eb6.954ee2","name":"Oven Temperature","topic":"sensetecnic/iiot/oven/temperature","qos":"","retain":"","broker":"ad677c90.c9374","x":699,"y":114,"wires":[]},{"id":"3f70cadf.e735c6","type":"debug","z":"268a2eb6.954ee2","name":"Temperature","active":true,"console":"false","complete":"payload","x":872,"y":92,"wires":[]},{"id":"376da0ee.763d8","type":"OpcUa-Endpoint","z":"268a2eb6.954ee2","endpoint":"opc.tcp://127.0.0.1:49320/KEPServerEX","login":false},{"id":"ad677c90.c9374","type":"mqtt-broker","z":"268a2eb6.954ee2","broker":"54.69.2.224","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"willTopic":"","willQos":"0","willPayload":"","birthTopic":"","birthQos":"0","birthPayload":""}]
```

As you can see from the flow, I didn't have the "datatype=Float" at the end of the topic. However, if I tried to add that in the inject node topic, I got the bad read error again. 
>  [error] [OpcUa-Client:OPC Read]        Bad read: BadNodeIdUnknown (0x80340000)

mikakaraila, 2017-02-03:
You need to add item node after inject, but there is something else that I have to check now.

mikakaraila, 2017-02-03:
Ups, this is corrected and working one. Put back kepserver address:
[{"id":"4f9f0995.eda168","type":"inject","z":"a2904723.4d3158","name":"Poll Temperature","topic":"ns=5;s=Sinusoid1;datatype=Double","payload":"","payloadType":"str","repeat":"5","crontab":"","once":false,"x":148.2000274658203,"y":151.1999969482422,"wires":[["1a17d582.1dfaba"]]},{"id":"d4202214.c377f","type":"switch","z":"a2904723.4d3158","name":"Filter","property":"payload","propertyType":"msg","rules":[{"t":"neq","v":"null","vt":"str"}],"checkall":"true","outputs":1,"x":488.2000274658203,"y":152.1999969482422,"wires":[["44a0f078.9e205","59dde44a.c4352c"]]},{"id":"59dde44a.c4352c","type":"mqtt out","z":"a2904723.4d3158","name":"Oven Temperature","topic":"sensetecnic/iiot/oven/temperature","qos":"","retain":"","broker":"fa122c31.6ec1b","x":684.2000274658203,"y":152.1999969482422,"wires":[]},{"id":"44a0f078.9e205","type":"debug","z":"a2904723.4d3158","name":"Temperature","active":false,"console":"false","complete":"payload","x":666.2000274658203,"y":104.19999694824219,"wires":[]},{"id":"1a17d582.1dfaba","type":"OpcUa-Client","z":"a2904723.4d3158","endpoint":"72203e80.075de","action":"read","time":"5","timeUnit":"s","name":"OPC Read","x":339.2000274658203,"y":152.1999969482422,"wires":[["d4202214.c377f","53b7ab06.5f9f44"]]},{"id":"53b7ab06.5f9f44","type":"debug","z":"a2904723.4d3158","name":"","active":true,"console":"false","complete":"false","x":520.4999847412109,"y":232.8000030517578,"wires":[]},{"id":"fa122c31.6ec1b","type":"mqtt-broker","z":"a2904723.4d3158","broker":"54.69.2.224","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"willTopic":"","willQos":"0","willPayload":"","birthTopic":"","birthQos":"0","birthPayload":""},{"id":"72203e80.075de","type":"OpcUa-Endpoint","z":"a2904723.4d3158","endpoint":"opc.tcp://localhost:53530/OPCUA/SimulationServer","login":false}]

mikakaraila, 2017-02-04:
Did you test flow above?

jmikerq, 2017-02-06:
Hello @mikakaraila, I just tried out with the flow you provided and the latest version you have on npm. Does the node suppose to work without the item node now?

with the flow you supplied and the latest version of your node on npm, I got the following message from cmd:

```
6 Feb 10:28:06 - [error] [OpcUa-Client:OPC Read]        Bad read: BadNodeIdUnknown (0x80340000)
6 Feb 10:28:06 - [error] [OpcUa-Client:OPC Read]        Message:ns=5;s=Painting.Oven.Temperature dataType:Double;ns=2;
6 Feb 10:28:06 - [error] [OpcUa-Client:OPC Read]        Data:{"statusCode":{"value":2150891520,"description":"The node id refers to a node that does not exist in the server address space.","name":"BadNodeIdUnknown"},"sourcePicoseconds":0,"serverPicoseconds":0}
```

mikakaraila, 2017-02-06:
The latest version is now 0.2.13 and this works:
[image]

mikakaraila, 2017-02-06:
Message seems to be wrong, can you attach your flow? If I have still parse error...
Message:ns=5;s=Painting.Oven.Temperature dataType:Double;ns=2;
Should be this way if you use topic:
ns=5;s=Painting.Oven.Temperature;datatype=Double

jmikerq, 2017-02-06:
I was using the flow you attached above to test, using the node version 0.2.13. I am currently using node-red@0.16.1 if that makes any different. 

```
[{"id":"6b0ffe8f.67ca9","type":"inject","z":"ad07d97e.e3a8d8","name":"Poll Temperature","topic":"ns=5;s=Painting.Oven.Temperature;datatype=Double;","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":172,"y":187,"wires":[["65d70fc.4e5c9f"]]},{"id":"a683e3bc.a80c5","type":"switch","z":"ad07d97e.e3a8d8","name":"Filter","property":"payload","propertyType":"msg","rules":[{"t":"neq","v":"null","vt":"str"}],"checkall":"true","outputs":1,"x":552,"y":182,"wires":[["97709778.790068","79be091.c4689f8"]]},{"id":"79be091.c4689f8","type":"mqtt out","z":"ad07d97e.e3a8d8","name":"Oven Temperature","topic":"sensetecnic/iiot/oven/temperature","qos":"","retain":"","broker":"f453a7d7.c95298","x":748,"y":182,"wires":[]},{"id":"97709778.790068","type":"debug","z":"ad07d97e.e3a8d8","name":"Temperature","active":true,"console":"false","complete":"payload","x":730,"y":134,"wires":[]},{"id":"65d70fc.4e5c9f","type":"OpcUa-Client","z":"ad07d97e.e3a8d8","endpoint":"a524dbea.285798","action":"read","time":"5","timeUnit":"s","name":"OPC Read","x":403,"y":182,"wires":[["a683e3bc.a80c5","36d5559a.667c8a"]]},{"id":"36d5559a.667c8a","type":"debug","z":"ad07d97e.e3a8d8","name":"","active":false,"console":"false","complete":"false","x":584.2999572753906,"y":262.6000061035156,"wires":[]},{"id":"e39bbe20.ed81c","type":"OpcUa-Item","z":"ad07d97e.e3a8d8","item":"ns=5;s=Painting.Oven.Temperature","datatype":"Double","value":"","name":"","x":256,"y":316,"wires":[["65d70fc.4e5c9f"]]},{"id":"f453a7d7.c95298","type":"mqtt-broker","z":"ad07d97e.e3a8d8","broker":"54.69.2.224","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"willTopic":"","willQos":"0","willPayload":"","birthTopic":"","birthQos":"0","birthPayload":""},{"id":"a524dbea.285798","type":"OpcUa-Endpoint","z":"ad07d97e.e3a8d8","endpoint":"opc.tcp://127.0.0.1:49320/KEPServerEX","login":false}]
```

You were right about the inject node topic, I removed the extra bit at the end. 

it still returns the following:

```
6 Feb 11:09:25 - [error] [OpcUa-Client:OPC Read]        Bad read: BadNodeIdUnknown (0x80340000)
6 Feb 11:09:25 - [error] [OpcUa-Client:OPC Read]        Message:ns=5;s=Painting.Oven.Temperature dataType:Double;
6 Feb 11:09:25 - [error] [OpcUa-Client:OPC Read]        Data:{"statusCode":{"value":2150891520,"description":"The node id refers to a node that does not exist in the server address space.","name":"BadNodeIdUnknown"},"sourcePicoseconds":0,"serverPicoseconds":0}
```

tried with the item node, still seeing the same thing.

mikakaraila, 2017-02-06:
Topic is still wrong in your flow:
ns=5;s=Sinusoid1;datatype=Double
[image]

jmikerq, 2017-02-06:
[image]

That is what I have

mikakaraila, 2017-02-06:
Attach flow here...

mikakaraila, 2017-02-06:
What is on Flow1 tab? You do not have repeat on. So are you sure debug
messages are coming from the Flow2 inject node?

jmikerq, 2017-02-06:
The flow is based on the one you provided:

```
[{"id":"ad07d97e.e3a8d8","type":"tab","label":"Flow 2"},{"id":"6b0ffe8f.67ca9","type":"inject","z":"ad07d97e.e3a8d8","name":"Poll Temperature","topic":"ns=5;s=Sinusoid1;datatype=Double","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":172,"y":187,"wires":[["65d70fc.4e5c9f"]]},{"id":"a683e3bc.a80c5","type":"switch","z":"ad07d97e.e3a8d8","name":"Filter","property":"payload","propertyType":"msg","rules":[{"t":"neq","v":"null","vt":"str"}],"checkall":"true","outputs":1,"x":552,"y":182,"wires":[["97709778.790068","79be091.c4689f8"]]},{"id":"79be091.c4689f8","type":"mqtt out","z":"ad07d97e.e3a8d8","name":"Oven Temperature","topic":"sensetecnic/iiot/oven/temperature","qos":"","retain":"","broker":"f453a7d7.c95298","x":748,"y":182,"wires":[]},{"id":"97709778.790068","type":"debug","z":"ad07d97e.e3a8d8","name":"Temperature","active":true,"console":"false","complete":"payload","x":730,"y":134,"wires":[]},{"id":"65d70fc.4e5c9f","type":"OpcUa-Client","z":"ad07d97e.e3a8d8","endpoint":"a524dbea.285798","action":"read","time":"5","timeUnit":"s","name":"OPC Read","x":403,"y":182,"wires":[["a683e3bc.a80c5","36d5559a.667c8a"]]},{"id":"36d5559a.667c8a","type":"debug","z":"ad07d97e.e3a8d8","name":"","active":false,"console":"false","complete":"false","x":584.2999572753906,"y":262.6000061035156,"wires":[]},{"id":"e39bbe20.ed81c","type":"OpcUa-Item","z":"ad07d97e.e3a8d8","item":"ns=5;s=Painting.Oven.Temperature","datatype":"Double","value":"","name":"","x":256,"y":316,"wires":[["65d70fc.4e5c9f"]]},{"id":"f453a7d7.c95298","type":"mqtt-broker","z":"ad07d97e.e3a8d8","broker":"54.69.2.224","port":"1883","clientid":"","usetls":false,"compatmode":true,"keepalive":"60","cleansession":true,"willTopic":"","willQos":"0","willPayload":"","birthTopic":"","birthQos":"0","birthPayload":""},{"id":"a524dbea.285798","type":"OpcUa-Endpoint","z":"ad07d97e.e3a8d8","endpoint":"opc.tcp://127.0.0.1:49320/KEPServerEX","login":false}]
```

For this flow, the response is 

```
6 Feb 12:17:47 - [error] [OpcUa-Client:OPC Read]        Bad read: BadNodeIdUnknown (0x80340000)
6 Feb 12:17:47 - [error] [OpcUa-Client:OPC Read]        Message:ns=5;s=Sinusoid1 dataType:Double
6 Feb 12:17:47 - [error] [OpcUa-Client:OPC Read]        Data:{"statusCode":{"value":2150891520,"description":"The node id refers to a node that does not exist in the server address space.","name":"BadNodeIdUnknown"},"sourcePicoseconds":0,"serverPicoseconds":0}
```

Flows in tab 1 were the original flows that worked with node version 0.2.7, and I purposely turn off the repeat options in all inject nodes for testing. I also set the debug tab to show the current flow only. So the responses I saw in the debug tab were only the ones from the flow that I attached above. 

I am seeing this error messages with and without using the item node in the flow.

mikakaraila, 2017-02-06:
I think it is Item node that has no input connected. Constructor gives
error message, but it will not be injected to client. Remove item node and
then it should work.
If you push inject then it will do read once.
[image: Inline image 1]

I added empty inject to get Item sent to client node this way:
[image: Inline image 2]

jmikerq, 2017-02-06:
I tried removing the item node. it still gives me the same response that:

```
6 Feb 13:24:03 - [error] [OpcUa-Client:OPC Read]        Bad read: BadNodeIdUnknown (0x80340000)
6 Feb 13:24:03 - [error] [OpcUa-Client:OPC Read]        Message:ns=5;s=Sinusoid1 dataType:Double
6 Feb 13:24:03 - [error] [OpcUa-Client:OPC Read]        Data:{"statusCode":{"value":2150891520,"description":"The node id refers to a node that does not exist in the server address space.","name":"BadNodeIdUnknown"},"sourcePicoseconds":0,"serverPicoseconds":0}
```

I tried playing with the item node. It sent some more meaningful msg to the OPC read node in my opinion, but still spits bad read response: BadNodeIdUnknown

```
6 Feb 13:21:35 - [error] [OpcUa-Client:OPC Read]        Bad read: BadNodeIdUnknown (0x80340000)
6 Feb 13:21:35 - [error] [OpcUa-Client:OPC Read]        Message:ns=5;s=Painting.Oven.Temperature; dataType:Double
6 Feb 13:21:35 - [error] [OpcUa-Client:OPC Read]        Data:{"statusCode":{"value":2150891520,"description":"The node id refers to a node that does not exist in the server address space.","name":"BadNodeIdUnknown"},"sourcePicoseconds":0,"serverPicoseconds":0}
```

```
6 Feb 13:23:00 - [error] [OpcUa-Client:OPC Read]        Bad read: BadNodeIdUnknown (0x80340000)
6 Feb 13:23:00 - [error] [OpcUa-Client:OPC Read]        Message:ns=5;s=Sinusoid1; dataType:Double
6 Feb 13:23:00 - [error] [OpcUa-Client:OPC Read]        Data:{"statusCode":{"value":2150891520,"description":"The node id refers to a node that does not exist in the server address space.","name":"BadNodeIdUnknown"},"sourcePicoseconds":0,"serverPicoseconds":0}
```

jmikerq, 2017-02-06:
I think the node works now. I think the problem was I used "ns=5" from the sample provided above. I switched to the ones I had before and it works now.

mikakaraila, 2017-02-07:
Heh,  correct error message as reason.
Your server just did not have that namespace... I was wondering what is
different.

7.2.2017 12.57 ap.

closed 2017-02-07 by mikakaraila