mikakaraila/node-red-contrib-opcua #28: How to add variable?

yonas124, 2017-02-02:
Hi Mika,

How can I add variable or node so that I can create my own OPC UA server?
Is the command not yet supported? Because I can only add equipment and physical asset at the moment.

mikakaraila, 2017-02-02:
I asked what types you will need?
I did not push new code to git yet...

yonas124, 2017-02-02:
Just a regular uint16 is enough

mikakaraila, 2017-02-02:
Another question: now you can create flat address space under Objects, I
think I should implement later addFolder & setFolder for more complex
address spaces.

yonas124, 2017-02-02:
addFolder sounds nice
What I want to do is actually to make a standalone opc ua server, consists of data from various sources.

biancode, 2017-02-02:
@yonas124 try https://www.npmjs.com/package/node-red-contrib-isa

mikakaraila, 2017-02-02:
Implemented, please test and enjoy.

closed 2017-02-02 by mikakaraila

yonas124, 2017-02-03:
super

chess126, 2017-07-13:
then  how can i add my own  node(var, folder, method etc) in namespace now, thx