mikakaraila/node-red-contrib-opcua #30: Node-Red crashes when injecting incorrect NodeID into OpcUA-Client node

mjunghanns, 2017-02-21:
Hello, I am experiencing a crash of Node-Red when trying subscribe to a miss-typed/incorrect NodeID. I'm injecting the namespace and NodeID using the injection node. If I use the correct values, everything is working as expected. 

Let's assume, the correct NodeID would be "TestVerzeichnis.TestKnoten". When we mess it up to "TestVerzeichnis.TestKnotn" we are receiving the following error:
`"subscription.monitorItem:Error: BadNodeIdUnknown (0x80340000)"`

So far, so good - when I inject again, Node-Red would crash with the following error (sorry for the poor format, editor doesn't wrap correctly...):

`node-red-0 21 Feb 12:18:26 - [warn] [OpcUa-Client:94a092f2.3a43b] OpcUaClientNode: Monitoring Event: ns=2;s=TestVerzeichnis.TestKnotn by interval of 0 ms
node-red-0 AssertionError: self.subscription.session
node-red-0     at ClientMonitoredItem._monitor (/home/root/.node-red/node_modules/node-opcua/lib/client/client_monitored_item.js:190:5)
node-red-0     at Immediate.wait_for_subscription_and_monitor (/home/root/.node-red/node_modules/node-opcua/lib/client/client_subscription.js:558:27)
node-red-0     at runCallback (timers.js:637:20)
node-red-0     at tryOnImmediate (timers.js:610:5)
node-red-0     at processImmediate [as _immediateCallback] (timers.js:582:5)
node-red-0 21 Feb 12:18:26 - [red] Uncaught Exception:
node-red-0 21 Feb 12:18:26 - AssertionError: self.subscription.session
node-red-0     at ClientMonitoredItem._monitor (/home/root/.node-red/node_modules/node-opcua/lib/client/client_monitored_item.js:190:5)
node-red-0     at Immediate.wait_for_subscription_and_monitor (/home/root/.node-red/node_modules/node-opcua/lib/client/client_subscription.js:558:27)
node-red-0     at runCallback (timers.js:637:20)
node-red-0     at tryOnImmediate (timers.js:610:5)
node-red-0     at processImmediate [as _immediateCallback] (timers.js:582:5)
node-red-0 21 Feb 12:18:28 - [info] 
`

I am using Node-Red v0.16.2 with node-red-contrib-opcua v0.2.13 on an Ubuntu host, yet I did test with Node-Red v0.15.1 and node-red-contrib-opcua v0.2.11 on a Raspberry Pi with the same result.

I would greatly appreciate someone taking a look at it. Many thanks!!

Mark

mikakaraila, 2017-02-21:
Ack, needs better error handling. I will take a look later today...

mikakaraila, 2017-02-21:
Can you attach test flow? I cannot repeat crash in windows...

mjunghanns, 2017-02-21:
Mika, thanks for looking after it! I don't have access to the flow at this moment, I will attach it tomorrow morning. I also heard from colleagues, that these crashes obviously do not occur on Windows setups.

reto88, 2017-02-22:
Me as well, yesterday the same problem Uncaught Exception, now i can not start the node red server again.
Do you have already a solution?

mikakaraila, 2017-02-22:
Actual exception is inside node-opcua. We have to wait it will be fixed.
https://github.com/node-opcua/node-opcua/issues/313
I can try look look some workaround...

mjunghanns, 2017-02-22:
reto88, I am convinced that Mika and the maintainers of node-opcua will do their best to come up with a fix soon.
Regarding your issue of a non-starting node red server I would guess, that the problem is caused by an injection node, that is configured to "inject on startup" which would inject the erroneousdata into the opcua node. A possible solution:

- create a backup of your flow_<hostname>.json file in your .node-red directory
- delete your flow_<hostname>.json and restart node-red. Your node-red instance will now be empty.
- import the backup'd flow, but DO NOT DEPLOY yet.
- go to your inject node and disable the "inject on startup" option.

Now your node-red instance should be operable again.

erossignon, 2017-02-22:
Could it be related to the fact that the opcua connection is droped when the node we are trying ti monitored doesn't exists:
https://github.com/mikakaraila/node-red-contrib-opcua/blob/7926b5d8f7457df00f18e82c33ebb56ecb974a78/opcua/102-opcuaclient.js#L500 ?
could we do something smarter here ?

mikakaraila, 2017-02-23:
I suppose node service is not available on client side or is it?
How to validate nodeId before adding it? Any prpoposals...

erossignon, 2017-02-23:
I would simply check the ```err``` in return to the ````subscrition.monitor``` callback and only register the item in the ```monitoredItems``` list if sucessful.
moving the ```  monitoredItems.add({"topicName": msg.topic, mItem: monitoredItem});``` inside the callback somehow.
It would be beneficial to turn node-red-opcua command into async methods, usually we need to wait for server answer (which happens asynchronously ) before stating on the success or failure of a command.

mikakaraila, 2017-02-23:
Hmm, seems to generate a lot assertions. Another workaround is to read nodeId once to validate it before adding it to monitoredItems.

mikakaraila, 2017-02-24:
Fixed, some more testing and I will push this back to github.

mjunghanns, 2017-02-24:
Sounds great! Thank you very much!

mikakaraila, 2017-02-24:
I tested this couple of times. Please test before I will publish new version into npm. Thanks!

mjunghanns, 2017-02-26:
Mika, I've just run some tests, which did not make the node-red instance crash anymore. Your fix works for me just perfectly. Many thanks for your quick and accurate support!
Regards,
Mark

mikakaraila, 2017-02-26:
OK, I published new version.

closed 2017-02-26 by mikakaraila