mikakaraila/node-red-contrib-opcua #31: Missing some browse attributes while browsing

giminni, 2017-02-22:
Hi Mika,

thanks for pointing me how to browse/read OPC/UA data points using node-RED.
Browsing an OPC/UA data point I need the field description and the datatype (Bool, Double, ...)

Can you add this missing fields to the main stream?

Thanks a lot

mikakaraila, 2017-02-22:
Bit busy today, but I will look tomorrow...
For what you will need these? I´m just curious.

giminni, 2017-02-24:
Here my answer:
- To identify a specific object I need the description field to extract some info from there
- To read a data value I must know first which data type it is (double, bool, ....)
- Some data points can only be read and/or written, therefore I need this info too.

mikakaraila, 2017-02-24:
Ack, perhaps I will introduce one new node for this... easier to use.

mikakaraila, 2017-02-28:
I actually added it into client code, I have to use existing session. Some work to encode data into payload.

mikakaraila, 2017-02-28:
Is this enough?
msg : Object
{ "topic": "ns=5;s=Sinusoid1", "payload": { "description": "", "browseName": "Sinusoid1", "userAccessLevel": "CurrentRead | CurrentWrite | HistoryRead", "accessLevel": "CurrentRead | CurrentWrite | HistoryRead", "type": "Double" }, "typeDefinition": "NUMERIC", "_msgid": "1b2ebd48.574623" }

mikakaraila, 2017-03-06:
Still waiting for feedback... code is otherwise ready.

giminni, 2017-03-08:
Hi Mika,
sorry I was busy. I have time to test tomorrow. Come back with findings and thanks.

mikakaraila, 2017-03-09:
New version pushed. Waiting for comments.

giminni, 2017-03-09:
Hi Mika,

right now I got only 0.2.14, I think I must wait a little longer to get the latest one.

Despite the fact, I use a browse node to get data out of a data point using 0.2.14 
and  could not see any new added items.

Am I doing something wrong?

mikakaraila, 2017-03-09:
Did you try get it fromgithub?
I will not publish it to npm.
Waiting for comments from others...

9.3.2017 6.10 ip.

giminni, 2017-03-10:
Hi Mika,
I tested it and i got the data from the client info node.

Wouldn't it be better to integrale the data into the browse node?

mikakaraila, 2017-03-19:
I do not want to break existing use of browse node.
Here are some examples how to use browse / info together.
I will add flow later to examples.
[{"id":"c8b446ba.bfa058","type":"OpcUa-Browser","z":"7e58ca8c.d78944","endpoint":"acd2f698.88bd28","item":"","datatype":"","topic":"ns=5;s=85/0:Simulation","items":[],"x":460.5000457763672,"y":33.79999542236328,"wires":[["83243287.79201","c9d289e3.ea1088"]]},{"id":"481f49fe.55a3a8","type":"inject","z":"7e58ca8c.d78944","name":"Browse","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":78.5,"y":31.79999542236328,"wires":[["3eb58696.791f7a"]]},{"id":"83243287.79201","type":"function","z":"7e58ca8c.d78944","name":"Decode & filter","func":"var items = msg.payload;\nfor (var i=0; i<items.length; i++) {\n    var item = items[i];\n    console.log(item);\n    var ref = item.item;\n    var nodeClass = ref.$nodeClass;\n    //console.log(nodeClass.value); // value==1 Variable\n    var typeDef1 = ref.typeDefinition.identifierType;\n    var typeDef2 = ref.nodeId.identifierType;\n    //var typeDef=ref.$nodeClass.value;\n    console.log(\"TYPE=\"+JSON.stringify(typeDef1));\n    console.log(\"TYPE=\"+JSON.stringify(typeDef2));\n    var bname = ref.browseName;\n    var ns=bname.namespaceIndex;\n    var name=bname.name;\n    \n    // Select only wanted namespace variables\n    if (ns==5) {\n        var newmsg={};\n        newmsg.topic = \"ns=\"+ns+\";s=\"+name;\n        newmsg.payload=\"\";\n        newmsg.typeDefinition=typeDef1;\n        node.send(newmsg);\n    }\n}\n","outputs":1,"noerr":0,"x":260.5000457763672,"y":139,"wires":[["877457b3.9972c8"]]},{"id":"de80c421.108fa8","type":"OpcUa-Client","z":"7e58ca8c.d78944","endpoint":"acd2f698.88bd28","action":"subscribe","time":"1","timeUnit":"s","name":"SUBSCRIBE","x":643.5,"y":95.20001220703125,"wires":[["6b267258.eb8e9c"]]},{"id":"6b267258.eb8e9c","type":"debug","z":"7e58ca8c.d78944","name":"","active":false,"console":"false","complete":"payload","x":854.2000427246094,"y":99.20001220703125,"wires":[]},{"id":"3eb58696.791f7a","type":"function","z":"7e58ca8c.d78944","name":"Set browse address","func":"msg.topic=\"ns=5;s=85/0:Simulation\";\nmsg.actiontype=\"browse\";\n\nreturn msg;","outputs":1,"noerr":0,"x":251.50001525878906,"y":32,"wires":[["c8b446ba.bfa058"]]},{"id":"c9d289e3.ea1088","type":"debug","z":"7e58ca8c.d78944","name":"","active":true,"console":"false","complete":"payload","x":701.933349609375,"y":34.933326721191406,"wires":[]},{"id":"3587fb90.5e3dd4","type":"OpcUa-Client","z":"7e58ca8c.d78944","endpoint":"acd2f698.88bd28","action":"browse","time":10,"timeUnit":"s","name":"","x":483.3666687011719,"y":227.0666732788086,"wires":[["a850713e.eee3a","f9fb5db7.02101"]]},{"id":"98fb0658.354ad8","type":"inject","z":"7e58ca8c.d78944","name":"Browse","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":101.93331909179688,"y":227.93331909179688,"wires":[["9b029e27.a3677"]]},{"id":"9b029e27.a3677","type":"function","z":"7e58ca8c.d78944","name":"Set browse address","func":"msg.topic=\"ns=0;i=85\";\nmsg.actiontype=\"browse\";\n\nreturn msg;","outputs":1,"noerr":0,"x":271.93333435058594,"y":226.1333236694336,"wires":[["3587fb90.5e3dd4"]]},{"id":"a850713e.eee3a","type":"debug","z":"7e58ca8c.d78944","name":"","active":false,"console":"false","complete":"true","x":675.933349609375,"y":224.93331909179688,"wires":[]},{"id":"cca0fd89.3345f","type":"OpcUa-Client","z":"7e58ca8c.d78944","endpoint":"acd2f698.88bd28","action":"subscribe","time":"10","timeUnit":"s","name":"","x":753.9333801269531,"y":265.93331146240234,"wires":[["b1ac92dd.97925"]]},{"id":"b1ac92dd.97925","type":"debug","z":"7e58ca8c.d78944","name":"","active":true,"console":"false","complete":"true","x":920.9333801269531,"y":265.93331146240234,"wires":[]},{"id":"f128c5f1.5ae0f8","type":"OpcUa-Client","z":"7e58ca8c.d78944","endpoint":"acd2f698.88bd28","action":"info","time":10,"timeUnit":"s","name":"","x":576.933349609375,"y":374.9333190917969,"wires":[["dfbd6595.32fbf8"]]},{"id":"480c6ff3.524aa","type":"inject","z":"7e58ca8c.d78944","name":"Get object info","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":145.5,"y":371.7999496459961,"wires":[["6f24fecd.8ee4a"]]},{"id":"dfbd6595.32fbf8","type":"debug","z":"7e58ca8c.d78944","name":"","active":true,"console":"false","complete":"payload","x":789.5000305175781,"y":372.79996490478516,"wires":[]},{"id":"6f24fecd.8ee4a","type":"OpcUa-Item","z":"7e58ca8c.d78944","item":"ns=5;s=Counter1","datatype":"Int32","value":"","name":"","x":355.3666687011719,"y":373.93335723876953,"wires":[["f128c5f1.5ae0f8"]]},{"id":"f9fb5db7.02101","type":"function","z":"7e58ca8c.d78944","name":"nodeId -> Topic","func":"if (msg.nodeId.toString().indexOf(\"ns=5\")>=0) {\n    msg.topic=msg.nodeId;\n    return msg;\n}","outputs":1,"noerr":0,"x":537.1666564941406,"y":305.06668853759766,"wires":[["a62010d6.04247","f128c5f1.5ae0f8"]]},{"id":"a62010d6.04247","type":"debug","z":"7e58ca8c.d78944","name":"","active":false,"console":"false","complete":"payload","x":796.933349609375,"y":313.9333190917969,"wires":[]},{"id":"877457b3.9972c8","type":"OpcUa-Client","z":"7e58ca8c.d78944","endpoint":"acd2f698.88bd28","action":"info","time":10,"timeUnit":"s","name":"Get object info","x":500.9333801269531,"y":140.93331146240234,"wires":[["948180e9.ea325"]]},{"id":"948180e9.ea325","type":"debug","z":"7e58ca8c.d78944","name":"","active":true,"console":"false","complete":"true","x":667.9333801269531,"y":141.93331146240234,"wires":[]},{"id":"acd2f698.88bd28","type":"OpcUa-Endpoint","z":"7e58ca8c.d78944","endpoint":"opc.tcp://localhost:53530/OPCUA/SimulationServer","login":false}]

sanlop73, 2017-05-03:
Hi:

Can not paste using Import->Clipboard.
Stays grey.
Any solution?

Thanks

mikakaraila, 2017-05-04:
Here is copy of browse & subscribe part of the flow:
[{"id":"815603e5.b0849","type":"OpcUa-Client","z":"f3e00663.6ac2e8","endpoint":"acd2f698.88bd28","action":"browse","time":10,"timeUnit":"s","name":"","x":516.433349609375,"y":102.9333267211914,"wires":[["f97b230a.72275"]]},{"id":"e2485f58.d95a4","type":"inject","z":"f3e00663.6ac2e8","name":"Browse","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":135,"y":103.79997253417969,"wires":[["7f3e5637.931e28"]]},{"id":"7f3e5637.931e28","type":"function","z":"f3e00663.6ac2e8","name":"Set
browse
address","func":"msg.topic=\"ns=0;i=85\";\nmsg.actiontype=\"browse\";\n\nreturn
msg;","outputs":1,"noerr":0,"x":305.00001525878906,"y":101.9999771118164,"wires":[["815603e5.b0849"]]},{"id":"90d27c78.6620f","type":"OpcUa-Client","z":"f3e00663.6ac2e8","endpoint":"acd2f698.88bd28","action":"subscribe","time":"10","timeUnit":"s","name":"","x":896.0001754760742,"y":104.79997825622559,"wires":[["d469b96.3c1ef48"]]},{"id":"d469b96.3c1ef48","type":"debug","z":"f3e00663.6ac2e8","name":"","active":true,"console":"false","complete":"true","x":1052.000072479248,"y":104.79998016357422,"wires":[]},{"id":"f97b230a.72275","type":"function","z":"f3e00663.6ac2e8","name":"nodeId
-> Topic","func":"if (msg.nodeId.toString().indexOf(\"ns=5\")>=0) {\n
 msg.topic=msg.nodeId;\n    return
msg;\n}","outputs":1,"noerr":0,"x":698.2333450317383,"y":103.9333610534668,"wires":[["90d27c78.6620f"]]},{"id":"acd2f698.88bd28","type":"OpcUa-Endpoint","z":"7e58ca8c.d78944","endpoint":"opc.tcp://localhost:53530/OPCUA/SimulationServer","login":false}]

sanlop73, 2017-05-04:
Good morning:

Sorry but tried to copy and paste using import and is still grey.

Can not import .

Could you please check it?

Thanks

[{"id":"815603e5.b0849","type":"OpcUa-Client","z":"f3e00663.6ac2e8","endpoint":"acd2f698.88bd28","action":"browse","time":10,"timeUnit":"s","name":"","x":516.433349609375,"y":102.9333267211914,"wires":[["f97b230a.72275"]]},{"id":"e2485f58.d95a4","type":"inject","z":"f3e00663.6ac2e8","name":"Browse","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":135,"y":103.79997253417969,"wires":[["7f3e5637.931e28"]]},{"id":"7f3e5637.931e28","type":"function","z":"f3e00663.6ac2e8","name":"Set
browse
address","func":"msg.topic=\"ns=0;i=85\";\nmsg.actiontype=\"browse\";\n\nreturn
msg;","outputs":1,"noerr":0,"x":305.00001525878906,"y":101.9999771118164,"wires":[["815603e5.b0849"]]},{"id":"90d27c78.6620f","type":"OpcUa-Client","z":"f3e00663.6ac2e8","endpoint":"acd2f698.88bd28","action":"subscribe","time":"10","timeUnit":"s","name":"","x":896.0001754760742,"y":104.79997825622559,"wires":[["d469b96.3c1ef48"]]},{"id":"d469b96.3c1ef48","type":"debug","z":"f3e00663.6ac2e8","name":"","active":true,"console":"false","complete":"true","x":1052.000072479248,"y":104.79998016357422,"wires":[]},{"id":"f97b230a.72275","type":"function","z":"f3e00663.6ac2e8","name":"nodeId
-> Topic","func":"if (msg.nodeId.toString().indexOf(\"ns=5\")>=0) {\n
msg.topic=msg.nodeId;\n return
msg;\n}","outputs":1,"noerr":0,"x":698.2333450317383,"y":103.9333610534668,"wires":[["90d27c78.6620f"]]},{"id":"acd2f698.88bd28","type":"OpcUa-Endpoint","z":"7e58ca8c.d78944","endpoint":"opc.tcp://localhost:53530/OPCUA/SimulationServer","login":false}]

mikakaraila, 2017-05-04:
Yes, copy to here seems to add newlines... I removed newlines and then it worked.

sanlop73, 2017-05-04:
Could you please send me into a file?

Thanks

mikakaraila, 2017-05-04:
[br_sub_copy.txt](https://github.com/mikakaraila/node-red-contrib-opcua/files/975314/br_sub_copy.txt)

sanlop73, 2017-05-04:
Hi:

Thanks a lot. It works!!

sanlop73, 2017-05-04:
Hi:

I have tested the flow but unfortunately does not work as expected.

I am using Ignition OPC server Simulation in my test.

I have setup the first function block "Set browse address" with ns=1;s=[Demo]_Meta:Refrigeration/Compressor2.

There are 8 tags inside this folder, the flow Subscribe the last tag 8 times , so i am getting 8 times every subscription period the same tag value.

Could you please check it what is wrong?

Thanks a lot

5/4/2017, 1:57:03 PMnode: e4b0d5ec.8d80a8<http://172.24.16.177:1880/#>ns=1;s=[Demo]_Meta:Refrigeration/Compressor2/presSuctionSP : msg : Object
{ topic: "ns=1;s=[Demo]_Meta:Refrigerati…", nodeId: "ns=1;s=[Demo]_Meta:Refrigerati…", browseName: "presSuctionSP", nodeClassType: "", typeDefinition: "VariableNode" … }
5/4/2017, 1:57:03 PMnode: e4b0d5ec.8d80a8<http://172.24.16.177:1880/#>ns=1;s=[Demo]_Meta:Refrigeration/Compressor2/presSuctionSP : msg : Object
{ topic: "ns=1;s=[Demo]_Meta:Refrigerati…", nodeId: "ns=1;s=[Demo]_Meta:Refrigerati…", browseName: "presSuctionSP", nodeClassType: "", typeDefinition: "VariableNode" … }
5/4/2017, 1:57:03 PMnode: e4b0d5ec.8d80a8<http://172.24.16.177:1880/#>ns=1;s=[Demo]_Meta:Refrigeration/Compressor2/presSuctionSP : msg : Object
{ topic: "ns=1;s=[Demo]_Meta:Refrigerati…", nodeId: "ns=1;s=[Demo]_Meta:Refrigerati…", browseName: "presSuctionSP", nodeClassType: "", typeDefinition: "VariableNode" … }
5/4/2017, 1:57:03 PMnode: e4b0d5ec.8d80a8<http://172.24.16.177:1880/#>ns=1;s=[Demo]_Meta:Refrigeration/Compressor2/presSuctionSP : msg : Object
{ topic: "ns=1;s=[Demo]_Meta:Refrigerati…", nodeId: "ns=1;s=[Demo]_Meta:Refrigerati…", browseName: "presSuctionSP", nodeClassType: "", typeDefinition: "VariableNode" … }
5/4/2017, 1:57:03 PMnode: e4b0d5ec.8d80a8<http://172.24.16.177:1880/#>ns=1;s=[Demo]_Meta:Refrigeration/Compressor2/presSuctionSP : msg : Object
{ topic: "ns=1;s=[Demo]_Meta:Refrigerati…", nodeId: "ns=1;s=[Demo]_Meta:Refrigerati…", browseName: "presSuctionSP", nodeClassType: "", typeDefinition: "VariableNode" … }
5/4/2017, 1:57:03 PMnode: e4b0d5ec.8d80a8<http://172.24.16.177:1880/#>ns=1;s=[Demo]_Meta:Refrigeration/Compressor2/presSuctionSP : msg : Object
{ topic: "ns=1;s=[Demo]_Meta:Refrigerati…", nodeId: "ns=1;s=[Demo]_Meta:Refrigerati…", browseName: "presSuctionSP", nodeClassType: "", typeDefinition: "VariableNode" … }
5/4/2017, 1:57:03 PMnode: e4b0d5ec.8d80a8<http://172.24.16.177:1880/#>ns=1;s=[Demo]_Meta:Refrigeration/Compressor2/presSuctionSP : msg : Object
{ topic: "ns=1;s=[Demo]_Meta:Refrigerati…", nodeId: "ns=1;s=[Demo]_Meta:Refrigerati…", browseName: "presSuctionSP", nodeClassType: "", typeDefinition: "VariableNode" … }
5/4/2017, 1:57:03 PMnode: e4b0d5ec.8d80a8<http://172.24.16.177:1880/#>ns=1;s=[Demo]_Meta:Refrigeration/Compressor2/presSuctionSP : msg : Object
{ topic: "ns=1;s=[Demo]_Meta:Refrigerati…", nodeId: "ns=1;s=[Demo]_Meta:Refrigerati…", browseName: "presSuctionSP", nodeClassType: "", typeDefinition: "VariableNode" … }
5/4/2017, 1:57:03 PMnode: e4b0d5ec.8d80a8<http://172.24.16.177:1880/#>ns=1;s=[Demo]_Meta:Refrigeration/Compressor2/presSuctionSP : msg : Object
{ topic: "ns=1;s=[Demo]_Meta:Refrigerati…", nodeId: "ns=1;s=[Demo]_Meta:Refrigerati…", browseName: "presSuctionSP", nodeClassType: "", typeDefinition: "VariableNode" … }
5/4/2017, 1:57:03 PMnode: e4b0d5ec.8d80a8<http://172.24.16.177:1880/#>ns=1;s=[Demo]_Meta:Refrigeration/Compressor2/presSuctionSP : msg : Object
{ topic: "ns=1;s=[Demo]_Meta:Refrigerati…", nodeId: "ns=1;s=[Demo]_Meta:Refrigerati…", browseName: "presSuctionSP", nodeClassType: "", typeDefinition: "VariableNode" … }
5/4/2017, 1:57:03 PMnode: e4b0d5ec.8d80a8<http://172.24.16.177:1880/#>ns=1;s=[Demo]_Meta:Refrigeration/Compressor2/presSuctionSP : msg : Object
{ topic: "ns=1;s=[Demo]_Meta:Refrigerati…", nodeId: "ns=1;s=[Demo]_Meta:Refrigerati…", browseName: "presSuctionSP", nodeClassType: "", typeDefinition: "VariableNode" … }
5/4/2017, 1:57:03 PMnode: e4b0d5ec.8d80a8<http://172.24.16.177:1880/#>ns=1;s=[Demo]_Meta:Refrigeration/Compressor2/presSuctionSP : msg : Object
{ topic: "ns=1;s=[Demo]_Meta:Refrigerati…", nodeId: "ns=1;s=[Demo]_Meta:Refrigerati…", browseName: "presSuctionSP", nodeClassType: "", typeDefinition: "VariableNode" … }
5/4/2017, 1:57:03 PMnode: e4b0d5ec.8d80a8<http://172.24.16.177:1880/#>ns=1;s=[Demo]_Meta:Refrigeration/Compressor2/presSuctionSP : msg : Object
{ topic: "ns=1;s=[Demo]_Meta:Refrigerati…", nodeId: "ns=1;s=[Demo]_Meta:Refrigerati…", browseName: "presSuctionSP", nodeClassType: "", typeDefinition: "VariableNode" … }
5/4/2017, 1:57:03 PMnode: e4b0d5ec.8d80a8<http://172.24.16.177:1880/#>ns=1;s=[Demo]_Meta:Refrigeration/Compressor2/presSuctionSP : msg : Object
{ topic: "ns=1;s=[Demo]_Meta:Refrigerati…", nodeId: "ns=1;s=[Demo]_Meta:Refrigerati…", browseName: "presSuctionSP", nodeClassType: "", typeDefinition: "VariableNode" … }
5/4/2017, 1:57:03 PMnode: e4b0d5ec.8d80a8<http://172.24.16.177:1880/#>ns=1;s=[Demo]_Meta:Refrigeration/Compressor2/presSuctionSP : msg : Object
{ topic: "ns=1;s=[Demo]_Meta:Refrigerati…", nodeId: "ns=1;s=[Demo]_Meta:Refrigerati…", browseName: "presSuctionSP", nodeClassType: "", typeDefinition: "VariableNode" … }
5/4/2017, 1:57:03 PMnode: e4b0d5ec.8d80a8<http://172.24.16.177:1880/#>ns=1;s=[Demo]_Meta:Refrigeration/Compressor2/presSuctionSP : msg : Object
{ topic: "ns=1;s=[Demo]_Meta:Refrigerati…", nodeId: "ns=1;s=[Demo]_Meta:Refrigerati…", browseName: "presSuctionSP", nodeClassType: "", typeDefinition: "VariableNode" … }
[... cut ...]
5/4/2017, 1:57:04 PMnode: e4b0d5ec.8d80a8<http://172.24.16.177:1880/#>ns=1;s=[Demo]_Meta:Refrigeration/Compressor2/presSuctionSP : msg : Object
{ topic: "ns=1;s=[Demo]_Meta:Refrigerati…", nodeId: "ns=1;s=[Demo]_Meta:Refrigerati…", browseName: "presSuctionSP", nodeClassType: "", typeDefinition: "VariableNode" … }
5/4/2017, 1:57:04 PMnode: e4b0d5ec.8d80a8<http://172.24.16.177:1880/#>ns=1;s=[Demo]_Meta:Refrigeration/Compressor2/presSuctionSP : msg : Object
{ topic: "ns=1;s=[Demo]_Meta:Refrigerati…", nodeId: "ns=1;s=[Demo]_Meta:Refrigerati…", browseName: "presSuctionSP", nodeClassType: "", typeDefinition: "VariableNode" … }

mikakaraila, 2018-03-25:
There seems to be in the client session new method that can be used to read all attribute values.
I will add it to my code...

mikakaraila, 2020-07-07:
Latest version (or long ago) support better browse attributes.

closed 2020-07-07 by mikakaraila