mikakaraila/node-red-contrib-opcua #32: Http Request Write To PLC

Riclas1, 2017-03-02:
Hi Mika,

right now, I’m trying to write data to an plc via our opc ua client module.  For Boolean I, have already find a way, with fixed values for every http slot (it is working….).  But now I’m struggling with Int values, because I like to use the data inside the payload. Did you have any suggestions?

Thank you upfront

Rico

mikakaraila, 2017-03-02:
Do you have example flow?

Riclas1, 2017-03-03:
Hi Mika,
thank you for your fast response. I´m new in this topic, and hopefully I didn´t overlooked something.
Here is my test example:

[{"id":"630db183.728c7","type":"tab","label":"Flow 1"},{"id":"ae114347.222d2","type":"http in","z":"630db183.728c7","name":"Http Post Start","url":"/Umrichter/Start","method":"post","swaggerDoc":"","x":85,"y":66,"wires":[["a74c1b3c.7de9b8","5c07382a.1faf88"]]},{"id":"eb3023ff.53524","type":"OpcUa-Client","z":"630db183.728c7","endpoint":"e29ba16a.aaa51","action":"write","time":"1","timeUnit":"s","name":"HMI_Start","x":880,"y":64,"wires":[[]]},{"id":"4bbe89cb.562df8","type":"OpcUa-Item","z":"630db183.728c7","item":"ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_start","datatype":"Boolean","value":"","name":"test","x":694,"y":65,"wires":[["eb3023ff.53524"]]},{"id":"43fce693.4adf78","type":"change","z":"630db183.728c7","name":"","rules":[{"t":"set","p":"payload","pt":"msg","to":"true","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":477,"y":67,"wires":[["4bbe89cb.562df8"]]},{"id":"a74c1b3c.7de9b8","type":"exec","z":"630db183.728c7","command":"inject","addpay":false,"append":"","useSpawn":"","timer":"","name":"","x":293,"y":67.5,"wires":[[],["43fce693.4adf78"],[]]},{"id":"5c07382a.1faf88","type":"http response","z":"630db183.728c7","name":"","x":337,"y":21,"wires":[]},{"id":"b33b1b3e.d83418","type":"exec","z":"630db183.728c7","command":"inject","addpay":false,"append":"","useSpawn":"","timer":"","name":"","x":297,"y":167,"wires":[[],["dd897d06.68ffe"],[]]},{"id":"dd897d06.68ffe","type":"change","z":"630db183.728c7","name":"","rules":[{"t":"set","p":"payload","pt":"msg","to":"false","tot":"msg"}],"action":"","property":"","from":"","to":"","reg":false,"x":481,"y":166.5,"wires":[["4a446d96.3d8244"]]},{"id":"f0e5468f.e1c688","type":"http in","z":"630db183.728c7","name":"Http Post Stop","url":"/Umrichter/Stop","method":"post","swaggerDoc":"","x":89,"y":165.5,"wires":[["b33b1b3e.d83418","88bcc26c.743ee"]]},{"id":"4a446d96.3d8244","type":"OpcUa-Item","z":"630db183.728c7","item":"ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_start","datatype":"Boolean","value":"","name":"test","x":698,"y":164.5,"wires":[["a35313d0.1f052"]]},{"id":"88bcc26c.743ee","type":"http response","z":"630db183.728c7","name":"","x":341,"y":120.5,"wires":[]},{"id":"a35313d0.1f052","type":"OpcUa-Client","z":"630db183.728c7","endpoint":"e29ba16a.aaa51","action":"write","time":"1","timeUnit":"s","name":"HMI_Stop","x":874,"y":163.5,"wires":[[]]},{"id":"2a24908b.3a4de","type":"http in","z":"630db183.728c7","name":"Http Post Pos","url":"/Umrichter/Pos","method":"post","swaggerDoc":"","x":75,"y":274.5,"wires":[["d84c37d2.721ab8","224a3d50.868262","4ca9ce35.1c7fc"]]},{"id":"d84c37d2.721ab8","type":"http response","z":"630db183.728c7","name":"","x":337,"y":229.5,"wires":[]},{"id":"f0396872.7945b8","type":"debug","z":"630db183.728c7","name":"","active":true,"console":"false","complete":"payload","x":291,"y":550,"wires":[]},{"id":"cef64516.ef8e58","type":"OpcUa-Client","z":"630db183.728c7","endpoint":"e29ba16a.aaa51","action":"write","time":"","name":"HMI_Pos","x":559,"y":413,"wires":[["d1ca3c6b.6b09c"]]},{"id":"69d65d77.90f824","type":"OpcUa-Item","z":"630db183.728c7","item":"ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos","datatype":"UInt32","value":"","name":"test","x":296,"y":414,"wires":[["cef64516.ef8e58"]]},{"id":"d1ca3c6b.6b09c","type":"debug","z":"630db183.728c7","name":"Write value","active":true,"console":"false","complete":"payload","x":837,"y":414,"wires":[]},{"id":"d35f9ae9.63d048","type":"inject","z":"630db183.728c7","name":"Write","topic":"","payload":"4000","payloadType":"str","repeat":"","crontab":"","once":false,"x":117,"y":414,"wires":[["69d65d77.90f824","f0396872.7945b8"]]},{"id":"8578999e.e7cad8","type":"debug","z":"630db183.728c7","name":"","active":true,"console":"false","complete":"true","x":568,"y":318,"wires":[]},{"id":"224a3d50.868262","type":"debug","z":"630db183.728c7","name":"","active":true,"console":"false","complete":"payload","x":108,"y":339,"wires":[]},{"id":"4ca9ce35.1c7fc","type":"function","z":"630db183.728c7","name":"","func":"msg = msg.req.body;\nreturn msg;","outputs":1,"noerr":0,"x":298,"y":301,"wires":[["69d65d77.90f824","8578999e.e7cad8"]]},{"id":"e29ba16a.aaa51","type":"OpcUa-Endpoint","z":"","endpoint":"opc.tcp://10.49.39.4:4840","login":false}]

mikakaraila, 2017-03-04:
I changed two nodes in your flow. I cannot test it, but check these:
1) function node, place http body into msg.payload
2) I changed payload to int and added topic into inject node

Modified flow:
[{"id":"fea15e31.cd8af","type":"http in","z":"12ebf4c8.88c21b","name":"Http Post Start","url":"/Umrichter/Start","method":"post","swaggerDoc":"","x":85,"y":66,"wires":[["b3ab4eae.9e1a2","ee956246.56359"]]},{"id":"4aff7580.f5bf5c","type":"OpcUa-Client","z":"12ebf4c8.88c21b","endpoint":"f0d38069.05e91","action":"write","time":"1","timeUnit":"s","name":"HMI_Start","x":880,"y":64,"wires":[[]]},{"id":"5383c8a.aa44638","type":"OpcUa-Item","z":"12ebf4c8.88c21b","item":"ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_start","datatype":"Boolean","value":"","name":"test","x":694,"y":65,"wires":[["4aff7580.f5bf5c"]]},{"id":"a4f015d8.9e7eb8","type":"change","z":"12ebf4c8.88c21b","name":"","rules":[{"t":"set","p":"payload","pt":"msg","to":"true","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":477,"y":67,"wires":[["5383c8a.aa44638"]]},{"id":"b3ab4eae.9e1a2","type":"exec","z":"12ebf4c8.88c21b","command":"inject","addpay":false,"append":"","useSpawn":"","timer":"","name":"","x":293,"y":67.5,"wires":[[],["a4f015d8.9e7eb8"],[]]},{"id":"ee956246.56359","type":"http response","z":"12ebf4c8.88c21b","name":"","x":337,"y":21,"wires":[]},{"id":"9ec4c2ff.e6f31","type":"exec","z":"12ebf4c8.88c21b","command":"inject","addpay":false,"append":"","useSpawn":"","timer":"","name":"","x":297,"y":167,"wires":[[],["ac619001.df0d3"],[]]},{"id":"ac619001.df0d3","type":"change","z":"12ebf4c8.88c21b","name":"","rules":[{"t":"set","p":"payload","pt":"msg","to":"false","tot":"msg"}],"action":"","property":"","from":"","to":"","reg":false,"x":481,"y":166.5,"wires":[["5c429c5a.f2f254"]]},{"id":"da3dd36a.f56d8","type":"http in","z":"12ebf4c8.88c21b","name":"Http Post Stop","url":"/Umrichter/Stop","method":"post","swaggerDoc":"","x":89,"y":165.5,"wires":[["9ec4c2ff.e6f31","ff10f0fe.d091d"]]},{"id":"5c429c5a.f2f254","type":"OpcUa-Item","z":"12ebf4c8.88c21b","item":"ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_start","datatype":"Boolean","value":"","name":"test","x":698,"y":164.5,"wires":[["a9419607.c91a28"]]},{"id":"ff10f0fe.d091d","type":"http response","z":"12ebf4c8.88c21b","name":"","x":341,"y":120.5,"wires":[]},{"id":"a9419607.c91a28","type":"OpcUa-Client","z":"12ebf4c8.88c21b","endpoint":"f0d38069.05e91","action":"write","time":"1","timeUnit":"s","name":"HMI_Stop","x":874,"y":163.5,"wires":[[]]},{"id":"2f7f7f0b.6e935","type":"http in","z":"12ebf4c8.88c21b","name":"Http Post Pos","url":"/Umrichter/Pos","method":"post","swaggerDoc":"","x":75,"y":274.5,"wires":[["4f8c0bb1.0c1c34","96e6ca19.23ea18","a105628c.365eb"]]},{"id":"4f8c0bb1.0c1c34","type":"http response","z":"12ebf4c8.88c21b","name":"","x":337,"y":229.5,"wires":[]},{"id":"2847449.ad306bc","type":"debug","z":"12ebf4c8.88c21b","name":"","active":true,"console":"false","complete":"payload","x":291,"y":550,"wires":[]},{"id":"86cd0452.c7a148","type":"OpcUa-Client","z":"12ebf4c8.88c21b","endpoint":"f0d38069.05e91","action":"write","time":"","name":"HMI_Pos","x":559,"y":413,"wires":[["2300c16e.7fb77e"]]},{"id":"fa54c328.9c428","type":"OpcUa-Item","z":"12ebf4c8.88c21b","item":"ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos","datatype":"UInt32","value":"","name":"test","x":296,"y":414,"wires":[["86cd0452.c7a148"]]},{"id":"2300c16e.7fb77e","type":"debug","z":"12ebf4c8.88c21b","name":"Write value","active":true,"console":"false","complete":"payload","x":837,"y":414,"wires":[]},{"id":"9ba2edaa.6d408","type":"inject","z":"12ebf4c8.88c21b","name":"Write","topic":"ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos;datatype=UInt32","payload":"4000","payloadType":"num","repeat":"","crontab":"","once":false,"x":117,"y":414,"wires":[["fa54c328.9c428","2847449.ad306bc"]]},{"id":"3fcbda90.c918d6","type":"debug","z":"12ebf4c8.88c21b","name":"","active":true,"console":"false","complete":"true","x":568,"y":318,"wires":[]},{"id":"96e6ca19.23ea18","type":"debug","z":"12ebf4c8.88c21b","name":"","active":true,"console":"false","complete":"payload","x":108,"y":339,"wires":[]},{"id":"a105628c.365eb","type":"function","z":"12ebf4c8.88c21b","name":"","func":"msg.payload = msg.req.body;\nreturn msg;","outputs":1,"noerr":0,"x":298,"y":301,"wires":[["fa54c328.9c428","3fcbda90.c918d6"]]},{"id":"f0d38069.05e91","type":"OpcUa-Endpoint","z":"","endpoint":"opc.tcp://10.49.39.4:4840","login":false}]

mikakaraila, 2017-03-04:
My test flow:
[{"id":"fea15e31.cd8af","type":"http in","z":"12ebf4c8.88c21b","name":"Http Post Start","url":"/Umrichter/Start","method":"post","swaggerDoc":"","x":85,"y":66,"wires":[["b3ab4eae.9e1a2","ee956246.56359"]]},{"id":"4aff7580.f5bf5c","type":"OpcUa-Client","z":"12ebf4c8.88c21b","endpoint":"f0d38069.05e91","action":"write","time":"1","timeUnit":"s","name":"HMI_Start","x":880,"y":64,"wires":[[]]},{"id":"5383c8a.aa44638","type":"OpcUa-Item","z":"12ebf4c8.88c21b","item":"ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_start","datatype":"Boolean","value":"","name":"test","x":694,"y":65,"wires":[["4aff7580.f5bf5c"]]},{"id":"a4f015d8.9e7eb8","type":"change","z":"12ebf4c8.88c21b","name":"","rules":[{"t":"set","p":"payload","pt":"msg","to":"true","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":477,"y":67,"wires":[["5383c8a.aa44638"]]},{"id":"b3ab4eae.9e1a2","type":"exec","z":"12ebf4c8.88c21b","command":"inject","addpay":false,"append":"","useSpawn":"","timer":"","name":"","x":293,"y":67.5,"wires":[[],["a4f015d8.9e7eb8"],[]]},{"id":"ee956246.56359","type":"http response","z":"12ebf4c8.88c21b","name":"","x":337,"y":21,"wires":[]},{"id":"9ec4c2ff.e6f31","type":"exec","z":"12ebf4c8.88c21b","command":"inject","addpay":false,"append":"","useSpawn":"","timer":"","name":"","x":297,"y":167,"wires":[[],["ac619001.df0d3"],[]]},{"id":"ac619001.df0d3","type":"change","z":"12ebf4c8.88c21b","name":"","rules":[{"t":"set","p":"payload","pt":"msg","to":"false","tot":"msg"}],"action":"","property":"","from":"","to":"","reg":false,"x":481,"y":166.5,"wires":[["5c429c5a.f2f254"]]},{"id":"da3dd36a.f56d8","type":"http in","z":"12ebf4c8.88c21b","name":"Http Post Stop","url":"/Umrichter/Stop","method":"post","swaggerDoc":"","x":89,"y":165.5,"wires":[["9ec4c2ff.e6f31","ff10f0fe.d091d"]]},{"id":"5c429c5a.f2f254","type":"OpcUa-Item","z":"12ebf4c8.88c21b","item":"ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_start","datatype":"Boolean","value":"","name":"test","x":698,"y":164.5,"wires":[["a9419607.c91a28"]]},{"id":"ff10f0fe.d091d","type":"http response","z":"12ebf4c8.88c21b","name":"","x":341,"y":120.5,"wires":[]},{"id":"a9419607.c91a28","type":"OpcUa-Client","z":"12ebf4c8.88c21b","endpoint":"f0d38069.05e91","action":"write","time":"1","timeUnit":"s","name":"HMI_Stop","x":874,"y":163.5,"wires":[[]]},{"id":"2f7f7f0b.6e935","type":"http in","z":"12ebf4c8.88c21b","name":"Http Post Pos","url":"/Umrichter/Pos","method":"post","swaggerDoc":"","x":75,"y":274.5,"wires":[["4f8c0bb1.0c1c34","96e6ca19.23ea18","a105628c.365eb"]]},{"id":"4f8c0bb1.0c1c34","type":"http response","z":"12ebf4c8.88c21b","name":"","x":337,"y":229.5,"wires":[]},{"id":"2847449.ad306bc","type":"debug","z":"12ebf4c8.88c21b","name":"","active":true,"console":"false","complete":"payload","x":291,"y":550,"wires":[]},{"id":"86cd0452.c7a148","type":"OpcUa-Client","z":"12ebf4c8.88c21b","endpoint":"f0d38069.05e91","action":"write","time":"","name":"HMI_Pos","x":559,"y":413,"wires":[["2300c16e.7fb77e"]]},{"id":"fa54c328.9c428","type":"OpcUa-Item","z":"12ebf4c8.88c21b","item":"ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos","datatype":"UInt32","value":"","name":"test","x":296,"y":414,"wires":[["86cd0452.c7a148"]]},{"id":"2300c16e.7fb77e","type":"debug","z":"12ebf4c8.88c21b","name":"Write value","active":true,"console":"false","complete":"payload","x":837,"y":414,"wires":[]},{"id":"9ba2edaa.6d408","type":"inject","z":"12ebf4c8.88c21b","name":"Write","topic":"ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos;datatype=UInt32","payload":"4000","payloadType":"num","repeat":"","crontab":"","once":false,"x":117,"y":414,"wires":[["fa54c328.9c428","2847449.ad306bc"]]},{"id":"3fcbda90.c918d6","type":"debug","z":"12ebf4c8.88c21b","name":"","active":true,"console":"false","complete":"true","x":568,"y":318,"wires":[]},{"id":"96e6ca19.23ea18","type":"debug","z":"12ebf4c8.88c21b","name":"","active":true,"console":"false","complete":"payload","x":108,"y":339,"wires":[]},{"id":"a105628c.365eb","type":"function","z":"12ebf4c8.88c21b","name":"","func":"msg.payload = msg.req.body;\nreturn msg;","outputs":1,"noerr":0,"x":298,"y":301,"wires":[["fa54c328.9c428","3fcbda90.c918d6"]]},{"id":"f0d38069.05e91","type":"OpcUa-Endpoint","z":"","endpoint":"opc.tcp://10.49.39.4:4840","login":false}]

Riclas1, 2017-03-06:
Hi Mikka,
i have try the changed example. I did also some changes at the function (“test”), now I have also in Topic in my Http line. The message from “DebugOutput” from the write line and the http line looks quite the sane right now.  But the write line is working and the http line I get some Error messages from “HMI_Pos”.

Thank you in advance

Messages Http:
6.3.2017, 09:10:52node: DebugOutput
ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos;datatype=UInt32 : msg.payload : number
3343

6.3.2017, 09:10:52node: HMI_Pos
msg : error
"TypeError: Converting circular structure to JSON"

6.3.2017, 09:10:52node: 63f31666.6f8278
ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos : msg.payload : number
3343

Message write.
6.3.2017, 09:12:13node: DebugOutput
ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos;datatype=UInt32 : msg.payload : number
4000

6.3.2017, 09:12:13node: 63f31666.6f8278
ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos : msg.payload : number
4000

Flow:
[{"id":"3340042f.488d8c","type":"http in","z":"6d577885.196908","name":"Http Post Start","url":"/Umrichter/Start","method":"post","swaggerDoc":"","x":123,"y":81,"wires":[["ed7b6c5c.89b3e","778118b7.8922f8"]]},{"id":"f39dbe5c.e0467","type":"OpcUa-Client","z":"6d577885.196908","endpoint":"10c5f1a1.ff00be","action":"write","time":"1","timeUnit":"s","name":"HMI_Start","x":918,"y":79,"wires":[[]]},{"id":"90353456.990458","type":"OpcUa-Item","z":"6d577885.196908","item":"ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_start","datatype":"Boolean","value":"","name":"test","x":732,"y":80,"wires":[["f39dbe5c.e0467"]]},{"id":"5ea83fb5.3a6c8","type":"change","z":"6d577885.196908","name":"","rules":[{"t":"set","p":"payload","pt":"msg","to":"true","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":548,"y":81,"wires":[["90353456.990458","de818c27.d74bb"]]},{"id":"778118b7.8922f8","type":"exec","z":"6d577885.196908","command":"inject","addpay":true,"append":"","useSpawn":"","timer":"","name":"","x":352,"y":81.5,"wires":[[],["5ea83fb5.3a6c8"],[]]},{"id":"ed7b6c5c.89b3e","type":"http response","z":"6d577885.196908","name":"","x":375,"y":36,"wires":[]},{"id":"e9a264a9.bd89e8","type":"http in","z":"6d577885.196908","name":"Http Post Stop","url":"/Umrichter/Stop","method":"post","swaggerDoc":"","x":127,"y":180.5,"wires":[["24d61cab.aa9d34","6b4f6234.bc445c"]]},{"id":"be68b6db.5c5968","type":"OpcUa-Item","z":"6d577885.196908","item":"ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_start","datatype":"Boolean","value":"","name":"test","x":736,"y":179.5,"wires":[["c4ba1acf.b2c0b8"]]},{"id":"24d61cab.aa9d34","type":"http response","z":"6d577885.196908","name":"","x":379,"y":135.5,"wires":[]},{"id":"c4ba1acf.b2c0b8","type":"OpcUa-Client","z":"6d577885.196908","endpoint":"10c5f1a1.ff00be","action":"write","time":"1","timeUnit":"s","name":"HMI_Stop","x":912,"y":178.5,"wires":[[]]},{"id":"243b06ff.9bf62a","type":"http in","z":"6d577885.196908","name":"Http Post Pos","url":"/Umrichter/Pos1","method":"post","swaggerDoc":"","x":113,"y":289.5,"wires":[["babfead9.401b18","ac2376f3.8e7548"]]},{"id":"babfead9.401b18","type":"http response","z":"6d577885.196908","name":"","x":375,"y":244.5,"wires":[]},{"id":"f08920d1.12824","type":"OpcUa-Client","z":"6d577885.196908","endpoint":"10c5f1a1.ff00be","action":"write","time":"","name":"HMI_Pos","x":915,"y":287,"wires":[["119dc5e3.d0095a"]]},{"id":"5506f79.ca08908","type":"inject","z":"6d577885.196908","name":"Write","topic":"ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos;datatype=UInt32","payload":"4000","payloadType":"num","repeat":"","crontab":"","once":false,"x":108,"y":419,"wires":[["68ce7309.76bb9c","433a8870.609118"]]},{"id":"ac2376f3.8e7548","type":"function","z":"6d577885.196908","name":"test","func":"msg.topic = \"ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos;datatype=UInt32\";\nmsg.payload = parseInt(msg.req.body);\nreturn msg;","outputs":1,"noerr":0,"x":353,"y":289,"wires":[["68ce7309.76bb9c","433a8870.609118"]]},{"id":"63f31666.6f8278","type":"debug","z":"6d577885.196908","name":"","active":true,"console":"false","complete":"false","x":947,"y":388,"wires":[]},{"id":"68ce7309.76bb9c","type":"debug","z":"6d577885.196908","name":"DebugOutput","active":true,"console":"false","complete":"payload","x":644,"y":433,"wires":[]},{"id":"de818c27.d74bb","type":"debug","z":"6d577885.196908","name":"","active":true,"console":"false","complete":"false","x":747,"y":122,"wires":[]},{"id":"9932bfa9.bca7f","type":"debug","z":"6d577885.196908","name":"","active":true,"console":"false","complete":"false","x":745,"y":230,"wires":[]},{"id":"22464e7.af852b2","type":"change","z":"6d577885.196908","name":"","rules":[{"t":"set","p":"payload","pt":"msg","to":"false","tot":"str"}],"action":"","property":"","from":"","to":"","reg":false,"x":551,"y":179,"wires":[["be68b6db.5c5968"]]},{"id":"6b4f6234.bc445c","type":"exec","z":"6d577885.196908","command":"inject","addpay":true,"append":"","useSpawn":"","timer":"","name":"","x":355,"y":179.5,"wires":[[],["22464e7.af852b2"],[]]},{"id":"433a8870.609118","type":"OpcUa-Item","z":"6d577885.196908","item":"ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos","datatype":"UInt32","value":"","name":"test","x":739,"y":288,"wires":[["f08920d1.12824","63f31666.6f8278"]]},{"id":"119dc5e3.d0095a","type":"debug","z":"6d577885.196908","name":"Write value","active":true,"console":"false","complete":"payload","x":1115,"y":288,"wires":[]},{"id":"10c5f1a1.ff00be","type":"OpcUa-Endpoint","z":"","endpoint":"opc.tcp://10.49.39.4:4840","login":false}]

mikakaraila, 2017-03-06:
Are you running with verbose on?
"node-red -v TEST.json"
> 6.3.2017, 09:10:52node: HMI_Pos
> msg : error
> "TypeError: Converting circular structure to JSON"
Can you try without verbose? So use just "node-red test.json"
There are some debug verbose message with JSON.stringify(msg) that can cause this.

Or another case can be is your msg.req.body is empty string -> msg.payload = NaN
And this causes error later...

Riclas1, 2017-03-06:
Hi Mikka,
i´m run node-red embedded in my mean-stack application. 
As you see below the payload has (msg.payload : number 3343) and the topic ( ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos;datatype=UInt32) right before the opc ua item.
6.3.2017, 09:10:52node: DebugOutput
ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos;datatype=UInt32 : msg.payload : number
3343
6.3.2017, 09:10:52node: HMI_Pos
msg : error
"TypeError: Converting circular structure to JSON"

Thank you in advanced

Rico

mikakaraila, 2017-03-06:
Can you add debug node before client node to Print out msg?

Riclas1, 2017-03-07:
Hi Mikka,

at first thank for your patience. 

**HTTP**
This is the debug right after the function node:
7.3.2017, 07:42:01node: DebugOutput
ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos;datatype=UInt32 : msg : Object
object
_msgid: "451d4237.0c49ec"
payload: 333
topic: "ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos;datatype=UInt32"
req: object
res: object

This is the debug right after your opcua-item node before opcua-client:
7.3.2017, 07:42:01node: 63f31666.6f8278
ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos : msg : Object
object
_msgid: "451d4237.0c49ec"
payload: 333
topic: "ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos"
datatype: "UInt32"
browseName: "test"
req: object
res: object

**Manual**
This is the debug right after the function node:
7.3.2017, 07:44:05node: DebugOutput
ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos;datatype=UInt32 : msg : Object
object
_msgid: "8ae91b05.e7ce78"
topic: "ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos;datatype=UInt32"
payload: 4000

This is the debug right after your opcua-item node before opcua-client:
ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos : msg : Object
object
_msgid: "8ae91b05.e7ce78"
topic: "ns=4;s=|var|CODESYS Control for Raspberry Pi SL.Application.PLC_PRG.HMI_Pos"
payload: 4000
datatype: "UInt32"
browseName: "test"

**One more thing:**
 
I got the same message from the “HMI_Stop” opcua-client node, if I doesn’t this injection node befor.
Maybe it help’s a little bit further.

Thank you in advanced

Rico

mikakaraila, 2017-03-07:
Hmm, here is explanation why we get this error message:
http://stackoverflow.com/questions/11616630/json-stringify-avoid-typeerror-converting-circular-structure-to-json
Now I just should find where this kind of structure is build...

Riclas1, 2017-03-16:
Hi Mika,

one small question. Are you still looking for it??

Best Regards 
Rico

mikakaraila, 2017-03-16:
Did you test with the latest version from github?
I will have time to look next weekend.... just now busy.

mikakaraila, 2017-03-19:
Look your flow and function node.
Modify code in the node and test please:
//msg.payload = msg.req.body;
var newmsg={};
newmsg.payload = msg.req.body;
return newmsg;

Riclas1, 2017-03-21:
Hi Mikka,

thank you for your pations. The last option with the function you gave worked out.

closed 2017-03-21 by Riclas1