mikakaraila/node-red-contrib-opcua #33: Server- / Source-Timestamp

MerkAndreas, 2017-03-02:
Hi Mika, 

Is it possible to get the timestamps back in the client read and subscribe functions? 
In the node-opcua library is it possible. 

Thanks and regards, 
Andreas

mikakaraila, 2017-03-03:
Ack, I will look during weekend.

mikakaraila, 2017-03-04:
Hmm, in many cases those fields sourceTimestamp and serverTimestamp seems to be null.
And sourcePicoseconds and serverPicoseconds are 0.

mikakaraila, 2017-03-04:
This I could add into msg object or payload, is this enough?
// Check if timestamps exists otherwise simulate them
					if (dataValue.serverTimestamp!=null) {
						msg.serverTimestamp = dataValue.serverTimestamp;
						msg.serverPicoseconds = dataValue.serverPicoseconds;
					}
					else {
						msg.serverTimestamp = new Date().getTime();;
						msg.serverPicoseconds = 0;
					}
					if (dataValue.sourceTimestamp!=null) {
						msg.sourceTimestamp = dataValue.sourceTimestamp;
						msg.sourcePicoseconds = dataValue.sourcePicoseconds;
					}
					else {
						msg.sourceTimestamp = new Date().getTime();;
						msg.sourcPicoseconds = 0;
					}

MerkAndreas, 2017-03-07:
Hello Mika,
I don´t understand. I think in all cases, the source- and servertimestamp shold be written from the server and return a value. Without the timestamp, you can´t use the data, because it´s only a value in the space. OPC UA is basically in use for time-series-data.

Your solution is more than enough. 
If the value returns null, in my opinion it´s better to set Unix time zero (1970-01-01T00:00:00Z) or forward null, because of the user know it´s no valid timestamp. The client should NOT overwrite the timestamp.
[http://blog.matrikonopc.com/index.php/ask-the-experts-opc-questions-and-answers/6-where-do-opc-timestamps-they-come-from/ ]

Greetz,
Andreas

mikakaraila, 2017-03-07:
Ok, I prefer null. I will test msg.serverTimestamp -> msg.payload.serverTimestamp this is perhaps better.
I will study why server is not updating timestamps before I will publish this.

MerkAndreas, 2017-03-08:
OK, thanks. I have no preference if msg.serverTimestamp or msg.payload.serverTimestamp. Essential is, we receive a value in node-red ;-) THANX!

mikakaraila, 2017-03-09:
Experimental code timestamp in msg object is in the new version. You can test, but I will modify this part.
So do not yet build bigger flow with this one.

mikakaraila, 2017-07-21:
I noticed this from the node-opcua:
https://github.com/node-opcua/node-opcua/issues/30

mikakaraila, 2017-07-21:
I suppose you could test this by adding this:
var TimestampsToReturn = require("node-opcua/lib/services/read_service").TimestampsToReturn;
and into monitor call
timestampsToReturn: TimestampsToReturn.Both

Do you need this? I can edit & publish code with this option.

nospam2000, 2017-07-22:
by the way, there is a typo in the code:
typo:
`msg.sourcPicoseconds = 0;`
correct:
`msg.sourcePicoseconds = 0;`

mikakaraila, 2017-07-24:
Yep, fixed typo and tested option, but I did not get timestamps from the server.
Have to test more...

mikakaraila, 2017-07-24:
OK, seems to be so that subscribing can be parametrized to return timestamps.
But read action will not return timestamps.

mikakaraila, 2017-07-24:
Fixed and tested. Read will use local current time.

MerkAndreas, 2017-07-28:
Good job, thanks a lot! It´s running!

brubbel, 2017-08-14:
Hi,

Would it be possible to include the source- and servertimestamps in the output message of the client?
Like this:
[image]

Thanks!

mikakaraila, 2017-08-14:
Hmm, this should be already in github. It is not yet in npm.
I changed subscription option to get Both timestamps (server&source).
So please test with the latest version from the github.

brubbel, 2017-08-14:
Could it be that this was corrected for [subscribe_monitoredItem](https://github.com/mikakaraila/node-red-contrib-opcua/blob/6dae1ebfe730bfee9dcce8f64f32e039c6e06371/opcua/102-opcuaclient.js#L578) but not yet in case of [read_action_input](https://github.com/mikakaraila/node-red-contrib-opcua/blob/6dae1ebfe730bfee9dcce8f64f32e039c6e06371/opcua/102-opcuaclient.js#L330)?

mikakaraila, 2017-08-17:
Ack, the function I used do not support timestamps. I should change code to support timestamps.

mikakaraila, 2018-03-25:
This is not directly possible, client session sets readVariableValue not to return timestamps:

var request = new read_service.ReadRequest({
        nodesToRead: nodesToRead,
        timestampsToReturn: read_service.TimestampsToReturn.Neither
});

mikakaraila, 2020-05-09:
Changed read to new function that will allow single read to return timestamps

commit b359a46c (2017-07-24): Version++, fixed #33
closed 2020-05-09 by mikakaraila