mikakaraila/node-red-contrib-opcua #34: node-red opc server can't be accessed by locally running opc sample clients

jtmoderate876, 2017-03-22:
Ideas?

I can connect to the Sample OPC Server (from opc foundation) from 2 different clients (Wonderware and OPC Sample Client).

But neither of the 2 clients can cannect to node-red running:
npm install -g node-red-contrib-opcua
node-RED v0.16.2 (latest)
nodejs v6.10.1 (latest stable)
windows 2012 server

I've added your latest cleint flows from OPCUA-TEST.json

and from within node-red the clients fail until the server creates the integer (as we might expect) then after the server create the the clients can read and write.

but non of this shows up external to node-red.

Security?

mikakaraila, 2017-03-23:
Which server creates the integer?
And do you mean external like ua expert cannot see new integer?
Have you tested rebrowse or reconnect? Address space change notification is not implemented or it is seldom used. So it depends on both ends how it works...

mikakaraila, 2017-04-10:
Did you test this with UA Expert?  
It requires to close address space if you have it open/focus on branch with new nodes and then re-browse it.

mikakaraila, 2017-04-13:
Issue #38 shows actual bug, sign options are not used as creating client.
Just fixing it...

mikakaraila, 2017-04-13:
OK now server registers itself. Please test.

mikakaraila, 2017-05-23:
Any results? I close this now and please re-open if not working...

closed 2017-05-23 by mikakaraila