mikakaraila/node-red-contrib-opcua #35: OPC/UA session is still open after node execution or redeployment

giminni, 2017-03-23:
Hi Mika,

I found out that I had a bunch of opened opc-ua session looking at my opc-ua client.

If I redeploy or trigger an opc-ua node the session persist and is not closed after execution.

Is there a way to close the session after execution or redeployment?

mikakaraila, 2017-03-23:
I think this is more cosmetic, but I will take a look... this can be caused by keep-alive-manager.
It will keep session alive even it is not reading/writing anything.
Redeploy options can effect full/flow/nodes modified will cause something.

mikakaraila, 2017-07-24:
Hmm, I tested this just now and here are some output and browse closes session:
[image]
[image]
So after browse there is no extra sessions open. Can you provide more detailed steps to reproduce error?

mikakaraila, 2017-07-24:
I added some code that will check node.session is existing and if not report error and reset client.

nospam2000, 2017-07-24:
Thanks a lot!
Last week I saw multiple times, that there have been up to 177 active OPC UA connections to my test PLC (that's its limit) and none of them was working. I think the reason was, that I started around 200 item subscriptions before the OPC connection was established. I will have an eye on it if your change also solves my problem.

mikakaraila, 2017-07-25:
Just pushed new fix into the github. Please test & provide feedback / examples if not working.

mikakaraila, 2017-07-27:
I think this is solved or #58 will solve connecting state message handling.

closed 2017-07-27 by mikakaraila

giminni, 2017-07-27:
Thanks a lot Mika!