mikakaraila/node-red-contrib-opcua #36: Timeout after 60 seconds - OpenSecureChannelChannel

mjunghanns, 2017-04-06:
Hello Mika,
we just encountered another issue. Our setup is the following:
- Linux (Yocto based) with 4.4 kernel
- NodeJS v6.9.2
- Node-Red v16.2
- node-contrib-opcua v0.2.13 / v0.2.14

The issue itself is quite strange. No matter how long the Node-Red instance was running, precisely 60 seconds after the first client subscription the node will time out ("waiting for response for OpenSecureChannelRequest") and cease functionality until the next Node-Red restart. By the way: It does not make any difference wether there is any payload being generated within the mentioned 60 seconds. Also, we successfully checked on a Windows based setup, this issue seems to be bound to Linux.

```
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8] ns=2;s=Production Data.Pieces.Good value has changed to 11114
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8]       Status-Code:Good (0x00000)
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8] Client status: active subscribed
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8] ns=2;s=Production Data.Pieces.Good value has changed to 11116
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8]       Status-Code:Good (0x00000)
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8] Client status: active subscribed
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8] ns=2;s=Production Data.Pieces.Good value has changed to 11117
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8]       Status-Code:Good (0x00000)
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8] Client status: active subscribed
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8] ns=2;s=Production Data.Pieces.Good value has changed to 11116
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8]       Status-Code:Good (0x00000)
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8] Client status: active subscribed
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8] ns=2;s=Production Data.Pieces.Good value has changed to 11119
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8]       Status-Code:Good (0x00000)
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8] Client status: active subscribed
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8] ns=2;s=Production Data.Pieces.Good value has changed to 11120
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8]       Status-Code:Good (0x00000)
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8] Client status: active subscribed
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8] ns=2;s=Production Data.Pieces.Good value has changed to 11121
node-red-0 3 Apr 15:12:46 - [info] [OpcUa-Client:b1c81faa.5f0af8]       Status-Code:Good (0x00000)
node-red-0 3 Apr 15:12:47 - [info] [OpcUa-Client:dfdf745a.f051d] Client status: active subscribed
node-red-0 3 Apr 15:12:47 - [info] [OpcUa-Client:dfdf745a.f051d] ns=2;s=Production Data.Pieces.Actual value has changed to 11150
node-red-0 3 Apr 15:12:47 - [info] [OpcUa-Client:dfdf745a.f051d]        Status-Code:Good (0x00000)
node-red-0 3 Apr 15:12:47 - [info] [OpcUa-Client:dfdf745a.f051d] Client status: active subscribed
node-red-0 3 Apr 15:12:47 - [info] [OpcUa-Client:dfdf745a.f051d] ns=2;s=Production Data.Pieces.Actual value has changed to 11153
node-red-0 3 Apr 15:12:47 - [info] [OpcUa-Client:dfdf745a.f051d]        Status-Code:Good (0x00000)
node-red-0 3 Apr 15:12:47 - [info] [OpcUa-Client:dfdf745a.f051d] Client status: active subscribed
node-red-0 3 Apr 15:12:47 - [info] [OpcUa-Client:dfdf745a.f051d] ns=2;s=Production Data.Pieces.Actual value has changed to 11155
node-red-0 3 Apr 15:12:47 - [info] [OpcUa-Client:dfdf745a.f051d]        Status-Code:Good (0x00000)
node-red-0 3 Apr 15:12:47 - [info] [OpcUa-Client:dfdf745a.f051d] Client status: active subscribed
node-red-0 3 Apr 15:12:47 - [info] [OpcUa-Client:dfdf745a.f051d] ns=2;s=Production Data.Pieces.Actual value has changed to 11157
node-red-0 3 Apr 15:12:47 - [info] [OpcUa-Client:dfdf745a.f051d]        Status-Code:Good (0x00000)
node-red-0  Timeout .... waiting for response for  OpenSecureChannelRequest { /*RequestHeader*/
node-red-0  authenticationToken           /* NodeId                           */: ns=0;i=0
node-red-0  timeStamp                     /* UtcTime                          */: 2017-04-03T15:11:47.876Z
node-red-0  requestHandle                 /* IntegerId                        */: 1               0x1
node-red-0  returnDiagnostics             /* UInt32                           */: 0               0x0
node-red-0  auditEntryId                  /* UAString                         */: null
node-red-0  timeoutHint                   /* UInt32                           */: 0               0x0
node-red-0  additionalHeader              /* ExtensionObject                  */: null
node-red-0 };
node-red-0 3 Apr 15:12:47 - [error] [OpcUa-Client:57e0e81b.0c0a7]  OPC UA connection error: Transaction has timed out
node-red-0 3 Apr 15:12:47 - [info] [OpcUa-Client:57e0e81b.0c0a7] Error: Transaction has timed out
node-red-0 3 Apr 15:12:47 - [info] [OpcUa-Client:57e0e81b.0c0a7] Client status: connection error
node-red-0 warning : disconnection : closing pending sessions
node-red-0  Timeout .... waiting for response for  OpenSecureChannelRequest { /*RequestHeader*/
node-red-0  authenticationToken           /* NodeId                           */: ns=0;i=0
node-red-0  timeStamp                     /* UtcTime                          */: 2017-04-03T15:11:47.894Z
node-red-0  requestHandle                 /* IntegerId                        */: 1               0x1
node-red-0  returnDiagnostics             /* UInt32                           */: 0               0x0
node-red-0  auditEntryId                  /* UAString                         */: null
node-red-0  timeoutHint                   /* UInt32                           */: 0               0x0
node-red-0  additionalHeader              /* ExtensionObject                  */: null
node-red-0 };
node-red-0  Timeout .... waiting for response for  OpenSecureChannelRequest { /*RequestHeader*/
node-red-0  authenticationToken           /* NodeId                           */: ns=0;i=0
node-red-0  timeStamp                     /* UtcTime                          */: 2017-04-03T15:11:47.898Z
node-red-0  requestHandle                 /* IntegerId                        */: 1               0x1
node-red-0  returnDiagnostics             /* UInt32                           */: 0               0x0
node-red-0  auditEntryId                  /* UAString                         */: null
node-red-0  timeoutHint                   /* UInt32                           */: 0               0x0
node-red-0  additionalHeader              /* ExtensionObject                  */: null
node-red-0 };
```

I would appreciate you taking a look at it.
Many thanks in advance!

Best regards
Mark

mikakaraila, 2017-04-06:
Hmm are you sure that all the variables are correct? If one has typo wrong name it could cause this kind of error.
Otherwise I will take a look during weekend...

mikakaraila, 2017-04-09:
Can you provide small test flow? I have tested now and cannot repeat problem immediately...
I have read, write and subscribe nodes in my flow.

mjunghanns, 2017-04-13:
Mika, unfortunately I caught the flu and cannot be at the office this week. I will try to have someone send me the affected flow. Thanks for your patience!

mikakaraila, 2017-07-04:
Status?

mjunghanns, 2017-07-10:
Sorry for the late reply, Mika. The problem did not occur in our test environment. Most likely it was caused by some misconfiguration. I apologize for wasting your time!

closed 2017-07-10 by mjunghanns

REugen, 2017-09-04:
Hi mjunghanns,
how did you solve your problem? I got the same issues, and cant find out how to fix it

regards REugen

mikakaraila, 2017-09-04:
Latest fix is in github not yet in npm.
Code will buffer actions until node is connected.