mikakaraila/node-red-contrib-opcua #37: Issues with node red OPC UA

johnjohnguinness, 2017-04-09:
Hi, Ive had node red OPC UA running on my PC for a few weeks now and its just stopped working. Ive uninstalled NODE JS and reinstalled both it and the npm install node red and OPC ua but when I run node red from CMD prompt it gets  so far through the start process and crashes. 
The errors are shown here:
9 Apr 22:53:35 - [error] [OpcUa-Client:6ee1b705.cfadd8] subscription.monitorEvent:Error: BadMonitoredItemFilterUnsupported (0x80440000)warning : disconnection : closing pending sessions
9 Apr 22:53:36 - [red] Uncaught Exception:
9 Apr 22:53:36 - AssertionError: self.subscription.session    at ClientMonitoredItem._monitor (C:\Users\John Egan\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua\lib\client\client_monitored_item.js:190:5)    at Immediate.wait_for_subscription_and_monitor (C:\Users\John Egan\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua\lib\client\client_subscription.js:558:27)    at runCallback (timers.js:666:20)    at tryOnImmediate (timers.js:639:5)    at processImmediate [as _immediateCallback] (timers.js:611:5). 
Slack as indicated that "Looks like the opcua node isn’t handling all the errors it should"
Is there a fix for this?

mikakaraila, 2017-04-10:
Ack, needs try /catch but I will check new node-opcua v0.0.64 it contains improvements & fixes.

johnjohnguinness, 2017-04-10:
Thanks,
Is V0.0.64 available for download?

johnjohnguinness, 2017-04-10:
Hi I tried run node red in verbose, I got the result below:
node-red -v
10 Apr 11:48:53 - [info]

Welcome to Node-RED
===================

10 Apr 11:48:53 - [info] Node-RED version: v0.16.2
10 Apr 11:48:53 - [info] Node.js  version: v6.10.2
10 Apr 11:48:53 - [info] Windows_NT 10.0.10586 x64 LE
10 Apr 11:49:09 - [info] Loading palette nodes
10 Apr 11:50:11 - [info] Dashboard version 2.3.3 started at /ui
10 Apr 11:50:12 - [warn] ------------------------------------------------------
10 Apr 11:50:12 - [warn] [rpi-gpio] Info : Ignoring Raspberry Pi specific node
10 Apr 11:50:12 - [warn] [tail] Not currently supported on Windows.
10 Apr 11:50:12 - [warn] ------------------------------------------------------
10 Apr 11:50:12 - [info] Settings file  : \Users\John Egan\.node-red\settings.js
10 Apr 11:50:12 - [info] User directory : \Users\John Egan\.node-red
10 Apr 11:50:12 - [info] Flows file     : \Users\John Egan\.node-red\flows_DESKTOP-I9CEULK.json
10 Apr 11:50:12 - [info] Server now running at http
10 Apr 11:50:12 - [info] Starting flows
10 Apr 11:50:12 - [warn] [OpcUa-Client:6ee1b705.cfadd8] OpcUaClientNode: create Client ...
10 Apr 11:50:12 - [info] [OpcUa-Client:6ee1b705.cfadd8] Client status: create client
10 Apr 11:50:12 - [info] [OpcUa-Client:6ee1b705.cfadd8] async series - connecting
10 Apr 11:50:12 - [info] [OpcUa-Client:6ee1b705.cfadd8] Client status: connecting
10 Apr 11:50:12 - [info] [OpcUa-Browser:e728f54a.716688] start browse client on 
10 Apr 11:50:12 - [info] [inject:COUNT] repeat = 1000
10 Apr 11:50:12 - [info] Started flows
10 Apr 11:50:12 - [warn] [OpcUa-Client:6ee1b705.cfadd8] OpcUaClientNode: can't work without OPC UA Session
10 Apr 11:50:12 - [info] [OpcUa-Client:6ee1b705.cfadd8] async series - create session ...
10 Apr 11:50:12 - [error] [OpcUa-Client:6ee1b705.cfadd8]  OPC UA connection error: self._secureChannel
10 Apr 11:50:12 - [info] [OpcUa-Client:6ee1b705.cfadd8] AssertionError: self._secureChannel
10 Apr 11:50:12 - [info] [OpcUa-Client:6ee1b705.cfadd8] Client status: connection error
10 Apr 11:50:12 - [info] [OpcUa-Client:6ee1b705.cfadd8] Client disconnected!
10 Apr 11:50:12 - [info] [OpcUa-Browser:e728f54a.716688] start browse session on opc.tcp://192.168.1.17:49320
10 Apr 11:50:12 - [warn] [OpcUa-Browser:e728f54a.716688] browseTopic:ns=0;i=85
10 Apr 11:50:12 - [warn] [OpcUa-Browser:e728f54a.716688] sending items 7
10 Apr 11:50:12 - [warn] [OpcUa-Browser:e728f54a.716688] close browse session
10 Apr 11:50:12 - [info] [mqtt-broker:ea9294b.872b9e8] Connected to broker: mqtt://iot.eclipse.org:1883
10 Apr 11:50:13 - [info] [OpcUa-Client:6ee1b705.cfadd8] Client disconnected!
10 Apr 11:50:13 - [warn] [OpcUa-Client:6ee1b705.cfadd8] OpcUaClientNode: create Client ...
10 Apr 11:50:13 - [info] [OpcUa-Client:6ee1b705.cfadd8] Client status: create client
10 Apr 11:50:13 - [info] [OpcUa-Client:6ee1b705.cfadd8] async series - connecting
10 Apr 11:50:13 - [info] [OpcUa-Client:6ee1b705.cfadd8] Client status: connecting
10 Apr 11:50:13 - [info] [OpcUa-Client:6ee1b705.cfadd8] async series - create session ...
10 Apr 11:50:13 - [info] [OpcUa-Client:6ee1b705.cfadd8] session active
10 Apr 11:50:13 - [info] [OpcUa-Client:6ee1b705.cfadd8] Client status: session active
10 Apr 11:50:13 - [info] [OpcUa-Client:6ee1b705.cfadd8] Action on input:events Item from Topic: ns=2;s=Channel2.Device1.COUNTER session Id: ns=1;g=8984097F-FA0C-487D-BBAA-0A6D4CDF3476
10 Apr 11:50:13 - [info] [OpcUa-Client:6ee1b705.cfadd8] subscribing events
10 Apr 11:50:13 - [info] [OpcUa-Client:6ee1b705.cfadd8] Subscription subscribed ID: 11
10 Apr 11:50:13 - [info] [OpcUa-Client:6ee1b705.cfadd8] Client status: subscribed
10 Apr 11:50:13 - [info] [OpcUa-Client:6ee1b705.cfadd8] Session subscriptionId: 11
10 Apr 11:50:13 - [info] [OpcUa-Client:6ee1b705.cfadd8] ns=2;s=Channel2.Device1.COUNTER samplingInterval 0
10 Apr 11:50:13 - [warn] [OpcUa-Client:6ee1b705.cfadd8] OpcUaClientNode: Monitoring Event: ns=2;s=Channel2.Device1.COUNTER by interval of 0 ms
10 Apr 11:50:13 - [info] [OpcUa-Client:6ee1b705.cfadd8] terminated monitored Event on ns=2;s=Channel2.Device1.COUNTER
10 Apr 11:50:13 - [error] [OpcUa-Client:6ee1b705.cfadd8] subscription.monitorEvent:Error: BadMonitoredItemFilterUnsupported (0x80440000)
warning : disconnection : closing pending sessions
10 Apr 11:50:14 - [info] [OpcUa-Client:6ee1b705.cfadd8] Client disconnected!
10 Apr 11:50:14 - [warn] [OpcUa-Client:6ee1b705.cfadd8] OpcUaClientNode: create Client ...
10 Apr 11:50:14 - [info] [OpcUa-Client:6ee1b705.cfadd8] Client status: create client
10 Apr 11:50:14 - [info] [OpcUa-Client:6ee1b705.cfadd8] async series - connecting
10 Apr 11:50:14 - [info] [OpcUa-Client:6ee1b705.cfadd8] Client status: connecting
10 Apr 11:50:14 - [info] [OpcUa-Client:6ee1b705.cfadd8] async series - create session ...
10 Apr 11:50:14 - [info] [OpcUa-Client:6ee1b705.cfadd8] session active
10 Apr 11:50:14 - [info] [OpcUa-Client:6ee1b705.cfadd8] Client status: session active
10 Apr 11:50:14 - [info] [OpcUa-Client:6ee1b705.cfadd8] Action on input:events Item from Topic: ns=2;s=Channel2.Device1.COUNTER session Id: ns=1;g=A8A71756-7D59-4622-BC78-B40E09FB9881
10 Apr 11:50:14 - [info] [OpcUa-Client:6ee1b705.cfadd8] subscribing events
10 Apr 11:50:14 - [info] [OpcUa-Client:6ee1b705.cfadd8] Client status: active subscribing
10 Apr 11:50:14 - [info] [OpcUa-Client:6ee1b705.cfadd8] Session subscriptionId: 11
10 Apr 11:50:14 - [info] [OpcUa-Client:6ee1b705.cfadd8] ns=2;s=Channel2.Device1.COUNTER samplingInterval 0
[... cut ...]
    at ClientMonitoredItem._monitor (C:\Users\John Egan\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua\lib\client\client_monitored_item.js:190:5)
    at Immediate.wait_for_subscription_and_monitor (C:\Users\John Egan\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\node_modules\node-opcua\lib\client\client_subscription.js:558:27)
    at runCallback (timers.js:666:20)
    at tryOnImmediate (timers.js:639:5)
    at processImmediate [as _immediateCallback] (timers.js:611:5)

not sure if this helps.

mikakaraila, 2017-04-10:
So you have browse node and event node. But this verbose seems to be interesting:
 by interval of 0 ms
Can you check how you have setup subscription node?

mikakaraila, 2017-04-10:
Another question: are you sending read request 1000ms interval?
10 Apr 11:50:12 - [info] [inject:COUNT] repeat = 1000
It will be better to use subscription.
Or if you send new subscription on every 1000ms it could cause this because all will be duplicate to first one...
OPC UA is actually designed to be very effective.

mikakaraila, 2017-04-10:
Yes node-opcua v0.0.64 is available from github. I have not checked npm
yet...

johnjohnguinness, 2017-04-10:
Hi,
Node red was working fine, then I start playing around with the timing of messages to MQTT client on my phone. I let the PC for a while when I came back Node red had stopped working and never recovered. I cant open it in the browser and it just crashes in the cmd prompt. I'm stuck:(

johnjohnguinness, 2017-04-10:
would you have a link to node-opcua v0.0.64?

mikakaraila, 2017-04-12:
https://github.com/node-opcua/node-opcua

mikakaraila, 2017-05-23:
Status?

mikakaraila, 2017-07-03:
Can be reopened if still valid.

closed 2017-07-03 by mikakaraila