mikakaraila/node-red-contrib-opcua #38: Can't connect to OPC UA server using endpoint security

Faerylnahr, 2017-04-11:
Hi,

I just tried to use a OPC UA client node in order to perform a subscription. The connection to the OPC UA server should be established with security policy "Basic128Rsa15 signed+encrypted". I can click "Update" when configuring the endpoint settings without getting any errors but when I try to connect to the server I get 
> 11 Apr 11:10:40 - [error] [OpcUa-Client:Subscription] Subscription OPC UA connection error:  End point must exist [...]
I found out that my endpoint configration wouldn't be saved correctly. It saves the endpoint URL and user credentials but not the value selected for "Sign" (in my case "Basic128Rsa15 signed+encrypted"). It always switchs back to "None" as security policy and this is (intentionally) not configured on the device. Thus no connection possible.

What am I missing?

Thanks in advance!

> node version: v6.9.5
> node-red version: v0.16.2

Faerylnahr, 2017-04-12:
After further research I found out, that the value of the property "sign" (-> node-input-sign) is completly ignored. Neither here
> 105-opcuaendpoint.html

`        defaults: {
            endpoint: {value: "", required: true},
            login: {value: false}
        },`

nor here
> 105-opcuaendpoint.js

`        this.endpoint = n.endpoint;
        this.login = n.login;`

Is this property intentionally ignored? In addition I didn't find any code snippet which might take the property value into account (e.g. when looking up possibly matching endpoints).

Thus, from my point of view, client connection using a security policy isn't yet supported. Am I right?

mikakaraila, 2017-04-12:
You put it also into node-opcua, let´s wait until Etienne comments...

mikakaraila, 2017-04-13:
Ack, just fixing this...

Faerylnahr, 2017-04-13:
Just opened a merge request and provided what I've changed so far :-) 
Thanks for your help 👍

mikakaraila, 2017-04-13:
Ops, I just made it... please test. And check if merge is still needed.

Faerylnahr, 2017-04-13:
[At this point](https://github.com/mikakaraila/node-red-contrib-opcua/commit/f406295e03822cd360c9328076d21cc511d9282f#diff-25ab3f180ba849e3daba62ff9e1e8c03R57) I wouldn't do the switch.
From my point of view it should be possible to select securityPolicy and securityMode separatly (as it is done e.g. in uaExport) from within the user interface (see here in [merge request](https://github.com/mikakaraila/node-red-contrib-opcua/pull/39/commits/8949d74b27df4e3580170650c3bbddf7c271d799#diff-a236f4d5dac89dcb6ac8696f24a5bbbaR24) and [here](https://github.com/mikakaraila/node-red-contrib-opcua/pull/39/files#diff-511a1a1fc0e55ef6dbf32eeb8cfd3506R27)).
The other changes look similar to my changes 👍

Faerylnahr, 2017-04-13:
If you won't merge my changes could you please have a look at [this line](https://github.com/mikakaraila/node-red-contrib-opcua/commit/83ccf7499b8f2a5deb94ba359096f4be92795271#diff-a236f4d5dac89dcb6ac8696f24a5bbbaR77)? Looks like the value is assigned twice.
[image]

mikakaraila, 2017-04-13:
Yes I will fix it on Monday. Already on vacation...

Faerylnahr, 2017-04-13:
Never mind :-) In the meantime I've forked your project again, fixed some typos and made some changes:
- SecurityPolicy and SecurityMode are configurable separatly
- OPCUAClient and OPCUABrowser now use the security settings
Looking forward to your update next week. Thanks so much for your help!
Now it's working! 👍

mikakaraila, 2017-04-18:
Merged.

closed 2017-04-18 by mikakaraila