mikakaraila/node-red-contrib-opcua #41: Write an Array?

brendonparker, 2017-04-20:
I'm not seeing how one would write to tag that holds an array of values?

Is this a gap, or am I just missing how to do this?

Best I can tell the `build_new_variant` would need to be modified to handle the case where `value` is an array.

A quick/naive implementation would look something like:
```javascript
module.exports.build_new_variant = function (opcua, datatype, value) {

    var nValue = new opcua.Variant({dataType: opcua.DataType.Float, value: 0.0});

    var arrVal;
    if(Array.isArray(value)){
        arrVal = value;
    }

    switch (datatype) {
        case "Float":
            nValue = new opcua.Variant({dataType: opcua.DataType.Float, value: arrVal || parseFloat(value)});
            break;
        case "Double":
            nValue = new opcua.Variant({dataType: opcua.DataType.Double, value: arrVal || parseFloat(value)});
            break;
        case "Int32":
            nValue = new opcua.Variant({dataType: opcua.DataType.Int32, value: arrVal || parseInt(value)});
            break;
        case "Int16":
            nValue = new opcua.Variant({dataType: opcua.DataType.Int16, value: arrVal || parseInt(value)});
            break;
        case "Int8":
            nValue = new opcua.Variant({dataType: opcua.DataType.Int8, value: arrVal || parseInt(value)});
            break;
        case "UInt32":
            nValue = new opcua.Variant({dataType: opcua.DataType.UInt32, value: arrVal || parseInt(value)});
            break;
        case "UInt16":
            nValue = new opcua.Variant({dataType: opcua.DataType.UInt16, value: arrVal || parseInt(value)});
            break;
        case "UInt8":
            nValue = new opcua.Variant({dataType: opcua.DataType.UInt8, value: arrVal || parseInt(value)});
            break;
        case "Boolean":
            if (value && value !== "false") {
                nValue = new opcua.Variant({dataType: opcua.DataType.Boolean, value: true});
            }
            else {
                nValue = new opcua.Variant({dataType: opcua.DataType.Boolean, value: false});
            }
            break;
        case "String":
            nValue = new opcua.Variant({dataType: opcua.DataType.String, value: arrVal || value});
            break;
        default:
            nValue = new opcua.Variant({dataType: opcua.DataType.BaseDataType, value: arrVal || value});
            break;
    }

    return nValue;
};
```

mikakaraila, 2017-04-20:
Not yet implemented into this...
I have done it for our server. Just have to find time to do it :-)

mikakaraila, 2017-07-04:
Here are some tips how to implement this:
https://github.com/node-opcua/node-opcua/issues/337
I have done this with Typed Arrays as it must be done. 
This is more than one day work to implement...

brendonparker, 2018-03-25:
> OK is this still needed? I can do it with our OPC UA Server. It supports
> 1,2 and 3-dimensional Arrays.

Not needed by me. No longer using node-red as an approach. Thanks!!

closed 2018-03-27 by mikakaraila

mikakaraila, 2021-01-27:
1-dimensional Array support implemented now.