mikakaraila/node-red-contrib-opcua #42: Link OPC UA browse to OPC UA Items

sanlop73, 2017-05-03:
Hi:

I am browsing all items in an specific path (ns=2;s=[Default]PLC-Test)  and  actually i am getting 6 items.
My problem is how to pass the items path found to set a read or subscription.  
In conclusion the goal is to read all tags defined in my Ignition server and create all Read/Subscription items automatically.
Could you please provide an example?

Thanks in advance
[image]

mikakaraila, 2017-05-03:
Yep,
I think it should be on my laptop. I will look and add it here tomorrow...

mikakaraila, 2017-07-04:
Sorry I forget that, but here it the flow that will do the trick:
[{"id":"c8b446ba.bfa058","type":"OpcUa-Browser","z":"76fbcb09.74594c","endpoint":"acd2f698.88bd28","item":"","datatype":"","topic":"ns=5;s=85/0:Simulation","items":[],"x":460.5000457763672,"y":37.79999542236328,"wires":[["83243287.79201"]]},{"id":"481f49fe.55a3a8","type":"inject","z":"76fbcb09.74594c","name":"Browse","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":78.5,"y":35.79999542236328,"wires":[["3eb58696.791f7a"]]},{"id":"83243287.79201","type":"function","z":"76fbcb09.74594c","name":"Decode & filter","func":"var items = msg.payload;\nfor (var i=0; i<items.length; i++) {\n    var item = items[i];\n\tvar ref = item.item;\n\tvar nodeClass = ref.$nodeClass;\n\tvar typeDef = ref.typeDefinition;\n\tvar bname = ref.browseName;\n\tvar ns=bname.namespaceIndex;\n\tvar name=bname.name;\n\t// Select only want namespace variables\n\tif (ns==5) {\n\t    var newmsg={};\n\t\tnewmsg.topic = \"ns=\"+ns+\";s=\"+name;\n\t\tnewmsg.payload=\"\";\n\t\tnode.send(newmsg);\n\t}\n}","outputs":1,"noerr":0,"x":260.5000457763672,"y":143,"wires":[["de80c421.108fa8"]]},{"id":"de80c421.108fa8","type":"OpcUa-Client","z":"76fbcb09.74594c","endpoint":"acd2f698.88bd28","action":"subscribe","time":"1","timeUnit":"s","name":"","x":451.49998474121094,"y":142.20001220703125,"wires":[["6b267258.eb8e9c"]]},{"id":"6b267258.eb8e9c","type":"debug","z":"76fbcb09.74594c","name":"","active":true,"console":"false","complete":"payload","x":632.2000274658203,"y":143.20001220703125,"wires":[]},{"id":"3eb58696.791f7a","type":"function","z":"76fbcb09.74594c","name":"Set browse address","func":"msg.topic='ns=5;s=85/0:Simulation';\nmsg.actiontype='browse';\nreturn msg;","outputs":1,"noerr":0,"x":251.50001525878906,"y":36,"wires":[["c8b446ba.bfa058"]]},{"id":"acd2f698.88bd28","type":"OpcUa-Endpoint","z":"7e58ca8c.d78944","endpoint":"opc.tcp://localhost:53530/OPCUA/SimulationServer","login":false}]

closed 2017-07-13 by mikakaraila