mikakaraila/node-red-contrib-opcua #43: Auto generated Server Status

philip-very, 2017-05-16:
Hello, 
im currenty working on my Masterthesis. For the implementation of my project im using Node-Red.
I've installed Node-red and Node-red-contrib-opcua on my Windows Computer and started my project. Everything worked fine so far. So I have to install my flows on a Linux server (Ubuntu 16.04.2). As far as I can tell the installation of Node-red and the other additional Modules worked fine and I'm able to start node-red. But while loading Node-red, i get the following Error-Messages:

16 May 12:55:20 - [warn] [OpcUa-Browser] Error: Cannot find module '_generated_/_auto_generated_ServerStatus'
16 May 12:55:20 - [warn] [OpcUa-Client] Error: Cannot find module '_generated_/_auto_generated_ServerStatus'
16 May 12:55:20 - [warn] [OpcUa-Event] Error: Cannot find module '_generated_/_auto_generated_ServerStatus'
16 May 12:55:20 - [warn] [OpcUa-Item] Error: Cannot find module '_generated_/_auto_generated_ServerStatus'
16 May 12:55:20 - [warn] [OpcUa-Server] Error: Cannot find module '_generated_/_auto_generated_ServerStatus'

I'v already checked for the latest version and I also tried to reinstall it. I rally hope you can help me, with my issue.

Kind regards
Philip

mikakaraila, 2017-05-17:
Simple, I should add into my code one line to load needed xml file into the server.
I will look later today...

mikakaraila, 2017-05-18:
Hmm, I just checked that server loads all needed XML files:
var xmlFiles = [path.join(__dirname, 'public/vendor/opc-foundation/xml/Opc.Ua.NodeSet2.xml'), // This one contains Server Diagnostic definitions
            path.join(__dirname, 'public/vendor/opc-foundation/xml/Opc.ISA95.NodeSet2.xml')];

Check that under node_modules/node-opcua/_generated_ are files like:
_auto_generated_ServerStatus.js

There must be something in the environment or installation because I can browse server diagnostics:
[image]

mikakaraila, 2017-07-04:
Installation problem? -> Closed.

closed 2017-07-04 by mikakaraila