mikakaraila/node-red-contrib-opcua #44: Can't write Byte Type

mscbpi, 2017-05-18:
Hi,

I am currently reviewing the node to interact with Beckhoff PLC.

I can read and write successfully boolean, int16.

I can read successfully Byte type (and datatype is indeed reported as byte).

I cannot write Byte type, no errors but the value never changes in the node.

Works from other OPC-UA client.

Thanks for help!

Martin

mscbpi, 2017-05-18:
I wiresharked and can notice that indeed, value is at null when the request goes on the network. [image] 

whereas it's OK when it comes to an Int16:
[image]

mscbpi, 2017-05-18:
actually it looks like it's missing there
https://github.com/mikakaraila/node-red-contrib-opcua/blob/master/opcua/opcua-basics.js

`module.exports.build_new_variant = function (opcua, datatype, value) {

    var nValue = new opcua.Variant({dataType: opcua.DataType.Float, value: 0.0});

    switch (datatype) {
        case "Float":
            nValue = new opcua.Variant({dataType: opcua.DataType.Float, value: parseFloat(value)});
            break;
        case "Double":
            nValue = new opcua.Variant({dataType: opcua.DataType.Double, value: parseFloat(value)});
            break;
        case "Int32":
            nValue = new opcua.Variant({dataType: opcua.DataType.Int32, value: parseInt(value)});
            break;
        case "Int16":
            nValue = new opcua.Variant({dataType: opcua.DataType.Int16, value: parseInt(value)});
            break;
        case "Int8":
            nValue = new opcua.Variant({dataType: opcua.DataType.Int8, value: parseInt(value)});
            break;
        case "UInt32":
            nValue = new opcua.Variant({dataType: opcua.DataType.UInt32, value: parseInt(value)});
            break;
        case "UInt16":
            nValue = new opcua.Variant({dataType: opcua.DataType.UInt16, value: parseInt(value)});
            break;
        case "UInt8":
            nValue = new opcua.Variant({dataType: opcua.DataType.UInt8, value: parseInt(value)});
            break;
        case "Boolean":
            if (value && value !== "false") {
                nValue = new opcua.Variant({dataType: opcua.DataType.Boolean, value: true});
            }
            else {
                nValue = new opcua.Variant({dataType: opcua.DataType.Boolean, value: false});
            }
            break;
        case "String":
            nValue = new opcua.Variant({dataType: opcua.DataType.String, value: value});
            break;
        default:
            nValue = new opcua.Variant({dataType: opcua.DataType.BaseDataType, value: value});
            break;
    }

    return nValue;
};`

mikakaraila, 2017-05-18:
Yes I know, do you want add it? I can do it later today...

mikakaraila, 2017-05-23:
Fix merged

closed 2017-05-23 by mikakaraila