mikakaraila/node-red-contrib-opcua #50: Extention for browse results?

Faerylnahr, 2017-06-20:
Hi,
if I use the OPCua browser I currently get something like
`
{
   "referenceTypeId": "ns=0;i=47",
   "isForward": "true",
   "nodeId": "ns=4;s=MY.PATH",
   "browseName": {
      "namespaceIndex": "4",
      "name": "MY_VARIABLE"
   },
   "displayName": {
      "text": "MY_VARIABLE",
      "locale": ""
   },
   "nodeClass": "Variable",
   "typeDefinition": "ns=0;i=63"
}
`
I'm missing information about datatype and value and I don't know if it's an "issue" of this project or of node-opcua. I just need some additional information, something like the following
`
{
   ...
   "value": {
      "value": 0,
      "datatype": "Int16"
   },
   ...
}
`
That would suit all my needs :-)
Kind regards,
Stefi

mikakaraila, 2017-06-21:
I supose you have same kind of needs as in issue #31 Have looked solution that I provided there?

Faerylnahr, 2017-06-21:
Thanks :-) I'll have a look on it tomorrow!

mikakaraila, 2017-07-03:
Any comments or wishes?

Faerylnahr, 2017-07-03:
Sorry for my late response and thx for your help. I've tested the solution you've provided and looks good for me :-)

mikakaraila, 2017-07-04:
Ok, I will close this one and look later if I have time to add own node to get browse results without additional function nodes.

closed 2017-07-04 by mikakaraila