mikakaraila/node-red-contrib-opcua #51: Unsubscibe?

msuing, 2017-06-21:
Hi,

I'm succesfully using the Client to subscribe to a server on a PLC. Now I would like to unsubscribe from a topic or close the session of the Client. I can't find how to do this. Is it possible with the current version?

Regards,
 Markus

PS: Thanx for the great work.

mikakaraila, 2017-06-21:
No I have not needed it by myself. I will look this later or is this "hot" topic?

msuing, 2017-06-22:
No, it's not urgent since I'm doing a POC project. But it would be needed for everyday use of my solution since the OPC server I have to work with sometimes shows unwanted behaviour if you don't properly close sessions.

mikakaraila, 2017-06-25:
Ack, I will be back after a week from business trip. I will look this then...

mikakaraila, 2017-07-03:
Implemented now, I will do some testing and push it to github.

mikakaraila, 2017-07-03:
New release & example available in github, please test.

closed 2017-07-13 by mikakaraila