mikakaraila/node-red-contrib-opcua #52: 0x891e307 Fatal error in , line 0 - unreachable code

scsi1042, 2017-06-22:
I am running node-red on a Yocto Linux System (Processor: Intel Quark® x1020 400 MHz).

My application includes a S7-Node (node-red-contrib-s7) to read out values of a PLC (this is currently a test value which changes every 15sec from true to false and vice versa).
An OPCUA-Server-Node (node-red-contrib-opcua) is initialized with the creation of a Boolean-variable and creates an OPCUA-Server to further read out the Data via OPC protocol.
An OPC-Client writes the incoming true/false values from the PLC to the OPCUA-Server.

After an unspecified time period (approx. 20-40 min) node-red crashes.

```
root@iot2000:/usr/lib/node_modules# node ./node-red/red

Welcome to Node-RED
===================

22 Jun 07:53:37 - [info] Node-RED version: v0.15.2
22 Jun 07:53:37 - [info] Node.js  version: v6.9.2
22 Jun 07:53:37 - [info] Linux 4.4.18-yocto-standard ia32 LE
22 Jun 07:53:52 - [info] Loading palette nodes
22 Jun 07:55:13 - [info] Dashboard version 2.1.0 started at /ui
22 Jun 07:55:14 - [warn] ------------------------------------------------------
22 Jun 07:55:14 - [warn] [rpi-gpio] Info : Ignoring Raspberry Pi specific node
22 Jun 07:55:14 - [warn] ------------------------------------------------------
22 Jun 07:55:14 - [info] Settings file  : /home/root/.node-red/settings.js
22 Jun 07:55:14 - [info] User directory : /home/root/.node-red
22 Jun 07:55:14 - [info] Flows file     : /home/root/.node-red/flows_iot2000.json
22 Jun 07:55:15 - [info] Server now running at http://127.0.0.1:1880/
22 Jun 07:55:15 - [debug] red/runtime/nodes/credentials.load : no user key present
22 Jun 07:55:15 - [debug] red/runtime/nodes/credentials.load : using default key
22 Jun 07:55:15 - [info] Starting flows
22 Jun 07:55:35 - [info] Started flows
22 Jun 07:55:36 - [info] [debug:9e445757.340408]
{ opcuaCommand: 'addVariable' }
22 Jun 07:56:39 - [info] [debug:7f92b9a0.ab1b8] true
22 Jun 07:56:39 - [error] [OpcUa-Client:WRITE ] WRITE  OPC UA connection error: connect ECONNREFUSED 127.0.0.1:53880
22 Jun 07:56:39 - [info] [OpcUa-Client:WRITE ] TESTLOG: RESET CLIENT NACH CONNECTION ERROR
 Bad Session in  _on_ActivateSessionRequest 030691667bf38f6e20b540681a3c7a3e
22 Jun 07:57:29 - [error] [s7 endpoint:ac2f8250.edec] Failure (Bad values)
22 Jun 07:57:31 - [info] [debug:7f92b9a0.ab1b8] false
22 Jun 07:57:37 - [info] [debug:7f92b9a0.ab1b8] true
 Timeout .... waiting for response for  OpenSecureChannelRequest { /*RequestHeader*/
 authenticationToken           /* NodeId                           */: ns=0;i=0
 timeStamp                     /* UtcTime                          */: 2017-06-22T07:56:39.362Z
 requestHandle                 /* IntegerId                        */: 1               0x1
 returnDiagnostics             /* UInt32                           */: 0               0x0
 auditEntryId                  /* UAString                         */: null
 timeoutHint                   /* UInt32                           */: 0               0x0
 additionalHeader              /* ExtensionObject                  */: null
};
 Timeout .... waiting for response for  CloseSecureChannelRequest { /*RequestHeader*/
 authenticationToken           /* NodeId                           */: ns=0;i=0
 timeStamp                     /* UtcTime                          */: 2017-06-22T07:56:39.667Z
 requestHandle                 /* IntegerId                        */: 2               0x2
 returnDiagnostics             /* UInt32                           */: 0               0x0
 auditEntryId                  /* UAString                         */:
 timeoutHint                   /* UInt32                           */: 0               0x0
 additionalHeader              /* ExtensionObject                  */: null
};
22 Jun 07:57:43 - [error] [s7 endpoint:ac2f8250.edec] Failure (Bad values)
22 Jun 07:57:43 - [error] [OpcUa-Client:WRITE ] WRITE  OPC UA connection error: Transaction has timed out
22 Jun 07:57:43 - [info] [OpcUa-Client:WRITE ] TESTLOG: RESET CLIENT NACH CONNECTION ERROR
22 Jun 07:57:56 - [info] [debug:7f92b9a0.ab1b8] false
22 Jun 07:58:06 - [info] [debug:9e445757.340408]
{ opcuaCommand: 'addVariable' }
22 Jun 07:58:08 - [info] [debug:7f92b9a0.ab1b8] true
22 Jun 07:58:23 - [info] [debug:7f92b9a0.ab1b8] false
22 Jun 07:58:39 - [info] [debug:7f92b9a0.ab1b8] true
22 Jun 07:58:55 - [info] [debug:7f92b9a0.ab1b8] false
22 Jun 07:59:10 - [info] [debug:7f92b9a0.ab1b8] true
...
...
...
22 Jun 08:17:57 - [info] [debug:7f92b9a0.ab1b8] true
22 Jun 08:18:13 - [info] [debug:7f92b9a0.ab1b8] false
22 Jun 08:18:28 - [info] [debug:7f92b9a0.ab1b8] true
22 Jun 08:18:44 - [info] [debug:7f92b9a0.ab1b8] false
22 Jun 08:18:59 - [info] [debug:7f92b9a0.ab1b8] true

#
# Fatal error in , line 0
# unreachable code
#

==== C stack trace ===============================

 1: V8_Fatal
 2: 0x891e307
 3: v8::internal::LCodeGenBase::GenerateBody()
 4: v8::internal::LCodeGen::GenerateCode()
 5: v8::internal::LChunk::Codegen()
 6: v8::internal::OptimizedCompileJob::GenerateCode()
 7: v8::internal::Compiler::FinalizeOptimizedCompileJob(v8::internal::OptimizedCompileJob*)
 8: v8::internal::OptimizingCompileDispatcher::InstallOptimizedFunctions()
 9: v8::internal::StackGuard::HandleInterrupts()
10: v8::internal::Runtime_StackGuard(int, v8::internal::Object**, v8::internal::Isolate*)
11: 0x3dd0a07e
12: 0x3c43c8df
13: 0x3dd1fd9e
14: 0x3dd1bd23
15: 0x85da06a
16: v8::internal::Execution::Call(v8::internal::Isolate*, v8::internal::Handle<v8::internal::Object>, v8::internal::Handle<v8::internal::Object>, int, v8::internal::Handle<v8::internal::Object>*)
17: v8::Function::Call(v8::Local<v8::Context>, v8::Local<v8::Value>, int, v8::Local<v8::Value>*)
18: v8::Function::Call(v8::Local<v8::Value>, int, v8::Local<v8::Value>*)
19: node::AsyncWrap::MakeCallback(v8::Local<v8::Function>, int, v8::Local<v8::Value>*)
20: 0x89a5cbb
21: 0x89f1d30
22: uv_run
23: node::Start(int, char**)
24: main
25: __libc_start_main
Illegal instruction
```
Can anybody help?

mikakaraila, 2017-06-22:
Can you run those separate first? I mean s7 flow works correctly and then OPCUA is working in own flow normally? After it if you combine them error occurs?

scsi1042, 2017-06-22:
When I run only s7 flow it works fine.
From the previous post, I deleted the s7 flow and added a input to simulate the 15sec true/false signals. It ended in the same error:

```
Welcome to Node-RED
===================

22 Jun 13:02:46 - [info] Node-RED version: v0.15.2
22 Jun 13:02:46 - [info] Node.js  version: v6.9.2
22 Jun 13:02:46 - [info] Linux 4.4.18-yocto-standard ia32 LE
22 Jun 13:03:01 - [info] Loading palette nodes
22 Jun 13:04:24 - [info] Dashboard version 2.1.0 started at /ui
22 Jun 13:04:25 - [warn] ------------------------------------------------------
22 Jun 13:04:25 - [warn] [rpi-gpio] Info : Ignoring Raspberry Pi specific node
22 Jun 13:04:25 - [warn] ------------------------------------------------------
22 Jun 13:04:25 - [info] Settings file  : /home/root/.node-red/settings.js
22 Jun 13:04:25 - [info] User directory : /home/root/.node-red
22 Jun 13:04:25 - [info] Flows file     : /home/root/.node-red/flows_iot2000.jso                                                     n
22 Jun 13:04:26 - [info] Server now running at http://127.0.0.1:1880/
22 Jun 13:04:26 - [debug] red/runtime/nodes/credentials.load : no user key prese                                                     nt
22 Jun 13:04:26 - [debug] red/runtime/nodes/credentials.load : using default key
22 Jun 13:04:26 - [info] Starting flows
22 Jun 13:04:44 - [info] Started flows
22 Jun 13:04:44 - [info] [debug:9e508b3b.d9eee]
{ opcuaCommand: 'addVariable' }
22 Jun 13:05:47 - [info] [debug:e6d55ecc.6283c8] true
22 Jun 13:05:47 - [error] [OpcUa-Client:WRITE ] WRITE  OPC UA connection error:                                                      connect ECONNREFUSED 127.0.0.1:53880
22 Jun 13:05:47 - [info] [OpcUa-Client:WRITE ] TESTLOG: RESET CLIENT NACH CONNEC                                                     TION ERROR
22 Jun 13:06:25 - [info] [debug:e6d55ecc.6283c8] false
22 Jun 13:06:25 - [info] [debug:e6d55ecc.6283c8] true
 Bad Session in  _on_ActivateSessionRequest 308375d91a15ad1afb14e90f867da36a
22 Jun 13:06:40 - [info] [debug:e6d55ecc.6283c8] false
 Bad Session in  _on_ActivateSessionRequest a59b318809dbaa4beecde0092dff2a2a
 Timeout .... waiting for response for  OpenSecureChannelRequest { /*RequestHeader*/
 authenticationToken           /* NodeId                           */: ns=0;i=0
 timeStamp                     /* UtcTime                          */: 2017-06-22T13:05:47.291Z
 requestHandle                 /* IntegerId                        */: 1               0x1
 returnDiagnostics             /* UInt32                           */: 0               0x0
 auditEntryId                  /* UAString                         */: null
 timeoutHint                   /* UInt32                           */: 0               0x0
 additionalHeader              /* ExtensionObject                  */: null
};
 Timeout .... waiting for response for  CloseSecureChannelRequest { /*RequestHeader*/
 authenticationToken           /* NodeId                           */: ns=0;i=0
 timeStamp                     /* UtcTime                          */: 2017-06-22T13:05:47.598Z
 requestHandle                 /* IntegerId                        */: 2               0x2
 returnDiagnostics             /* UInt32                           */: 0               0x0
 auditEntryId                  /* UAString                         */:
 timeoutHint                   /* UInt32                           */: 0               0x0
 additionalHeader              /* ExtensionObject                  */: null
};
22 Jun 13:06:47 - [error] [OpcUa-Client:WRITE ] WRITE  OPC UA connection error: Transaction has timed out
22 Jun 13:06:47 - [info] [OpcUa-Client:WRITE ] TESTLOG: RESET CLIENT NACH CONNECTION ERROR
22 Jun 13:06:57 - [info] [debug:e6d55ecc.6283c8] true
warning : disconnection : closing pending sessions
22 Jun 13:07:13 - [info] [debug:e6d55ecc.6283c8] false
22 Jun 13:07:14 - [info] [debug:9e508b3b.d9eee]
{ opcuaCommand: 'addVariable' }
22 Jun 13:07:27 - [info] [debug:e6d55ecc.6283c8] true
22 Jun 13:07:42 - [info] [debug:e6d55ecc.6283c8] false
22 Jun 13:07:57 - [info] [debug:e6d55ecc.6283c8] true
22 Jun 13:08:12 - [info] [debug:e6d55ecc.6283c8] false
...
...
...
22 Jun 13:19:13 - [info] [debug:e6d55ecc.6283c8] false
22 Jun 13:19:28 - [info] [debug:e6d55ecc.6283c8] true
22 Jun 13:19:43 - [info] [debug:e6d55ecc.6283c8] false

#
[... 62 lines cut ...]
                                callback();
                            }
                            else {
                                set_node_status_to("session error");
                                callback(err);
                            }
                        });
                    } catch (err) {
                        callback(err);
                    }
                }
            ], function (err) {
                if (err) {
                    node.error(node.name + " OPC UA connection error: " + err.message);
                    verbose_log(err);
                    node.session = null;
                    close_opcua_client(set_node_status_to("connection error"));

                    verbose_log("RESET OPC UA CLIENT nach CONNECTION ERROR", opcuaEndpoint.endpoint);
                    node.log("RESET OPC UA CLIENT nach CONNECTION ERROR", opcuaEndpoint.endpoint)
                    reset_opcua_client(connect_opcua_client);
                }
            });
        }
```

mikakaraila, 2017-06-22:
Hmm, it would be better to delay client.
Now reset will close session and cause the crash most probably...

scsi1042, 2017-06-23:
I reedited the changes in the function "connect_opcua_client". As a result the client doesn't connect and results in "Connection Error". When I continue to run the program like this, the v8_fatal error is still thrown after about 20 min.

mikakaraila, 2017-06-25:
Hmm, same happens also in Windows environment? I don´t have time to debug this during next week.

scsi1042, 2017-06-26:
When I run the same code on a windows VM, I get this error:
"AssertionError: fs.existsSync(this.certificateFile), "Certificate file must exist:" + this.certificateFile".

Since I need my application to run on my linux system, I have not examined further this error, yet.

mikakaraila, 2017-07-03:
Can you provide small test flow to study this?

scsi1042, 2017-07-04:
The OPC UA Server Node and 3 input nodes which add new Variables after a delay of 150 sec, which is necessary to wait until server is initialized. I have not tested with OPC UA Server node only.

`[{"id":"ab443549.fa54b","type":"debug","z":"d21a72c9.d644","name":"","active":true,"console":"true","complete":"payload","x":630,"y":120,"wires":[]},{"id":"aa8bec06.163e98","type":"OpcUa-Server","z":"d21a72c9.d644","port":"53880","name":"OPC-UA Server","endpoint":"UA/SimpleNodeRedServer","x":680,"y":280,"wires":[["ab443549.fa54b"]]},{"id":"9c2ffd0d.92992","type":"inject","z":"d21a72c9.d644","name":"New variable (Vereinzler)","topic":"ns=4;s=Vereinzler;datatype=Float","payload":"{ \"opcuaCommand\": \"addVariable\" }","payloadType":"json","repeat":"","crontab":"","once":true,"x":170,"y":60,"wires":[["ab443549.fa54b","264edb2a.f2360c"]]},{"id":"5f6ada63.894abc","type":"inject","z":"d21a72c9.d644","name":"New variable (Entdeckeln)","topic":"ns=4;s=Entdeckeln;datatype=Float","payload":"{ \"opcuaCommand\": \"addVariable\" }","payloadType":"json","repeat":"","crontab":"","once":true,"x":170,"y":120,"wires":[["ab443549.fa54b","fecfd61a.ec6f1"]]},{"id":"d5c0fcfe.b3424","type":"inject","z":"d21a72c9.d644","name":"New variable (Entleeren)","topic":"ns=4;s=Entleeren;datatype=Float","payload":"{ \"opcuaCommand\": \"addVariable\" }","payloadType":"json","repeat":"","crontab":"","once":true,"x":170,"y":180,"wires":[["ab443549.fa54b","970b5dd1.320e7"]]},{"id":"264edb2a.f2360c","type":"delay","z":"d21a72c9.d644","name":"","pauseType":"delay","timeout":"150","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":437,"y":254,"wires":[["aa8bec06.163e98"]]},{"id":"fecfd61a.ec6f1","type":"delay","z":"d21a72c9.d644","name":"","pauseType":"delay","timeout":"150","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":439,"y":296,"wires":[["aa8bec06.163e98"]]},{"id":"970b5dd1.320e7","type":"delay","z":"d21a72c9.d644","name":"","pauseType":"delay","timeout":"150","timeoutUnits":"seconds","rate":"1","nbRateUnits":"1","rateUnits":"second","randomFirst":"1","randomLast":"5","randomUnits":"seconds","drop":false,"x":437,"y":342,"wires":[["aa8bec06.163e98"]]}]`

I also learned that starting node-red with `node --nocrankshaft red` deactivates the optimizing controller. By doing this there are no crashs anymore, but unfortunately the connection between Server and an external client disconnects periodically every few seconds. And when I run the whole application including a s7-node, these values are not read continuously nor are any errors (e.g. bad values) displayed.

In the last days I implemented my application without node-red. This isn't as convenient as the node-red dashboard, but works so far.

mikakaraila, 2017-07-04:
Ack, this sounds like there is something with node.js and the hardware if the optimizer is not working.

closed 2017-07-04 by mikakaraila