mikakaraila/node-red-contrib-opcua #53: Connection Issue

Calvin93, 2017-06-30:
Hi, I have the following Problem with the OPC UA Client:

Everytime when i'm connecting my OPC UA Client with the Server, the connection is getting lost after a few seconds. In the console i get the following warning:

30 Jun 13:25:20 - [info] Starting flows
[BaseClient:connect] Connecting to IoTF with host : ssl://cje5yu.messaging.internetofthings.ibmcloud.com:8883
30 Jun 13:25:20 - [info] Started flows
[BaseClient:onClose] Connection was closed.
[BaseClient:disconnect] Disconnected from the client.
[GatewayClient:connect] GatewayClient Connected
 message was
undefined
 message was 2: { /*OpenSecureChannelRequest*/
 requestHeader                 /* RequestHeader                    */: {
   authenticationToken         /* NodeId                           */: ns=0;i=0
   timeStamp                   /* UtcTime                          */: 2017-06-30T11:25:20.630Z
   requestHandle               /* IntegerId                        */: 1               0x1
   returnDiagnostics           /* UInt32                           */: 0               0x0
   auditEntryId                /* UAString                         */: null
   timeoutHint                 /* UInt32                           */: 0               0x0
   additionalHeader            /* ExtensionObject                  */: null
 }
 clientProtocolVersion         /* UInt32                           */: 0               0x0
 requestType                   /* SecurityTokenRequestType         */: ISSUE ( 0)
 securityMode                  /* MessageSecurityMode              */: NONE ( 1)
 clientNonce                   /* ByteString                       */: null
 requestedLifetime             /* UInt32                           */: 600000               0x927c0
};
nodeStr=ns=3;s="WerkPos1AVorh"
30 Jun 13:25:38 - [info] [filter:6c8aaadc.1cc564] node-red-filter: message dropped (not matching the 'WerkstInGreifer' regular expression)
30 Jun 13:25:38 - [info] [filter:72441345.a65cfc] node-red-filter: message dropped (not matching the 'WerksFarbe_' regular expression)
30 Jun 13:25:38 - [info] [filter:242ba887.2510b8] node-red-filter: message dropped (not matching the 'sHlftxt' regular expression)
 Timeout .... waiting for response for  OpenSecureChannelRequest { /*RequestHeader*/
 authenticationToken           /* NodeId                           */: ns=0;i=0
 timeStamp                     /* UtcTime                          */: 2017-06-30T11:25:20.630Z
 requestHandle                 /* IntegerId                        */: 1               0x1
 returnDiagnostics             /* UInt32                           */: 0               0x0
 auditEntryId                  /* UAString                         */: null
 timeoutHint                   /* UInt32                           */: 0               0x0
 additionalHeader              /* ExtensionObject                  */: null
};
30 Jun 13:26:20 - [error] [OpcUa-Client:a9eae1b0.136b]  OPC UA connection error: Transaction has timed out
warning : disconnection : closing pending sessions
 warning : ClientSessionKeepAliveManager#ping_server   ServiceResult is BadSessionIdInvalid (0x80250000)

Hopefully anyone have an idea to solve this Problem.

mikakaraila, 2017-07-03:
Have you tested with some other client? Do you have ping statistics? 
VPN/firewall can slow down connection....

mikakaraila, 2017-07-13:
Is this your endpoint: ssl://cje5yu.messaging.internetofthings.ibmcloud.com:8883 ?
I suppose it is incorrect... have you tested server with some other client like UA Expert?

mikakaraila, 2017-07-27:
Solved? You can reopen if still existing...

closed 2017-07-27 by mikakaraila