mikakaraila/node-red-contrib-opcua #54: Problem: connection with credentials

TillBu, 2017-07-05:
Hey
For a university project i try to connect a opc client in node red to a molding machine with an OPC UA server. 

The connection is with no  SecurityPolicy and  SecurityMode but with credentials.
The Connection works with other clients (UA Expert) but node-red opcua returns:

Error: Cannot find ANONYMOUS user token policy in end point description"

Shouldn't there be a username token policy when i use credentials?

i´m very thankful for some help!

mikakaraila, 2017-07-05:
Did you activate credentials & filled user/passwd? 
I tested with Prosys OPC UA simulation server & user/passwd and everything is working.
[image]

allatrue, 2017-07-06:
There was once a problem, solved with #17. Do you have the latest version?

mikakaraila, 2018-03-17:
I suppose this works now with node-opcua v0.2.2.

closed 2018-03-17 by mikakaraila

TillBu, 2018-04-04:
Hello
Did not have to deal with this problem for a while, but now it is very current again.
As you posted i hoped very much it would be solved with the new version (im running 0.2.3 ) but the problem persists.
I will try to include some more information and see if i can help to find the reason.

the Endpoint, username and password are cofirmed working on UAExpert

[image]

i noticed that i get 2 different error messages even though the nodes they are based on the same endpoint. Also the Identity token should be of the Type UserName.

maybe the log from a connection attempt with another client who exposes the server policy can help

[image]

here the conection failed with: unsupportet Security policy

Thank you for your support, i will try looking at the code but im not very fluent.

Sincerly
Till

COPS95, 2018-10-22:
can you solve it? actually i'm trying to connect Machine Arburg with node-red.
But i can´t solve this issue, so i hope your answer .

Thankss!!

TillBu, 2018-10-22:
Hello,

I did not solve the problem. I ended up writing my own specific client based on the OPC Foundation .Net Stack.
I guess(!) the problem is that Arburg does not supply a certificate on thier OPC Servers and thus does not fulfill the OPC-UA Specification. 

if you find a solution, please let me know.

best regards