mikakaraila/node-red-contrib-opcua #55: Clients are unable to modify items on node-red OPC UA server

Vomimu, 2017-07-12:
Hello,

in a node red flow I added some items to a node-red OPC UA server using the 'addVariable' command.

My problem is that  OPC UA clients are not able to modify these items. It seams that the missing setter for the item value in '104-opcuaserver.js' is the reason for this  behavier. The following code resolves the issue:

```javascript
	...
	var opcuaBasics = require('./opcua-basics');
	...
	addressSpace.addVariable({
		componentOf: addressSpace.findNode(parentFolder.nodeId),
		nodeId: name,
		browseName: browseName, // or displayName
		dataType: opcuaDataType,
		value: {
			get: function() {
				return new opcua.Variant({dataType: opcuaDataType, value: variables[browseName]})
			},
			set: function (variant) {
				variables[browseName] = opcuaBasics.build_new_value_by_datatype(variant.dataType, variant.value);
				return opcua.StatusCodes.Good;
			}
		}
	});
	...
```
Best regards,
Volker

closed 2017-07-13 by mikakaraila

Vomimu, 2017-07-13:
Hi Mika. Thanks for adding the setter function. But as 'opcuaBasics' is not defined in '104-opcuaserver.js' the code doesn't work. Could you please also add the following line to the code:
```javascript
var opcuaBasics = require('./opcua-basics');
```
Thanks
Volker

mikakaraila, 2017-07-13:
Ack, fixed missing line. Rainy day. Enough time to fix code :-)
Mfg,
Meta

commit 1f969fde (2017-07-13): Fixed #55 missing line.