mikakaraila/node-red-contrib-opcua #58: subscriptions not working when injected

nospam2000, 2017-07-26:
Last week I saw multiple times, that there have been up to 177 active OPC UA connections to my test PLC (that's its limit) and none of them was working. The reason was, that I started around 200 item subscriptions before the OPC connection was established, it was in state "connecting".
This happens when you use the flag "inject once at start" of the 'inject' node to trigger the subscriptions to start the flow automatically.

I get the following traces:
[info] Settings file  : /data/settings.js
[info] User directory : /data
[info] Flows file     : /data/flows.json
[info] Server now running at http://127.0.0.1:1880/
[info] Starting flows
[info] [OpcUa-Client:OPC UA Subscribe] [object Object]
**[warn] [OpcUa-Client:OPC UA Subscribe] OPC UA Subscribe: create Client ...**
[info] [OpcUa-Client:OPC UA Subscribe] [object Object]
**[info] [OpcUa-Client:OPC UA Subscribe] Client status: create client**
**[info] [OpcUa-Client:OPC UA Subscribe] async series - connecting 
[info] [OpcUa-Client:OPC UA Subscribe] Client status: connecting**
[info] [inject:cyclic processing] repeat = 60000
[info] Started flows
**[warn] [OpcUa-Client:OPC UA Subscribe] OPC UA Subscribe: can't work without OPC UA Session**
[warn] [OpcUa-Client:OPC UA Subscribe] OPC UA Subscribe: can't work without OPC UA Session
[warn] [OpcUa-Client:OPC UA Subscribe] OPC UA Subscribe: can't work without OPC UA Session
[info] [OpcUa-Client:OPC UA Subscribe] Client disconnected!
[warn] [OpcUa-Client:OPC UA Subscribe] OPC UA Subscribe: create Client ...
[info] [OpcUa-Client:OPC UA Subscribe] [object Object]
**[info] [OpcUa-Client:OPC UA Subscribe] Client status: create client
[info] [OpcUa-Client:OPC UA Subscribe] async series - connecting 
[info] [OpcUa-Client:OPC UA Subscribe] Client status: connecting**
[warn] [OpcUa-Client:OPC UA Subscribe] OPC UA Subscribe: can't work without OPC UA Session
[warn] [OpcUa-Client:OPC UA Subscribe] OPC UA Subscribe: can't work without OPC UA Session
[warn] [OpcUa-Client:OPC UA Subscribe] OPC UA Subscribe: can't work without OPC UA Session
[info] [OpcUa-Client:OPC UA Subscribe] Client disconnected!
[warn] [OpcUa-Client:OPC UA Subscribe] OPC UA Subscribe: create Client ...
[info] [OpcUa-Client:OPC UA Subscribe] [object Object]
[info] [OpcUa-Client:OPC UA Subscribe] Client status: create client

In a function node between the inject node and the OPC UA Client node, I use roughly the following code (but with around 200 items):
```
const interval = 1000;
let items = [];
items.push({'name': 'hello_world', 'topic': 'ns=3;s="Dummy"', 'datatype': 'Float', 'payload': interval});
return items;

```

I already made a fix which I'm still testing, see my fork.

mikakaraila, 2017-07-27:
Ack, seems to be nice & clean solution. Inform me as you have tested it.

nospam2000, 2017-07-31:
I tested my change and it solves the described problem to 90%.

There are still some problems with dangling OPC UA connections (e.g. currently there are 4 open OPC UA connections, but I only have one OPC UA node object). On my fork I created a new branch "conn_handling" where I try to fix the handling of the sessions and client objects.
I think that the session must be closed, before the session variable is set to null, but I'm still learning to understand the code.

mikakaraila, 2017-08-02:
Ok I will wait...

mikakaraila, 2017-08-24:
I will merge solution without session fix... or should I wait until next week?

nospam2000, 2017-08-25:
**You can merge the version on my "master" branch. This should fix most of the problems.**

On the "conn_handling" branch I made additional changes, but it is not clear if they help.
1. The parameters 'done' of the `node.on("close"` eventhandler is now used (parameter 'removed' is not used)

2. `close_opcua_client(set_node_status_to("closed"))` is probably not doing what was intended: it first calls set_node_status_to and passes the return-value to close_opcua_client. I changed it to a callback:
```
       close_opcua_client(function() {
         set_node_status_to("closed");
         done();
       }); 
```
When you look at the 'TODO' comments in the code, you can see the open issues.

closed 2017-08-28 by mikakaraila