mikakaraila/node-red-contrib-opcua #59: Change Endpoint

srebling, 2017-08-08:
Any way to programmatically set or change the Endpoint for the OpcUa Client node?

msg.endpoint = "opc.tcp://...";  (does not seem to work)

Setting/changing msg.action and msg.topic works great...

mikakaraila, 2017-08-09:
Ack, I added this as enhancement. I will some time, but not now... bit later.

dgP86, 2018-03-07:
This enhancement would be really helpful to change the endpoints e.g. by dropdown menus and without redeploying the flows.

mikakaraila, 2018-03-08:
OK, I will look this today after work.

mikakaraila, 2018-03-08:
Code written but some work with v0.2.2 and testing...

mikakaraila, 2018-03-12:
Hmm, still something. Now client closes connection as I try to open new one.
More debugging... but progressing.

mikakaraila, 2018-03-12:
OK this seems to work now. You can use inject to reconnect to same client.
Not an another one as defined in OpcUaClient node.
I will do some other testing before pushing this to GIT and publishing to npm.

mikakaraila, 2018-03-13:
OK works now. 
I will test other nodes just to check that I did not broke anything with updating to node-opcua v0.2.2
[image]

mikakaraila, 2018-03-17:
Action reconnect can be now used to reconnect or change endpoint.

closed 2018-03-17 by mikakaraila

S-Idris, 2018-04-13:
Hi,
I already wrote one endpoint in OPC UA Client node and by appending a function block before this node I tried to connect the client to a new endpoint by  configuring a **msg.action="reconnect",msg.endpoint="opc.tcp.<new endpoint>"**

still I am able to reconnect to the previous endpoint but  I am not able to change the previous endpoint with new one.
can you please help me with changing the endpoint of node dynamically?

mikakaraila, 2018-04-13:
Test with this one:
[{"id":"6dc7f81e.bfaff8","type":"OpcUa-Client","z":"f555c05.a35cec","endpoint":"4885da78.65011c","action":"read","time":10,"timeUnit":"s","name":"","x":559.5,"y":145.59999084472656,"wires":[["5fd1a1b9.8ca57"]]},{"id":"5fd1a1b9.8ca57","type":"debug","z":"f555c05.a35cec","name":"","active":true,"console":"false","complete":"false","x":815.5,"y":147.39999389648438,"wires":[]},{"id":"5e0698a5.fab678","type":"inject","z":"f555c05.a35cec","name":"ProSys","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":109.5,"y":94,"wires":[["19a957aa.4ef46"]]},{"id":"4285ebb4.5d47c4","type":"inject","z":"f555c05.a35cec","name":"Test Server2","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":131.1999969482422,"y":223.1999969482422,"wires":[["a93afbff.01d608"]]},{"id":"19a957aa.4ef46","type":"function","z":"f555c05.a35cec","name":"ProSys params","func":"msg.topic=\"Reconnect\";\nmsg.action=\"reconnect\";\nmsg.OpcUaEndpoint = {\n    credentials: {},\n    endpoint: 'opc.tcp://TREL16087112:53530/OPCUA/SimulationServer',\n    securityPolicy: 'None',\n    securityMode: 'NONE',\n    login: false,\n    user: undefined,\n    password: undefined \n}\n\nreturn msg;","outputs":1,"noerr":0,"x":283.49998474121094,"y":95.19999694824219,"wires":[["6dc7f81e.bfaff8"]]},{"id":"a93afbff.01d608","type":"function","z":"f555c05.a35cec","name":"Server2 params","func":"msg.topic=\"Reconnect\";\nmsg.action=\"reconnect\";\nmsg.OpcUaEndpoint = {\n    credentials: {},\n    endpoint: 'opc.tcp://127.0.0.1:4840/simulation/',\n    securityPolicy: 'None',\n    securityMode: 'NONE',\n    login: false,\n    user: undefined,\n    password: undefined \n}\n\nreturn msg;","outputs":1,"noerr":0,"x":314.20001220703125,"y":221.1999969482422,"wires":[["6dc7f81e.bfaff8"]]},{"id":"492d4ed6.841f38","type":"OpcUa-Item","z":"f555c05.a35cec","item":"ns=3;s=SByteDataItem","datatype":"SByte","value":"","name":"SByte","x":306.0000305175781,"y":143.1999969482422,"wires":[["6dc7f81e.bfaff8"]]},{"id":"a1c4ea0d.b1fff","type":"inject","z":"f555c05.a35cec","name":"Test","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":108.00001525878906,"y":143.39999389648438,"wires":[["492d4ed6.841f38"]]},{"id":"ee7e4b5d.1bf0b8","type":"OpcUa-Item","z":"f555c05.a35cec","item":"ns=2;i=2","datatype":"Float","value":"","name":"Float","x":348.20001220703125,"y":282.20001220703125,"wires":[["6dc7f81e.bfaff8"]]},{"id":"6b8d3577.7c34e4","type":"inject","z":"f555c05.a35cec","name":"Test","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":150.1999969482422,"y":282.40000915527344,"wires":[["ee7e4b5d.1bf0b8"]]},{"id":"4885da78.65011c","type":"OpcUa-Endpoint","z":"f555c05.a35cec","endpoint":"opc.tcp://TREL16087112:53530/OPCUA/SimulationServer","secpol":"None","secmode":"NONE","login":false}]

S-Idris, 2018-04-19:
@mikakaraila, thanks for sharing the example, it helped me a lot.

@dgP86 Yes I also faced the same issue at the first time, then I realized that json is not correctly formed so with small changes I made it a valid json.
Please try to import this updated node-red-flow.

[{"id":"6dc7f81e.bfaff8","type":"OpcUa-Client","z":"f555c05.a35cec","endpoint":"4885da78.65011c","action":"read","time":10,"timeUnit":"s","name":"","x":559.5,"y":145.59999084472656,"wires":[["5fd1a1b9.8ca57"]]},{"id":"5fd1a1b9.8ca57","type":"debug","z":"f555c05.a35cec","name":"","active":true,"console":"false","complete":"false","x":815.5,"y":147.39999389648438,"wires":[]},{"id":"5e0698a5.fab678","type":"inject","z":"f555c05.a35cec","name":"ProSys","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":109.5,"y":94,"wires":[["19a957aa.4ef46"]]},{"id":"4285ebb4.5d47c4","type":"inject","z":"f555c05.a35cec","name":"Test Server2","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":131.1999969482422,"y":223.1999969482422,"wires":[["a93afbff.01d608"]]},{"id":"19a957aa.4ef46","type":"function","z":"f555c05.a35cec","name":"ProSys params","func":"msg.topic=\"Reconnect\";\nmsg.action=\"reconnect\";\nmsg.OpcUaEndpoint = {\n credentials: {},\n endpoint: 'opc.tcp://TREL16087112:53530/OPCUA/SimulationServer',\n securityPolicy: 'None',\n securityMode: 'NONE',\n login: false,\n user: undefined,\n password: undefined \n}\n\nreturn msg;","outputs":1,"noerr":0,"x":283.49998474121094,"y":95.19999694824219,"wires":[["6dc7f81e.bfaff8"]]},{"id":"a93afbff.01d608","type":"function","z":"f555c05.a35cec","name":"Server2 params","func":"msg.topic=\"Reconnect\";\nmsg.action=\"reconnect\";\nmsg.OpcUaEndpoint = {\n credentials: {},\n endpoint: 'opc.tcp://127.0.0.1:4840/simulation/',\n securityPolicy: 'None',\n securityMode: 'NONE',\n login: false,\n user: undefined,\n password: undefined \n}\n\nreturn msg;","outputs":1,"noerr":0,"x":314.20001220703125,"y":221.1999969482422,"wires":[["6dc7f81e.bfaff8"]]},{"id":"492d4ed6.841f38","type":"OpcUa-Item","z":"f555c05.a35cec","item":"ns=3;s=SByteDataItem","datatype":"SByte","value":"","name":"SByte","x":306.0000305175781,"y":143.1999969482422,"wires":[["6dc7f81e.bfaff8"]]},{"id":"a1c4ea0d.b1fff","type":"inject","z":"f555c05.a35cec","name":"Test","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":108.00001525878906,"y":143.39999389648438,"wires":[["492d4ed6.841f38"]]},{"id":"ee7e4b5d.1bf0b8","type":"OpcUa-Item","z":"f555c05.a35cec","item":"ns=2;i=2","datatype":"Float","value":"","name":"Float","x":348.20001220703125,"y":282.20001220703125,"wires":[["6dc7f81e.bfaff8"]]},{"id":"6b8d3577.7c34e4","type":"inject","z":"f555c05.a35cec","name":"Test","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":150.1999969482422,"y":282.40000915527344,"wires":[["ee7e4b5d.1bf0b8"]]},{"id":"4885da78.65011c","type":"OpcUa-Endpoint","z":"f555c05.a35cec","endpoint":"opc.tcp://TREL16087112:53530/OPCUA/SimulationServer","secpol":"None","secmode":"NONE","login":false}]

LucaCarminati-it, 2021-01-14:
> @mikakaraila, thanks for sharing the example, it helped me a lot.
>
> @dgP86 Yes I also faced the same issue at the first time, then I realized that json is not correctly formed so with small changes I [...]

Hi S-Idris,
Thanks for your code (corrected), it helped me a lot.
I have , I think , the same problem that you had, so if I try to change the Endpoint while there is a connection yet estabilished it works and I am able to switch and connect dinamically two or more device.
If I try to change the Endpoint of a connection that is not yet estabilished, or is closed, it doesn't work, and in the debug node I receive the error : 

TypeError: Cannot read property 'close' of undefined

Does somebody find a solution about that ?

I attach my node flow : 

`[{"id":"4d6f4bc7.443eb4","type":"tab","label":"Test OPC","disabled":false,"info":""},{"id":"9bb621f2.c9f6e","type":"debug","z":"4d6f4bc7.443eb4","name":"","active":true,"tosidebar":true,"console":true,"tostatus":false,"complete":"true","targetType":"full","statusVal":"","statusType":"auto","x":790,"y":240,"wires":[]},{"id":"16c7b350.0ad55d","type":"inject","z":"4d6f4bc7.443eb4","name":"Read","props":[{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"ns=3;s=\"opctest\".\"var\"","x":310,"y":220,"wires":[["adb7bc03.ce77c"]]},{"id":"5f5219a8.2e4658","type":"function","z":"4d6f4bc7.443eb4","name":"","func":"msg.topic=\"Reconnect\";\nmsg.action=\"reconnect\";\nmsg.OpcUaEndpoint = {\n credentials: {},\n endpoint: 'opc.tcp://172.31.224.21:4840/simulation/',\n securityPolicy: 'None',\n securityMode: 'NONE',\n login: false,\n user: undefined,\n password: undefined \n}\n\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":460,"y":180,"wires":[["adb7bc03.ce77c"]]},{"id":"b67141a3.d5d06","type":"inject","z":"4d6f4bc7.443eb4","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"str","x":310,"y":180,"wires":[["5f5219a8.2e4658"]]},{"id":"5f2ad9cb.ecdd38","type":"function","z":"4d6f4bc7.443eb4","name":"","func":"msg.topic=\"Reconnect\";\nmsg.action=\"reconnect\";\nmsg.OpcUaEndpoint = {\n credentials: {},\n endpoint: 'opc.tcp://172.31.224.2:4840/simulation/',\n securityPolicy: 'None',\n securityMode: 'NONE',\n login: false,\n user: undefined,\n password: undefined \n}\n\nreturn msg;","outputs":1,"noerr":0,"initialize":"","finalize":"","x":460,"y":300,"wires":[["adb7bc03.ce77c"]]},{"id":"c7de412.51578c","type":"inject","z":"4d6f4bc7.443eb4","name":"","props":[{"p":"payload"},{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"","payload":"","payloadType":"str","x":310,"y":300,"wires":[["5f2ad9cb.ecdd38"]]},{"id":"162d8352.4956ad","type":"inject","z":"4d6f4bc7.443eb4","name":"Read","props":[{"p":"topic","vt":"str"}],"repeat":"","crontab":"","once":false,"onceDelay":0.1,"topic":"ns=3;s=\"opcua\".\"plc2\"","x":310,"y":260,"wires":[["adb7bc03.ce77c"]]},{"id":"adb7bc03.ce77c","type":"OpcUa-Client","z":"4d6f4bc7.443eb4","endpoint":"94949813.9a64e8","action":"read","deadbandtype":"a","deadbandvalue":1,"time":10,"timeUnit":"s","certificate":"n","localfile":"","localkeyfile":"","securitymode":"None","securitypolicy":"None","name":"","x":640,"y":240,"wires":[["9bb621f2.c9f6e"]]},{"id":"94949813.9a64e8","type":"OpcUa-Endpoint","endpoint":"opc.tcp://0.0.0.0:0000","secpol":"None","secmode":"None","login":false}]`

Thanks to all!