mikakaraila/node-red-contrib-opcua #60: Problem with write to the OPC server

Jwenda, 2017-08-10:
Hi,
I'm using this node on Raspberry pi 3. I'm using node-opcua as a server.

When I tried to use your example ExampleFlow.json and I just changed the OPC server, it is not writing to the OPC server variables. The value is still the same.

Here is the writing and reading example:
[{"id":"5f334c75.a5b744","type":"OpcUa-Item","z":"f7029c5.3c5f86","item":"ns=1;i=1020ffaa;datatype=Double","datatype":"Double","value":"20","name":"","x":281.00000762939453,"y":107.00000381469727,"wires":[["b7f8a70b.9b2898"]]}]

Here is the server code: https://github.com/node-opcua/node-opcua/blob/master/documentation/sample_server.js

Thank you for your help.

mikakaraila, 2017-08-14:
Code does not support opaque b=0x1020ffaa item. sample_client.js is now simple_client.js
simple_client.js add folloging code before // Add view

    // add a variable named MyVariable2 to the newly created folder "MyDevice"
    var variable2 = 10.0;
        
    var node2 = addressSpace.addVariable({
            
            organizedBy: myDevices,
            
//            nodeId: "ns=1;b=1020FFAA", // some opaque NodeId in namespace 4
            nodeId: "ns=1;s=MyVariable2",
            browseName: "MyVariable2",
            
            dataType: "Double",    
            
            value: {
                get: function () {
                    return new opcua.Variant({dataType: opcua.DataType.Double, value: variable2 });
                },
                set: function (variant) {
                    variable2 = parseFloat(variant.value);
                    return opcua.StatusCodes.Good;
                }
            }
    });
Flow example:
[{"id":"cc68f044.32062","type":"OpcUa-Item","z":"c95d316b.eeb148","item":"ns=1;s=MyVariable2","datatype":"Double","value":"","name":"","x":203.49998474121094,"y":85.19999694824219,"wires":[["6b11c241.821294"]]},{"id":"6b11c241.821294","type":"OpcUa-Client","z":"c95d316b.eeb148","endpoint":"17b87a31.bbc01e","action":"write","time":10,"timeUnit":"s","name":"","x":392.5,"y":83.80000305175781,"wires":[["8d7048ed.d2da68"]]},{"id":"8d7048ed.d2da68","type":"debug","z":"c95d316b.eeb148","name":"","active":true,"console":"false","complete":"false","x":586.5,"y":85,"wires":[]},{"id":"a16c4cf2.6f7df8","type":"inject","z":"c95d316b.eeb148","name":"Write","topic":"","payload":"22.5","payloadType":"num","repeat":"","crontab":"","once":false,"x":70.5,"y":34,"wires":[["cc68f044.32062"]]},{"id":"17b87a31.bbc01e","type":"OpcUa-Endpoint","z":"c95d316b.eeb148","endpoint":"opc.tcp://localhost:26543","secpol":"None","secmode":"NONE","login":false}]

Jwenda, 2017-08-15:
Hi,
thank you very much for your example. It is working perfectly.

closed 2017-08-15 by Jwenda