mikakaraila/node-red-contrib-opcua #61: Error at RaspberryPi but on Windows it's working

vallik800i, 2017-08-22:
Hello there,
I am a student from Germany and at this moment I am working at a company.

My task is to read something out of a PLC with OPC UA.

My programm has about 8 Flows with 13 OPCUA client nodes, which are reading the data from the PLC.
If I deploy only 1-2 Flows there are no problems on my RaspberryPi 3. But if I deploy more, I get the warning: 

warning : ClientSessionKeepAliveManager#ping_server   ServiceResult is BadSessionIdInvalid (0x80250000)
 warning : ClientSessionKeepAliveManager#ping_server   ServiceResult is BadSessionIdInvalid (0x80250000)
 message was 
undefined
 message was 2: { /*OpenSecureChannelRequest*/
 requestHeader                 /* RequestHeader                    */: {
   authenticationToken         /* NodeId                           */: ns=0;i=0
   timeStamp                   /* UtcTime                          */: 2017-08-18T19:50:29.933Z
   requestHandle               /* IntegerId                        */: 1               0x1
   returnDiagnostics           /* UInt32                           */: 0               0x0
   auditEntryId                /* UAString                         */: null
   timeoutHint                 /* UInt32                           */: 0               0x0
   additionalHeader            /* ExtensionObject                  */: null
 }
 clientProtocolVersion         /* UInt32                           */: 0               0x0
 requestType                   /* SecurityTokenRequestType         */: ISSUE ( 0)
 securityMode                  /* MessageSecurityMode              */: NONE ( 1)
 clientNonce                   /* ByteString                       */: null
 requestedLifetime             /* UInt32                           */: 600000               0x927c0
};

For about one minute, the programm will run now. But after one minute I get this Error:

Timeout .... waiting for response for  OpenSecureChannelRequest { /*RequestHeader*/
 authenticationToken           /* NodeId                           */: ns=0;i=0
 timeStamp                     /* UtcTime                          */: 2017-08-18T19:50:29.933Z
 requestHandle                 /* IntegerId                        */: 1               0x1
 returnDiagnostics             /* UInt32                           */: 0               0x0
 auditEntryId                  /* UAString                         */: null
 timeoutHint                   /* UInt32                           */: 0               0x0
 additionalHeader              /* ExtensionObject                  */: null
};
18 Aug 19:51:29 - [error] [OpcUa-Client:7ad5e19b.271e7]  OPC UA connection error: Transaction has timed out
warning : disconnection : closing pending sessions

But I get this Errors only at the RaspPi. On Windows I get sometimes the warning but not the Error..
What is my fault? I am searching for a week now and i don't know what's wrong...

Can you help me?
Thanks in advance and sorry for my bad english..

Valentin

nospam2000, 2017-08-22:
can you please check the number of open tcp connections to your OPC UA server, either by using the following command on the client machine (replace "4840" by the port number of your OPC UA server)

`netstat -a -p tcp | grep 4840`

When there are more connections than OPC UA nodes in your flows, you might have run into issue #58.
PS: is it a secret for which company you are working?

vallik800i, 2017-08-23:
I have changed the programm a little bit. 
now I have 13 OpcUa Items which are connected to one OpcUa Client. 
Also there are 8 Flows.
In the first Flow there are the OpcUa things. I am writing them in a databank.
In the other Flows I read datas out of the databank and put them in other db's and process them and show them on dashboard.

If I only activate the Flow 1 (OpcUa reading) i have no errors and I have only one connection to my OPCUA server (netstat -a -p tcp | grep 4840)
But if I activate one or two more Flows, I will get the error and I have between 5 and 14 connections to the OpcUa Server.
But why? In the other Flows there is nothing with OpcUa...
I have no idea...

I don't know if I am allowed to tell my company name.. But we are a medium-sized company with less than 1000 employees.

Thanks for the fast reply.

mikakaraila, 2017-08-23:
One cause could be that the other flow nodes will block or cause delay so OPC UA will try to use keepalive session(s) to reconnect... this is just idea that you should check.

vallik800i, 2017-08-23:
I think this is the problem.
The other Flows cause a delay that the OpcUa isn't fast enough and more Clients will be opened..
So the RaspPi-processor is too slow?
Is there a possibility to avoid this?

mikakaraila, 2017-08-23:
One question is how you are using flows. Are there a flow that will save data to database?
Is database local or remote... etc.

vallik800i, 2017-08-23:
All Flows are saving data to databases. I am using SQLite so the databases are local

mikakaraila, 2017-08-23:
Ok are those implemented async? So storing data does not block other actions/flows/events?
Is the db-node some public existing one or have you implemented it by yourself?

vallik800i, 2017-08-23:
The db-node is public existing from node-red-node-sqlite.
I split som Flows and read something out of other db's and join them after that together. 
Picture from OpcUa Flow:
https://picload.org/view/rwcporiw/1.png.html

Picture from one other Flow(all other Flows are near to that):
https://picload.org/view/rwcporpi/2.png.html

mikakaraila, 2017-08-23:
How fast are you getting&storing data? Every 1s or faster? SQlite reconnect timeout is 20s.
I think it is related to CPU load and how fast Raspberry can store data.
Which model you are using? Model2 or model3? SDcard speed can also effect.
Check cpu load to narrow the bottleneck...

vallik800i, 2017-08-23:
I am getting and storing every 1s. I tried it with 3 and 5s but same error..
I am using the RaspPi 3. CPU-Usage is between 5 and 30%.
The node-red-Task needs 1-3% CPU usage

mikakaraila, 2017-08-23:
Have you created db and tables before you start storing?
Model3 should be capable to run this. This must have some principle error...

vallik800i, 2017-08-23:
Yes, I created the table before.
The datas writing correctly in the db's.

mikakaraila, 2017-08-23:
Do you have index on table? Can you add/check timestamp behaviour?

vallik800i, 2017-08-23:
The db is every time writing when the OpcUa is running. If the error pops out, The db ( named "daten", as you can see in the first picture from the comment with the pictures) stop writing because it gets no datas.
The other db's which are reading from the main db ("daten") only read the last entry.

vallik800i, 2017-08-24:
I've got the Solution!! 
If I start the interval for reading after deploying, there is no error!
I am just not allowed to start it instantly.
But why? :D

mikakaraila, 2017-08-24:
Ack, there is fix / solution coming, see #58  I will merge it soon.

vallik800i, 2017-08-24:
Okay top

mikakaraila, 2017-08-28:
Merge done, please test.

vallik800i, 2017-08-28:
you've merged #62 with #58 but not #61, or am I wrong?

mikakaraila, 2017-08-28:
I suppose pull request contained more fixes... did you test this?
I will have more time after work.

closed 2017-08-28 by mikakaraila