mikakaraila/node-red-contrib-opcua #63: catching node.error-

sowcow, 2017-08-30:


mikakaraila, 2017-08-31:
If you have done this already, please make PR.

nospam2000, 2019-07-10:
The second parameter needs to be the msg and not a string. Otherwise this causes Node-Red to completely crash when you use a "catch" node. 

See https://github.com/node-red/node-red/issues/2228 for the problem this causes

See also the documentation in https://nodered.org/docs/user-guide/writing-functions:
_Handling errors_
_If the function encounters an error that should halt the current flow, it should return nothing. To trigger a Catch node on the same tab, the function should call node.error **with the original message as a second argument**:_

`node.error("hit an error", msg);
`

mikakaraila, 2020-08-10:
Sorry I somehow forgot this one but now as I cleanup code I fixed this one.
At least v0.2.78 should contain this fix.

closed 2020-09-02 by mikakaraila