mikakaraila/node-red-contrib-opcua #67: Missing genrsa -rand option

giminni, 2017-11-14:
Hi Mika,
I am trying to use the lib with alpine linux with libressl, but the -rand option is missing.
The installation went wrong,  and produced the following error:

> OpenSSL version :  LibreSSL 2.6.3
ERROR  Command failed: "openssl" genrsa  -out  private/cakey.pem -rand random.rnd 2048
usage: genrsa [args] [numbits]
 -des            encrypt the generated key with DES in cbc mode
 -des3           encrypt the generated key with DES in ede cbc mode (168 bit key)
 -idea           encrypt the generated key with IDEA in cbc mode
 -aes128, -aes192, -aes256
                 encrypt PEM output with cbc aes
 -camellia128, -camellia192, -camellia256
                 encrypt PEM output with cbc camellia
 -out file       output the key to 'file
 -passout arg    output file pass phrase source
 -f4             use F4 (0x10001) for the E value
 -3              use 3 for the E value

> Command failed: "openssl" genrsa  -out  private/cakey.pem -rand random.rnd 2048
usage: genrsa [args] [numbits]
 -des            encrypt the generated key with DES in cbc mode
 -des3           encrypt the generated key with DES in ede cbc mode (168 bit key)
 -idea           encrypt the generated key with IDEA in cbc mode
 -aes128, -aes192, -aes256
                 encrypt PEM output with cbc aes
 -camellia128, -camellia192, -camellia256
                 encrypt PEM output with cbc camellia
 -out file       output the key to 'file
 -passout arg    output file pass phrase source
 -f4             use F4 (0x10001) for the E value
 -3              use 3 for the E value

done ... (0)

Can the installation produces rsa keys without using -rand?

giminni, 2017-11-17:
Klaus said it is a node-opcua issue. I'll close this because I guess it is the same kind of issue.

closed 2017-11-17 by giminni

mikakaraila, 2017-11-18:
Ack, node-opcua packaging has changed and I think I should update to latest one.
I will look when I will have time.

giminni, 2017-11-19:
Thx Mika