mikakaraila/node-red-contrib-opcua #68: Certificate Error

dlabudovGP, 2017-11-20:
Node-Red V. 0.17.5
NodeJS V8.9.1

Error Message while trying to utilize OPC UA browser to a UI template node.
Utilizing Kepware ServerEX OPC UA server with all security turned off for testing purposes.
> "Error: cannot locate certificate file C:\Users\DLABUDOV\.node-red\node_modules\node-opcua\certificates\client_selfsigned_cert_1024.pem"

I have checked this path, and the file exists.

Node Clipboard 
`[{"id":"c4c870b0.f6c58","type":"OpcUa-Browser","z":"47b2a9d7.488c74","endpoint":"4ba0ce65.716ad8","item":"","datatype":"","topic":"","items":[],"x":380,"y":159,"wires":[["f11356c8.de74e8"]]},{"id":"4ba0ce65.716ad8","type":"OpcUa-Endpoint","z":"47b2a9d7.488c74","endpoint":"opc.tcp://170.168.11.78:4840","secpol":"None","secmode":"NONE","login":false}]`

Any suggestions?

mikakaraila, 2017-11-20:
Have you tested with the latest one of node-red-contrib-opcua?
If I remember correctly there was something...

dlabudovGP, 2017-11-20:
I think its the latest, its version 0.2.22

mikakaraila, 2017-11-22:
Cannot reproduce error, did you check that node-opcua installation was successful?
In the error message is there missing home path separator after C:\Users\DLABUDOV ?
C:\Users\DLABUDOV**\\**.node-red\node_modules\node-opcua\certificates\client_selfsigned_cert_1024.pem"

mblackstock, 2018-01-08:
I am seeing similar errors on Mac OS High Sierra 10.13.2

The error I see in the NR console when running is:

[error] [OpcUa-Client:2e2ef53f.d96ada] Error:  cannot locate certificate file .../node_modules/node-opcua/certificates/client_selfsigned_cert_1024.pem

I believe updating node-opcua to the latest may fix this error: https://github.com/node-opcua/node-opcua/issues/254

mikakaraila, 2018-03-17:
This should work now with new node-opcua v0.2.2

Gawel13, 2018-04-12:
Unfortunately I get the same issue.
node-red-contrib-ua version 0.2.27
node-opcua version 0.2.3
Have I missed something ?

mikakaraila, 2018-04-12:
Did you uninstall older node-opcua?
If not then it can still load old npm packages and thus use wrong relative path.
npm -g does not work because code uses relative path under .node-red folder.
Check and comment please.

mikakaraila, 2018-04-14:
OK added, by default code will use relative path, but if you add absolute path then it should use it.
Please test.

Gawel13, 2018-04-16:
Uninstalled and reinstalled node-red-contrib-opcua, now works just fine !
Thanks a lot

closed 2018-04-16 by mikakaraila