mikakaraila/node-red-contrib-opcua #69: OPCUA Client Subscribe value

mrsalhaoui, 2017-11-22:
Hello !

please I still having this problem, even I upgrade the OPCUA node : 

"subscription.monitorItem:Error: BadNodeIdUnknown (0x80340000)"

mikakaraila, 2017-11-22:
What kind of nodeId are you using?
Do you have correct namespace?
Can you access it with UaExpert?

mrsalhaoui, 2017-11-22:
I'm new in node-red, I'm using inject node and also a drop down from dashboard in order to have a connection with a plc, but I have the same error in boat of them.
for UaExpert I don't know how to use it.

[image]

mikakaraila, 2017-11-22:
Can you browse address space?
Debug window message tolds that check topic format. 
I expect it should be something like ns=3;s=OPC_UA_Values.Mode

mrsalhaoui, 2017-11-22:
[image]
I changed but it stills the same problem

mikakaraila, 2017-11-22:
As Debug says BadNodeIdUnknow.
Try this flow, setup server endpoint and then browse address to find out correct nodeIds.
This is tested with Prosys OPC UA Simulation server.
[{"id":"c458779b.de0bb8","type":"OpcUa-Browser","z":"45b1a8ba.9778e","endpoint":"81c1880c.2bb588","item":"","datatype":"","topic":"ns=2;s=MyObjectsFolder","items":[],"x":567.5000610351562,"y":92.80000305175781,"wires":[["72d231d4.6c1e3"]]},{"id":"dd33332b.6e79d8","type":"inject","z":"45b1a8ba.9778e","name":"Browse","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":159.5,"y":92.79999542236328,"wires":[["977dc05.fcd1ac"]]},{"id":"72d231d4.6c1e3","type":"debug","z":"45b1a8ba.9778e","name":"","active":true,"console":"false","complete":"payload","x":768.2000122070312,"y":95.20001220703125,"wires":[]},{"id":"977dc05.fcd1ac","type":"function","z":"45b1a8ba.9778e","name":"Set browse address","func":"msg.topic='ns=0;i=85';\nmsg.actiontype='browse';\nreturn msg;","outputs":1,"noerr":0,"x":354.50001525878906,"y":93,"wires":[["c458779b.de0bb8"]]},{"id":"81c1880c.2bb588","type":"OpcUa-Endpoint","z":"45b1a8ba.9778e","endpoint":"opc.tcp://TREL16087112.vstage.co:53530/OPCUA/SimulationServer","secpol":"None","secmode":"NONE","login":false}]

mrsalhaoui, 2017-11-23:
I need just OPC UA client for this Client, why the browser ? 
I tried your example but I dont know how to setup the blocks, I only change the address of my PLC in the OPC UA browser, and I dont know how to find out correct nodes, it still have the same problem Sir.

[image]

mikakaraila, 2017-11-23:
Browse node is just to list your address space, but if you can share PLC configuration or 
some other view that will show nodeId it will be enough.
Now the given string ns=3;s=OPC_UA_Values.Temperature is not correct.
Do you have any other OPC UA Client that can access the server?

mikakaraila, 2017-11-23:
Your flow should be like this one (OPC UA Item is missing from your flow or it contains item address in wrong format.
[image]

Flow:
[{"id":"8c4eedfb.a5398","type":"OpcUa-Item","z":"7964702c.fd52a","item":"ns=5;s=Counter1","datatype":"opcua.DataType.Double","value":"","name":"","x":316,"y":113.19999694824219,"wires":[["a4d61800.667ce8"]]},{"id":"326404a1.888f9c","type":"inject","z":"7964702c.fd52a","name":"Subscribe","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"x":142,"y":112.19999694824219,"wires":[["8c4eedfb.a5398"]]},{"id":"a4d61800.667ce8","type":"OpcUa-Client","z":"7964702c.fd52a","endpoint":"4120b567.1f3ddc","action":"subscribe","time":"1","timeUnit":"s","name":"Test server (subscribe item)","x":576,"y":113.19999694824219,"wires":[["e49a58e1.78e2c"]]},{"id":"e49a58e1.78e2c","type":"debug","z":"7964702c.fd52a","name":"Subscribed values","active":true,"console":"false","complete":"true","x":814,"y":113.19999694824219,"wires":[]},{"id":"4120b567.1f3ddc","type":"OpcUa-Endpoint","z":"7964702c.fd52a","endpoint":"opc.tcp://TREL16087112.vstage.co:53530/OPCUA/SimulationServer","secpol":"None","secmode":"NONE","login":false}]

mrsalhaoui, 2017-11-23:
I'm working to subscribe and write a value in a PLC using node-Red in the IoT2040 of Siemens, I just followed the tutorial of a PDF of siemens, to uses OPC UA of node-red you find attached the file. and also in the example they use the item but for it still the same problem 

[OPC_UA_Client_node-red_V1.0.pdf](https://github.com/mikakaraila/node-red-contrib-opcua/files/1500104/OPC_UA_Client_node-red_V1.0.pdf)

[image]

allatrue, 2017-11-23:
If you want to work with opc ua, you should get and learn UaExpert. It's a "must have"  tool for troubleshooting and a good start point for beginners. Basic functions of UaExpert are easy to use.

mikakaraila, 2017-11-24:
Yes I agree, trying to help people is not easy if they cannot use basic tools.
Here is download link for UaExpert:
https://www.unified-automation.com/products/development-tools/uaexpert.html

mrsalhaoui, 2017-11-25:
Thank you Mr.Mika and Mr.alla for your help, I had installed UaExpert as a client, and I can change the state of my database from UaExpert to the PLC, I had searched also in Siemens forum, and I found that the NodeId must be in the same name as on my PLC, as you also said before, so I had just changed the name of the database that I use in my PLC in Node-red and it works, and this NodeID is always showing in UaExpert in the correct way, I tried it again with other program in my PLC to change the state of some variables, I connected the UaExpert client, I took the NodeID from it, I put it in Node-red and I got the connection.

[image]

mikakaraila, 2017-11-26:
Good problem solved :-)

closed 2017-11-26 by mikakaraila