mikakaraila/node-red-contrib-opcua #70: Monitored item twice

cabsil, 2017-11-24:
Hello,

I have installed the example flow from here https://github.com/mikakaraila/node-red-contrib-opcua/blob/eeba8e62e0068ac64a155fc0e3381554690e61a5/examples/ExampleFlow.json

I only used the "Subscribe (100ms)" inject and I configured the item according to my opc ua server.  As a result I got always two messages at every interval. For your information, my item is a simple counter with a +1 every 500 msec.

Is this normal ?

Regards

Christophe Absil

PS : you are doing a great job keep going.

mikakaraila, 2017-11-26:
Should not give it twice, ExampleFlow.json is bit old.
But did you inject subscription twice?

cabsil, 2017-11-27:
No I deployed the flow and click once. So I assume there is only one subscription.

mikakaraila, 2018-03-17:
Check this with the latest version. I am just cleaning up issues. Reopen if not yet ok.

closed 2018-03-17 by mikakaraila