mikakaraila/node-red-contrib-opcua #71: Write Double value issue

wlabsoft, 2018-02-20:
Using the same flow to write Boolean from my client as in **examples/ProsysOpcUaFlow.json** I'm able to set a Boolean value, but when my variable is of type Double it send **NaN** instead of **1** as value to my OPC server!

mikakaraila, 2018-02-20:
Can you attach flow here? You should use Double as it is the correct type on the server.
Boolean is True or False and conversion will result NaN for Double.
I have not implemented different type conversions. You can add one function node to make
needed conversion from True to 1.0 and False to 0.0 if needed.

wlabsoft, 2018-02-20:
When debug the output of the OpcUa-Item box I saw msg.payload = ""
As workaround, I've added a function to set msg.payload to 1.0 and it works now!

This is the json of the flow with the workaround:
=====================
[{"id":"5168a837.9c22a8","type":"inject","z":"6ba32c06.77deb4","name":"Write","topic":"","payload":"","payloadType":"str","repeat":"","crontab":"","once":false,"onceDelay":"","x":97,"y":814,"wires":[["4f9bb995.458848"]]},{"id":"4f9bb995.458848","type":"OpcUa-Item","z":"6ba32c06.77deb4","item":"ns=3;s=Double","datatype":"Double","value":"1","name":"Double (1.0)","x":297,"y":814,"wires":[["6fb1abfc.c852a4"]]},{"id":"a6cd15ef.dd28b8","type":"OpcUa-Client","z":"6ba32c06.77deb4","endpoint":"a131bb7d.771f88","action":"write","time":10,"timeUnit":"s","name":"Client Write","x":751,"y":822,"wires":[["df908b08.b04678"]]},{"id":"df908b08.b04678","type":"debug","z":"6ba32c06.77deb4","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"true","x":961,"y":823,"wires":[]},{"id":"6fb1abfc.c852a4","type":"function","z":"6ba32c06.77deb4","name":"","func":"msg.payload = 1.0\nreturn msg;","outputs":1,"noerr":0,"x":507,"y":822,"wires":[["a6cd15ef.dd28b8"]]},{"id":"a131bb7d.771f88","type":"OpcUa-Endpoint","z":"","endpoint":"opc.tcp://127.0.0.1:53530/OPCUA/SimulationServer","secpol":"None","secmode":"NONE","login":false}]

mikakaraila, 2018-03-17:
OK I will close this.

closed 2018-03-17 by mikakaraila