mikakaraila/node-red-contrib-opcua #72: Turn on/off and reset client connection

dgP86, 2018-03-07:
Hi all,
I'm currently trying to connect 3 different, not permanently to the network connected devices via OPC-UA.
When a device is connected to the network and I e.g. deploy the new flows a connection by the client node is established and it's working.
If one of the devices leaves the network and rejoins the OPC-UA client node is unable to reestablish the connection due to the timeout error.
Is there a way to start/stop the connection manually and/or retrigger the connection e.g. by a switch node?

Thanks in advance.
dgP

mikakaraila, 2018-03-08:
I will look this together with #59 .

mikakaraila, 2018-03-08:
Inject should be able to rejoin with pre-defined configuration or you can inject new endpoint.
Code needs some testing before I will push it to GIT & publish to npm.

dgP86, 2018-03-09:
Sounds perfekt, thanks a lot!
I guess I'll inject a new endpoint to the client nodes when a device is present and then get the data.

mikakaraila, 2018-03-12:
This can reconnect client node to Server:
[image]

pablopani2002, 2018-03-15:
Hi! Is this fix comming out anytime soon? If I understood correctly, the node will rejoin a predefined enpoint when an inject message is received, but this is not something now implemented. Right? Is there any workaround for this problem until the fix comes out?
Thanks a lot for your help!

mikakaraila, 2018-03-15:
Not today, I will redo some tests. Tomorrow most probably...

mikakaraila, 2018-03-15:
Not today, I will redo some tests. Tomorrow most probably...
Some other bugs found and fixed (caused by new v0.2.2 node-opcua changes).
[image]

mikakaraila, 2018-03-16:
Minor things still to fix...
[image]
[image]

mikakaraila, 2018-03-17:
OK just pushed and published new version.

closed 2018-03-17 by mikakaraila