mikakaraila/node-red-contrib-opcua #73: Incompatibility of node-red-contrib-opcua@0.2.23 with "node-opcua@0.2.2",

vijraman, 2018-03-20:
19 Mar 19:16:47 - [warn] [OpcUa-Client] Error: Cannot find module 'node-opcua/lib/datamodel/nodeid' (line:24)

node-opcua@0.2.2 has different hierarchy. Here is a partial list of changes I made -
//  var nodeId = require('node-opcua/lib/datamodel/nodeid');                                                                                                              
  var nodeId = require('node-opcua-nodeid/src/nodeid');
  //  var UAProxyManager = require("node-opcua/lib/client/proxy").UAProxyManager;                                                                                           
  var UAProxyManager = require("node-opcua-client-proxy/src/proxy").UAProxyManager;
  //  var coerceNodeId = require("node-opcua/lib/datamodel/nodeid").coerceNodeId;                                                                                           
  var coerceNodeId = require("node-opcua-nodeid/src/nodeid").coerceNodeId;
  //  var makeNodeId = require("node-opcua/lib/datamodel/nodeid").makeNodeId;                                                                                               
  var makeNodeId = require("node-opcua-nodeid/src/nodeid").makeNodeId
19 Mar 19:41:26 - [warn] [OpcUa-Client] ReferenceError: requinode is not defined

node version v6.11.1

mikakaraila, 2018-03-20:
Thanks, my fault. I had for some reason old & new in my development environment.
Will fix ASAP!

vijraman, 2018-03-20:
> Closed #73 <https://github.com/mikakaraila/node-red-contrib-opcua/issues/73> via 33b56f6 <https://github.com/mikakaraila/node-red-contrib-opcua/commit/33b56f65733142c6aeba29472bafd7b252ab92e9>.

I still see the issue as partially fixed. Do you want me to file this as a new issue?

20 Mar 11:30:52 - [warn] [OpcUa-Client] Error: Cannot find module 'node-opcua/lib/services/browse_service'

mikakaraila, 2018-03-20:
Ack , I will check all again. I check only beginning of file. And of course I do not have clean environment at the moment to test properly. Thanks and be patient.

closed 2018-03-20 by mikakaraila

vijraman, 2018-03-20:
Thanks. The fix works. I am just testing this and posting errors. take your time - Regards

20 Mar 13:35:27 - [warn] [OpcUa-Browser:da30dac6.85a758] browseTopic:ns=0;i=85
ERROR !!! , please check here !!!! callback may be called twice !!  TypeError: browse_result.forEach is not a function

commit 33b56f65 (2018-03-20): Fixed #73.