mikakaraila/node-red-contrib-opcua #74: Position to put certificates

corn98, 2018-03-20:
Hi Mika, 

Firstable thank for great plug-in!
Anyway I'm facing with an annoying problem, probably because I'm novice with OPC UA itself.
What I understand for now is there must be consistency between the certificates generated when a connection is wanted. 

What I'm trying to do is establish a connection with security policy=Basic128Rsa15 and mode=Sign&Encrypt to OPCUA server that stands in siemens TP Comfort. In siemens TP, I put in the right place the certificate generated. 

During installation I notice that the certificates folder is under:

.../node-red-contrib-opcua/node_modules/node-opcua/certificates

but I'm not able to understand where the certificate is generated/how to recognise it and where I have to put it, even if I suppose that is in the folder indicated above.

The error displayed in debug tab is:
Read OPC UA connection error: Transaction has timed out

Also I tried to connect to the server of the TP with UA Expert and all works.

I'm using the version 0.2.22 of opcua nodes and node-red 0.18.4.

Thanks in advance for the help!

Corn

mikakaraila, 2018-03-20:
New node-opcua uses certificates from:
Client C:\Users\USERNAME\.node-red\node_modules\node-opcua-client\certificates
Server C:\Users\USERNAME\.node-red\node_modules\node-opcua-server\certificates

corn98, 2018-03-21:
Thanks for the fast reply Mika!

Although your answer is complete I cannot find that directory in my system. Indeed I installed all nodes globally and the folder .../.node-red/node_modules is empty. 

Have you any suggestions considering this situation?

Thanks one more time for your Help!

Corn

mikakaraila, 2018-03-21:
You can run npm list -g to see where global libraries are installed.
On Unix systems they are normally placed in /usr/local/lib/node or /usr/local/lib/node_modules when installed globally. If you set the NODE_PATH environment variable to this path, the modules can be found by node.
Windows XP - %USERPROFILE%\Application Data\npm\node_modules
Windows 7, 8 and 10 - %AppData%\npm\node_modules

Then again under node-opcua-client/certificates and node-opcua-server\certificates.

corn98, 2018-03-21:
I tried to search (through search tool) globally in the system "node-opcua-client" folder but it doesn't exist.

Now I tried with without security mode and connection works. 

Coming back from "without security mode" to "security mode" another error occurs:

"Read OPC UA connection error: Connection has been closed by client , but this transaction cannot be honored"

Which is the case that make this error occurs?

Corn

mikakaraila, 2018-03-21:
Ack, I will test at home later this week with Pi 3 model B and with global installation.
Then I can look security..

corn98, 2018-03-21:
Thanks for your dedication Mika!

Hear you as soon as possible, with the required time!

Corn

mikakaraila, 2018-03-24:
I checked my code and it uses relative path to certificates. 
So it cannot use certificates from the global installation...

corn98, 2018-03-26:
So, the suggestion is to install the package node-red-contrib-opcua in 

...USER/.node-red/node-modules              ?

mikakaraila, 2018-03-26:
Yes, it is easiest and fastest way at the moment.

mikakaraila, 2018-04-16:
Test with the latest, now it is possible to define path to certificates.

aadrii, 2018-07-30:
Hi,
May I suggest you to use the 'tls-config' node, which is a default node type provided by node-red, and used by core nodes (http, mqtt, websockets). Even if node-opcua needs a path to the certificate file, the tls-config form is pretty cool :)
Best regards

mikakaraila, 2020-07-07:
Latest version finds from the global installed or locally installed package and certificates this way

closed 2020-07-07 by mikakaraila