mikakaraila/node-red-contrib-opcua #76: addVariable seems not to work anymore

Fietje10, 2018-04-05:
Hi,

thank you for the great work, which I used in some former projects on a Raspberry Pi to access and provide OPC-UA server/client connections. Now on another Node-Red instance on a windows PC in the newest version there seems to be a problem injecting an "addVariable" command leading to the following error:

""AssertionError [ERR_ASSERTION]: this.identifierType""

Was there any recent change that might caused this ? Adding a folder is still possible without any issues. I also tried your example OPCUA-TEST.json, just in case I did something wrong, but it shows the same issue.

Thanks !
Fietje

mikakaraila, 2018-04-07:
Ack, I will check this.

mikakaraila, 2018-04-08:
Fixed, addVariable parameter dataType must be string not opcua type. I pushed new version with the wrong issue number...

closed 2018-04-08 by mikakaraila