mikakaraila/node-red-contrib-opcua #77: No nodes appearing

Gawel13, 2018-04-06:
Dear Mika,
I have been trying to install it but I can't get it to work... No node appear in node-red. No error messages during installations, just this when I start node red:
6 Apr 11:35:46 - [info] Node-RED version: v0.18.4
6 Apr 11:35:46 - [info] Node.js  version: v8.11.1
6 Apr 11:35:46 - [info] Windows_NT 10.0.16299 x64 LE
6 Apr 11:35:47 - [info] Loading palette nodes
6 Apr 11:35:48 - [warn] ------------------------------------------------------
6 Apr 11:35:48 - [warn] [node-red/rpi-gpio] Info : Ignoring Raspberry Pi specific node
6 Apr 11:35:48 - [warn] [node-red/tail] Not currently supported on Windows.
6 Apr 11:35:48 - [warn] [node-red-contrib-opcua/OpcUa-Item] Error: Cannot find module 'semver'
6 Apr 11:35:48 - [warn] [node-red-contrib-opcua/OpcUa-Client] Error: Cannot find module 'semver'
6 Apr 11:35:48 - [warn] [node-red-contrib-opcua/OpcUa-Browser] Error: Cannot find module 'semver'
6 Apr 11:35:48 - [warn] [node-red-contrib-opcua/OpcUa-Server] Error: Cannot find module 'semver'
6 Apr 11:35:48 - [warn] [node-red-contrib-opcua/OpcUa-Event] Error: Cannot find module 'semver'
6 Apr 11:35:48 - [warn] ------------------------------------------------------
6 Apr 11:35:48 - [info] Settings file  : \Users\Gaël\.node-red\settings.js
6 Apr 11:35:48 - [info] User directory : \Users\Gaël\.node-red
6 Apr 11:35:48 - [warn] Projects disabled : set editorTheme.projects.enabled=true to enable
6 Apr 11:35:48 - [info] Flows file     : \Users\Gaël\.node-red\flows_GAEL-LAPTOP.json
6 Apr 11:35:48 - [info] Creating new flow file
6 Apr 11:35:48 - [info] Starting flows
6 Apr 11:35:48 - [info] Started flows
6 Apr 11:35:48 - [info] Server now running at http://127.0.0.1:1880/

Any idea ? I guessed a failed installation, I suppress and re-install all without any results.

petergmorgan, 2018-04-06:
Install the semver module  using `npm i -g semver `  then start node-red and the warning

> "Error: Cannot find module 'semver'"

 will go away and the opcua nodes should appear in the palette.

mikakaraila, 2018-04-06:
It is fixed latest node-opcua...

mikakaraila, 2018-04-08:
Fixed, addVariable parameter dataType must be string not opcua type.

mikakaraila, 2018-04-08:
Sorry wrong issue, this is waiting for new node-opcua version.

Gawel13, 2018-04-10:
A last full delete, reinstallation and installation of semver has fixed the problem.
Thank you petergmorgan and mikakaraila !

closed 2018-04-10 by Gawel13