mikakaraila/node-red-contrib-opcua #81: Unsubscribe

stephb59, 2018-05-28:
Hi Mika,
Than you for the work done on that subject. I am able to use this plugin to connect to a B&R PLC in order to collect data from it. I used the subscription method to receive data from it. I tried unsuccessfully to unsubscribe from that opc server following the sample flow provided.
Here's my flow. 
<details/>
[{"id":"566cd0e.ec0263","type":"tab","label":"Flow 1","disabled":false,"info":""},{"id":"75f8baae.2681ec","type":"OpcUa-Item","z":"566cd0e.ec0263","item":"ns=6;s=::ServerTask:VarZ","datatype":"Int16","value":"","name":"","x":382,"y":459.5,"wires":[["50ac7957.3e13d8"]]},{"id":"50ac7957.3e13d8","type":"OpcUa-Client","z":"566cd0e.ec0263","endpoint":"bb326e1e.6c2a98","action":"subscribe","time":"1","timeUnit":"s","name":"","x":663,"y":305.5,"wires":[["6c6a1211.2cb9ac","2aa94cac.3f6f5c","4d6bff5d.0c97e","6af6c727.9f6c3"]],"inputLabels":["item a souscrire"]},{"id":"6c6a1211.2cb9ac","type":"debug","z":"566cd0e.ec0263","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":1059.7666320800781,"y":484.2500305175781,"wires":[]},{"id":"2aa94cac.3f6f5c","type":"ui_gauge","z":"566cd0e.ec0263","name":"","group":"fcb379b1.261ed","order":0,"width":0,"height":0,"gtype":"gage","title":"gauge p1","label":"bar","format":"{{value}}","min":0,"max":"400","colors":["#00b500","#e6e600","#ca3838"],"seg1":"300","seg2":"330","x":1046.7666015625,"y":366.6499938964844,"wires":[]},{"id":"4d6bff5d.0c97e","type":"ui_chart","z":"566cd0e.ec0263","name":"","group":"fcb379b1.261ed","order":0,"width":0,"height":0,"label":"chart","chartType":"line","legend":"false","xformat":"HH:mm:ss","interpolate":"linear","nodata":"","dot":false,"ymin":"","ymax":"","removeOlder":1,"removeOlderPoints":"","removeOlderUnit":"3600","cutout":0,"useOneColor":false,"colors":["#1f77b4","#aec7e8","#ff7f0e","#2ca02c","#98df8a","#d62728","#ff9896","#9467bd","#c5b0d5"],"useOldStyle":false,"x":1038.7666320800781,"y":274.7166442871094,"wires":[[],[]]},{"id":"9093b068.b3a508","type":"ui_button","z":"566cd0e.ec0263","name":"","group":"55af4a6f.ab0fdc","order":0,"width":0,"height":0,"passthru":false,"label":"Subscribe","color":"","bgcolor":"","icon":"","payload":"","payloadType":"str","topic":"","x":213,"y":334,"wires":[["75f8baae.2681ec"]]},{"id":"4e8005d6.8f9ddc","type":"function","z":"566cd0e.ec0263","name":"Unsubscribe","func":"msg.action=\"unsubscribe\";\nreturn msg;","outputs":1,"noerr":0,"x":556,"y":619,"wires":[["50ac7957.3e13d8","df284572.78152"]]},{"id":"9551676e.b55f4","type":"ui_button","z":"566cd0e.ec0263","name":"","group":"fcb379b1.261ed","order":0,"width":0,"height":0,"passthru":false,"label":"Unsubscribe","color":"","bgcolor":"","icon":"","payload":"","payloadType":"str","topic":"","x":185,"y":544,"wires":[["6bd1c878.54eee"]]},{"id":"6bd1c878.54eee","type":"OpcUa-Item","z":"566cd0e.ec0263","item":"ns=6;s=::ServerTask:VarZ","datatype":"Int16","value":"","name":"","x":345,"y":653,"wires":[["4e8005d6.8f9ddc"]]},{"id":"6af6c727.9f6c3","type":"function","z":"566cd0e.ec0263","name":"INSERT statement","func":"msg.topic = \"INSERT INTO `mydata`(`varZ`) VALUES (\" + msg.payload + \")\"\nreturn msg;","outputs":1,"noerr":0,"x":917,"y":191,"wires":[["6d582dbe.64a4f4","42d57654.4e4e48"]]},{"id":"6d582dbe.64a4f4","type":"debug","z":"566cd0e.ec0263","name":"","active":false,"tosidebar":true,"console":false,"tostatus":false,"complete":"false","x":1181,"y":60,"wires":[]},{"id":"42d57654.4e4e48","type":"mysql","z":"566cd0e.ec0263","mydb":"fb1b5b0c.f3a2c","name":"","x":1273.5,"y":123,"wires":[[]]},{"id":"df284572.78152","type":"debug","z":"566cd0e.ec0263","name":"","active":true,"tosidebar":true,"console":false,"tostatus":false,"complete":"action","x":800,"y":555,"wires":[]},{"id":"bb326e1e.6c2a98","type":"OpcUa-Endpoint","z":"","endpoint":"opc.tcp://localhost:4841","secpol":"None","secmode":"NONE","login":false},{"id":"fcb379b1.261ed","type":"ui_group","z":"","name":"p1","tab":"6db06d46.0b1154","disp":true,"width":"6","collapse":false},{"id":"55af4a6f.ab0fdc","type":"ui_group","z":"","name":"Général page 1","tab":"6db06d46.0b1154","disp":true,"width":"6","collapse":false},{"id":"fb1b5b0c.f3a2c","type":"MySQLdatabase","z":"","host":"localhost","port":"3306","db":"data","tz":""},{"id":"6db06d46.0b1154","type":"ui_tab","z":"","name":"PIERRET","icon":"dashboard"}]
</details>

I have the following error in the debug window : "AssertionError [ERR_ASSERTION]: _.isFunction(done)"
Any idea ?
Thanks in advance

mikakaraila, 2018-05-28:
Ack, I will look this on Friday.

mikakaraila, 2018-06-03:
Code was out of date. New node-opcua uses monitoredItems through session. Fixed and updated example.

closed 2018-06-03 by mikakaraila

stephb59, 2018-06-05:
Hi Mika,
Sorry to open it again but i tested it again following recent review of the example and in my case it is not ok.
I just take out a short video of the problem.

[OPCUA.zip](https://github.com/mikakaraila/node-red-contrib-opcua/files/2072147/OPCUA.zip)

commit f2ee6b33 (2018-06-03): Fixed #81, updated code to match node-opcua refactored code.