mikakaraila/node-red-contrib-opcua #82: Unsubscribe

stephb59, 2018-06-05:
Hi Mika,
Sorry to ask to open #81  again but i tested it again following recent review of the example and in my case it is not ok.
I just take out a short video of the problem.
[OPCUA.zip](https://github.com/mikakaraila/node-red-contrib-opcua/files/2072237/OPCUA.zip)

mikakaraila, 2018-06-05:
Did you install also the latest node-opcua v0.4.1? 
Because framework has changed and my modifications are now according the latest one?
Or are you using global node-opcua? You can check this with command "npm list".

stephb59, 2018-06-05:
Mika here's the command output on my  server
[image]
I can not find something related to opcua ?
In the graphical interface (manage palette ) i can see node-red-contrib-opcua 0.2.28, upgrade possible to 0.2.31. 
How to upgrade to v0.4.1 ?
Do i need to install it from CLI ? If i remember well i had issues earlier installing from CLI and i had to install from graphical view.
Regards,
Stéphane

stephb59, 2018-06-05:
Just installed node-opcua like requested.
Here's the npm list output
<details/>
C:\Users\stb\.node-red>npm list
node-red-project@0.0.1 C:\Users\stb\.node-red
+-- node-opcua@0.4.1
| +-- colors@1.3.0
| +-- node-opcua-address-space@0.4.0
| | +-- async@2.6.1
| | | `-- lodash@4.17.10
| | +-- dequeue@1.0.5
| | +-- node-opcua-assert@0.4.0 deduped
| | +-- node-opcua-basic-types@0.4.0 deduped
| | +-- node-opcua-common@0.4.0 deduped
| | +-- node-opcua-constants@0.2.0 deduped
| | +-- node-opcua-data-access@0.4.0 deduped
| | +-- node-opcua-data-model@0.4.0 deduped
| | +-- node-opcua-data-value@0.4.0 deduped
| | +-- node-opcua-date-time@0.4.0
| | | +-- long@4.0.0
| | | +-- node-opcua-assert@0.4.0 deduped
| | | `-- underscore@1.9.1 deduped
| | +-- node-opcua-debug@0.4.0 deduped
| | +-- node-opcua-enum@0.4.0 deduped
| | +-- node-opcua-factory@0.4.0 deduped
| | +-- node-opcua-nodeid@0.4.0 deduped
| | +-- node-opcua-numeric-range@0.4.0 deduped
| | +-- node-opcua-object-registry@0.4.0
| | | +-- node-opcua-assert@0.4.0 deduped
| | | +-- node-opcua-debug@0.4.0 deduped
| | | `-- underscore@1.9.1 deduped
| | +-- node-opcua-service-browse@0.4.0 deduped
| | +-- node-opcua-service-call@0.4.0 deduped
| | +-- node-opcua-service-filter@0.4.0 deduped
| | +-- node-opcua-service-history@0.4.0 deduped
| | +-- node-opcua-service-translate-browse-path@0.4.0 deduped
| | +-- node-opcua-service-write@0.4.0 deduped
| | +-- node-opcua-status-code@0.4.0 deduped
| | +-- node-opcua-utils@0.4.0 deduped
| | +-- node-opcua-variant@0.4.0 deduped
| | +-- node-opcua-xml2json@0.4.0
| | | +-- bomstrip@0.1.4
| | | +-- ltx@2.7.1
| | | | `-- inherits@2.0.3 deduped
| | | +-- node-opcua-assert@0.4.0 deduped
| | | `-- underscore@1.9.1 deduped
| | +-- set-prototype-of@1.0.0
| | +-- underscore@1.9.1
| | `-- xml-writer@1.7.0
| +-- node-opcua-address-space-for-conformance-testing@0.4.0
| | +-- node-opcua-address-space@0.4.0 deduped
| | +-- node-opcua-assert@0.4.0 deduped
| | +-- node-opcua-basic-types@0.4.0 deduped
| | +-- node-opcua-data-access@0.4.0 deduped
| | +-- node-opcua-data-model@0.4.0 deduped
| | +-- node-opcua-data-value@0.4.0 deduped
| | +-- node-opcua-factory@0.4.0 deduped
| | +-- node-opcua-nodeid@0.4.0 deduped
| | +-- node-opcua-status-code@0.4.0 deduped
| | +-- node-opcua-variant@0.4.0 deduped
| | `-- underscore@1.9.1 deduped
| +-- node-opcua-assert@0.4.0
| | +-- better-assert@1.0.2
| | | `-- callsite@1.0.0
| | `-- chalk@2.4.1
| |   +-- ansi-styles@3.2.1
| |   | `-- color-convert@1.9.1
| |   |   `-- color-name@1.1.3
| |   +-- escape-string-regexp@1.0.5
| |   `-- supports-color@5.4.0
[... 686 lines cut ...]
|   | `-- to-array@0.1.4
|   `-- socket.io-parser@2.3.1
|     +-- component-emitter@1.1.2
|     +-- debug@2.2.0
|     | `-- ms@0.7.1
|     +-- isarray@0.0.1
|     `-- json3@3.3.2
`-- node-red-node-mysql@0.0.16
  `-- mysql@2.15.0
    +-- bignumber.js@4.0.4
    +-- readable-stream@2.3.3
    | +-- core-util-is@1.0.2
    | +-- inherits@2.0.3
    | +-- isarray@1.0.0
    | +-- process-nextick-args@1.0.7
    | +-- safe-buffer@5.1.1 deduped
    | +-- string_decoder@1.0.3
    | | `-- safe-buffer@5.1.1 deduped
    | `-- util-deprecate@1.0.2
    +-- safe-buffer@5.1.1 deduped
    `-- sqlstring@2.3.0
</details>
The situation  remain the same. I still have the assertion error.
Regards,
Stéphane

stephb59, 2018-06-05:
Just installed node-opcua like requested.
Here's the npm list output
The situation remain the same. Still have the assertion error

```
C:\Users\stb\.node-red>npm list
node-red-project@0.0.1 C:\Users\stb\.node-red
+-- node-opcua@0.4.1
| +-- colors@1.3.0
| +-- node-opcua-address-space@0.4.0
| | +-- async@2.6.1
| | | `-- lodash@4.17.10
| | +-- dequeue@1.0.5
| | +-- node-opcua-assert@0.4.0 deduped
| | +-- node-opcua-basic-types@0.4.0 deduped
| | +-- node-opcua-common@0.4.0 deduped
| | +-- node-opcua-constants@0.2.0 deduped
| | +-- node-opcua-data-access@0.4.0 deduped
| | +-- node-opcua-data-model@0.4.0 deduped
| | +-- node-opcua-data-value@0.4.0 deduped
| | +-- node-opcua-date-time@0.4.0
| | | +-- long@4.0.0
| | | +-- node-opcua-assert@0.4.0 deduped
| | | `-- underscore@1.9.1 deduped
| | +-- node-opcua-debug@0.4.0 deduped
| | +-- node-opcua-enum@0.4.0 deduped
| | +-- node-opcua-factory@0.4.0 deduped
| | +-- node-opcua-nodeid@0.4.0 deduped
| | +-- node-opcua-numeric-range@0.4.0 deduped
| | +-- node-opcua-object-registry@0.4.0
| | | +-- node-opcua-assert@0.4.0 deduped
| | | +-- node-opcua-debug@0.4.0 deduped
| | | `-- underscore@1.9.1 deduped
| | +-- node-opcua-service-browse@0.4.0 deduped
| | +-- node-opcua-service-call@0.4.0 deduped
| | +-- node-opcua-service-filter@0.4.0 deduped
| | +-- node-opcua-service-history@0.4.0 deduped
| | +-- node-opcua-service-translate-browse-path@0.4.0 deduped
| | +-- node-opcua-service-write@0.4.0 deduped
| | +-- node-opcua-status-code@0.4.0 deduped
| | +-- node-opcua-utils@0.4.0 deduped
| | +-- node-opcua-variant@0.4.0 deduped
| | +-- node-opcua-xml2json@0.4.0
| | | +-- bomstrip@0.1.4
| | | +-- ltx@2.7.1
| | | | `-- inherits@2.0.3 deduped
| | | +-- node-opcua-assert@0.4.0 deduped
| | | `-- underscore@1.9.1 deduped
| | +-- set-prototype-of@1.0.0
| | +-- underscore@1.9.1
| | `-- xml-writer@1.7.0
| +-- node-opcua-address-space-for-conformance-testing@0.4.0
| | +-- node-opcua-address-space@0.4.0 deduped
| | +-- node-opcua-assert@0.4.0 deduped
| | +-- node-opcua-basic-types@0.4.0 deduped
| | +-- node-opcua-data-access@0.4.0 deduped
| | +-- node-opcua-data-model@0.4.0 deduped
| | +-- node-opcua-data-value@0.4.0 deduped
| | +-- node-opcua-factory@0.4.0 deduped
| | +-- node-opcua-nodeid@0.4.0 deduped
| | +-- node-opcua-status-code@0.4.0 deduped
| | +-- node-opcua-variant@0.4.0 deduped
| | `-- underscore@1.9.1 deduped
| +-- node-opcua-assert@0.4.0
| | +-- better-assert@1.0.2
| | | `-- callsite@1.0.0
| | `-- chalk@2.4.1
| |   +-- ansi-styles@3.2.1
| |   | `-- color-convert@1.9.1
| |   |   `-- color-name@1.1.3
[... 686 lines cut ...]
|   | | `-- better-assert@1.0.2 deduped
|   | +-- socket.io-parser@2.3.1 deduped
|   | `-- to-array@0.1.4
|   `-- socket.io-parser@2.3.1
|     +-- component-emitter@1.1.2
|     +-- debug@2.2.0
|     | `-- ms@0.7.1
|     +-- isarray@0.0.1
|     `-- json3@3.3.2
`-- node-red-node-mysql@0.0.16
  `-- mysql@2.15.0
    +-- bignumber.js@4.0.4
    +-- readable-stream@2.3.3
    | +-- core-util-is@1.0.2
    | +-- inherits@2.0.3
    | +-- isarray@1.0.0
    | +-- process-nextick-args@1.0.7
    | +-- safe-buffer@5.1.1 deduped
    | +-- string_decoder@1.0.3
    | | `-- safe-buffer@5.1.1 deduped
    | `-- util-deprecate@1.0.2
    +-- safe-buffer@5.1.1 deduped
    `-- sqlstring@2.3.0

```

mikakaraila, 2018-06-05:
You can run node-red -v OPCUA-UNSUBSCRIBE.json

But look also 102-opcuaclient.js in my code file there is this function
that will do unscubscribe:
I wonder if palette loads node implementation to another folder than CLI...
can you search if the 102-opcuaclient.js is multiple times on your hard
drive?

stephb59, 2018-06-06:
Here's the output from node red when unsubscribing

```
6 Jun 09:38:28 - [info] [OpcUa-Client:2b4e6634.3785a2] initialized monitoredItem
 on ns=6;s=::AsGlobalPV:Mode_Travail
6 Jun 09:38:29 - [info] [OpcUa-Client:2b4e6634.3785a2] Client status: active sub
scribed
6 Jun 09:38:29 - [info] [OpcUa-Client:2b4e6634.3785a2] ns=6;s=::AsGlobalPV:Mode_
Travail value has changed to 5
6 Jun 09:38:29 - [info] [OpcUa-Client:2b4e6634.3785a2] DataValue:
   value:           Variant(Scalar<Byte>, value: 5)
   statusCode:      Good (0x00000)
   serverTimestamp: 2018-06-06T07:38:28.840Z $ 0
   sourceTimestamp: 2018-06-06T07:38:28.840Z $ 0
6 Jun 09:38:29 - [info] [OpcUa-Client:2b4e6634.3785a2]  Status-Code:Good (0x0000
0)
6 Jun 09:38:32 - [info] [OpcUa-Client:2b4e6634.3785a2] Subscription keepalive ID
: 4058921796
6 Jun 09:38:32 - [info] [OpcUa-Client:2b4e6634.3785a2] Client status: keepalive
6 Jun 09:38:32 - [info] [OpcUa-Item:aba024e0.805a6] node value:
6 Jun 09:38:32 - [info] [OpcUa-Client:2b4e6634.3785a2] Action on msg:unsubscribe

6 Jun 09:38:32 - [info] [OpcUa-Client:2b4e6634.3785a2] Action on input:unsubscri
be Item from Topic: ns=6;s=::AsGlobalPV:Mode_Travail session Id: ns=1;i=40589317
98
6 Jun 09:38:32 - [info] [OpcUa-Client:2b4e6634.3785a2] unsubscribing
6 Jun 09:38:32 - [info] [OpcUa-Client:2b4e6634.3785a2] Client status: active sub
scribing
6 Jun 09:38:32 - [info] [OpcUa-Client:2b4e6634.3785a2] Session subscriptionId: 4
058921796
6 Jun 09:38:32 - [error] [OpcUa-Client:2b4e6634.3785a2] AssertionError [ERR_ASSE
RTION]: _.isFunction(done)
    at ClientMonitoredItem.terminate (C:\Users\stb\AppData\Roaming\npm\node_modu
les\node-red-contrib-opcua\node_modules\node-opcua-client\src\client_monitored_i
tem.js:67:5)
    at unsubscribe_monitoredItem (C:\Users\stb\AppData\Roaming\npm\node_modules\
node-red-contrib-opcua\opcua\102-opcuaclient.js:752:31)
    at unsubscribe_action_input (C:\Users\stb\AppData\Roaming\npm\node_modules\n
ode-red-contrib-opcua\opcua\102-opcuaclient.js:615:11)
    at OpcUaClientNode.processInputMsg (C:\Users\stb\AppData\Roaming\npm\node_mo
dules\node-red-contrib-opcua\opcua\102-opcuaclient.js:368:11)
    at emitOne (events.js:116:13)
    at OpcUaClientNode.emit (events.js:211:7)
    at OpcUaClientNode.Node.receive (C:\Users\stb\AppData\Roaming\npm\node_modul
es\node-red\red\runtime\nodes\Node.js:215:14)
    at FunctionNode.Node.send (C:\Users\stb\AppData\Roaming\npm\node_modules\nod
e-red\red\runtime\nodes\Node.js:202:14)
    at sendResults (C:\Users\stb\AppData\Roaming\npm\node_modules\node-red\nodes
\core\core\80-function.js:52:18)
    at FunctionNode.<anonymous> (C:\Users\stb\AppData\Roaming\npm\node_modules\n
ode-red\nodes\core\core\80-function.js:220:21)
6 Jun 09:38:35 - [info] [OpcUa-Client:2b4e6634.3785a2] Subscription keepalive ID
: 4058921796
6 Jun 09:38:35 - [info] [OpcUa-Client:2b4e6634.3785a2] Client status: keepalive
6 Jun 09:38:38 - [info] [OpcUa-Client:2b4e6634.3785a2] Subscription keepalive ID
: 4058921796

```
I have done search for 102-opcuaclient.js on the hard disk drive, there is only one located at C:\Users\stb\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\opcua

mikakaraila, 2018-06-06:
Ok check that in your
C:\Users\stb\AppData\Roaming\npm\node_modules\node-red-contrib-opcua\opcua\102-opcuaclient.js
is same function on line 752 as I have:

It should print out to debug tab " Invalid empty node in getObject".

stephb59, 2018-06-07:
Mika,
Did the comparison between github and my local 102-opcuaclient.js.
Here it is 
[102-opcuaclient compare stb github.zip](https://github.com/mikakaraila/node-red-contrib-opcua/files/2079271/102-opcuaclient.compare.stb.github.zip)
You can see differences.
I then replaced local file by the github one and unsubscribe seems to work perfectly.
Many thanks for your support. I really appreciate it